realsenseai/realsense-ros #2315: D455 SLAM

w-1016-Papman, 2022-04-11:
Hi buddy. I would like to ask you for some more detailed official information and materials about 'D455 Camera SLAM', I think you will be of great help to me, I have followed this [tutorial](https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i) successfully to display images in rviz and can save and play back rosbag files according to the documentation prompts. I'm curious to know: can I write and optimize SLAM solutions in python, does Intel have a relevant python library for that.
                     
thanks.

MartyG-RealSense, 2022-04-11:
Hi @w-1016-Papman  If you wanted a purely Python SLAM library rather than a Python binding for a SLAM project not originally designed for Python then **pySLAM-D** or **pySLAM** might suit your needs.  These are not projects that are tested for RealSense compatibility though.

https://github.com/armandok/pySLAM-D
https://github.com/luigifreda/pyslam

In regard to binding-based solutions, ORB2-SLAM has been used successfully in the past with RealSense 400 Series cameras such as D435, and the link below has a Python binding for ORB2-SLAM.

https://github.com/jskinn/ORB_SLAM2-PythonBindings/blob/master/README.md

w-1016-Papman, 2022-04-12:
Your information is very good, besides, Because I'm very satisfied with the D435i SLAM guide in the [IntelRealsense/realsense-ros](https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i) official repository, I was amazed by the effect it showed, so I would also like to know if you have more detailed information about 【SLAM with D435i】. Are there any more detailed and specific learning materials about "SLAM with D435i" that are publicly available, which will allow me to understand and master [SLAM with D435I] in realsense-ros more quickly.

Although my camera is D455, I know that both D455 and D435i use [SLAM with D435i]

MartyG-RealSense, 2022-04-12:
You could also investigate adding obstacle avoidance to your SLAM setup.  A RealSense ROS user at https://github.com/IntelRealSense/librealsense/issues/9294 did it with a D435 and **depthimage_to_laserscan**

https://github.com/ros-perception/depthimage_to_laserscan

Also, if you have more than one RealSense 400 Series camera then a natural next step may be to try to add a second camera to your SLAM setup by using the advice in Intel's multiple camera ROS guide in the link below.

https://www.intelrealsense.com/how-to-multiple-camera-setup-with-ros/

closed 2022-04-12 by w-1016-Papman