realsenseai/realsense-ros #2317: URDF for Intel D405

DimitrisKld, 2022-04-13:
Hi, 

I would like to ask you if there is somewhere the URDF for the Intel D405 camera or if you are going to upload it soon. I checked the branches related to the ROS2 version, which is supported by Intel D405, but I could not find the URDF file. 

Thank you in advance.

MartyG-RealSense, 2022-04-13:
Hi @DimitrisKld  New RealSense camera models do typically have URDF files for them added to the ROS wrapper over time, though I do not have information about plans for the release of a URDF for D405 for the ROS2 wrapper at the time of writing this.

MartyG-RealSense, 2022-04-22:
Hi @DimitrisKld  Do you require further assistance with this case, please?  Thanks!

DimitrisKld, 2022-04-25:
Hi @MartyG-RealSense , thank you for your response. No, I will follow the new releases in case this URDF file will be added.

MartyG-RealSense, 2022-04-25:
Thanks very much @DimitrisKld for the update - I will close this case as you will be following the new releases.  Thanks again!

closed 2022-04-25 by MartyG-RealSense

MartyG-RealSense, 2023-01-03:
Hi again @DimitrisKld  I wanted to update you that the RealSense ROS developer team has advised me that the addition of a URDF for D405 is planned for a future wrapper release.

weaponeer, 2023-03-07:
Also have been waiting for URDF/dae for the d405, any insight into when in the future these might be added to ros2-development?

SamerKhshiboun, 2023-04-11:
Hi @weaponeer 

> Also have been waiting for URDF/dae for the d405, any insight into when in the future these might be added to ros2-development?

While we are working to add D405 URDF, You can download the cad file of D405 from here:
https://dev.intelrealsense.com/docs/stereo-depth-camera-d400

Then you can convert this STEP file to dae and stl files 
(I managed to download FreeCAD, and convert (File -> Export) these STEP file to .stl and .dae files)

MartyG-RealSense, 2023-12-26:
Hi @weaponeer  A URDF file for D405 has been added to the ros2-development wrapper - please see https://github.com/IntelRealSense/realsense-ros/pull/2953#event-11339930771

weaponeer, 2023-12-26:
I see it there in ros2-development. 

Many Thanks!