realsenseai/realsense-ros #2320: Auto exposure of color sensor not working

LynxStassie, 2022-04-14:
I am trying to use _auto_exposure_ in **color sensor** via `color_sensor.set_option(rs.option.enable_auto_exposure, True)` option. And this function doesn't work. The exposure value is not automatically adjusted. 
Depth sensor works fine and _auto exposure_ is enabled. 
Are there any specific options to use _auto exposure_ with the **RGB sensor** of D455 camera?

MartyG-RealSense, 2022-04-14:
Hi @LynxStassie   Have you defined your **color_sensor** with an index number of '1' so that it accesses the color sensor instead of the depth sensor (which is index number '0')?  For example:

```
color_sensor = profile.get_device().query_sensors()[1]
color_sensor.set_option(rs.option.enable_auto_exposure, 1)
```

LynxStassie, 2022-04-16:
Hey @MartyG-RealSense 
Thank you for your reply! Yes, I have set the indexes for the sensors. 
And auto exposure for the RGB sensor works in RealSense Viewer via the button "Enable Auto Exposure". But when I set the auto exposure it doesn't make any effect under the changing light conditions. Maybe there is a way to detect auto exposure settings in the camera logs? And how can it be verified that the image is created by the sensor that works with auto exposure enabled? Is there a way to get the color image from the certain RGB sensor with auto exposure enabled? 

Intel RealSense D455 (FW: 05.13.00.50  on USB3.2) 
Windows 10

MartyG-RealSense, 2022-04-16:
Whilst depth exposure can be read from camera metadata, **RGB** exposure values cannot be read when auto-exposure is enabled, as advised by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/7267#issuecomment-688295655

When auto-exposure is enabled, exposure values do not update and so could give a false impression that the exposure value is not changing when auto-exposure is enabled.  Instead, you have to read the 'actual exposure value' from metadata instead.  As mentioned above though, retrieving actual **RGB** exposure from metadata is not supported.

If it appears that RGB auto-exposure is not working correctly then there may be additional environmental factors involved in this problem.  On page 8 (the final page) of the **Specification Clarification** Intel PDF guide linked to below, there is official advice specific to the D455 camera model about color problems that may be experienced by the RGB sensor under certain lighting conditions.  The PDF should automatically begin downloading in your browser when left-clicked on.

https://www.intelrealsense.com/download/13629/

LynxStassie, 2022-04-16:
@MartyG-RealSense The environmental conditions are ok. In the RealSense Viewer program auto exposure is successfully enabled with the "Enable auto exposure" button. So, in cpp this function works fine. The problem is to make it work in python code also.

MartyG-RealSense, 2022-04-16:
May I confirm that your Python script is not setting an exposure value like the script at https://github.com/IntelRealSense/librealsense/issues/4449#issue-469308511 does, as setting a manual exposure value automatically disables auto-exposure.

LynxStassie, 2022-04-17:
@MartyG-RealSense The exposure value is not set manually.  i want to get atomatic control of exposure color image like when i runing intel realsence viewer and click on "auto exposure enable" button and get that atomatic controls of frames exposure while "rgb camera" is opened. 
[image]
Can I ask you how does the RealSense Viewer software (in c++ code) enable the auto exposure on mouse click the "Enable auto exposure" button? In which header and cpp file can i see the calls to camera that enable auto exposure mechanism in debug mode with running realsense-viewer.exe with realsense-viewer.pdb. I want to see the calls in WinDbg or Visual Studio (with debug mode attached to realsense-viewer.exe with realsense-viewer.pdb) and then to compare the results with python calls sequence.

MartyG-RealSense, 2022-04-18:
The code provided earlier in this discussion at https://github.com/IntelRealSense/realsense-ros/issues/2320#issuecomment-1099594426 for setting the sensor index to '1' (RGB) and then setting enable_auto_exposure to '1' to enable RGB exposure should work.  You could try printing the status of RGB auto-exposure after enabling it to confirm that it is enabled (1), even if it does not appear to be having the desired effect.

`print (color_sensor.get_option(rs.option.enable_auto_exposure))`

I believe that the Viewer's RGB auto-exposue status is handled in the C++ code at the link below.

https://github.com/IntelRealSense/librealsense/blob/master/common/model-views.cpp#L1171-L1181

closed 2022-05-04 by MartyG-RealSense