realsenseai/realsense-ros #2323: multiple D455 camera sync issue

snakehaihai, 2022-04-16:
Hi 
I`m using 1 Arduino as the master clock and Arduino is sending pulses to 2 D455 to sync. I used the same setup for IDS camera and pointgrey camera before, but there wasn`t an issue. But when triggering d455. I counter some issues.

I start slow with a 1sec pulse and 1-sec downtime to test out the system sync function. By right, i should get 0.5hz camera feed. But i couldn`t get it.

I`m using realsense ros 2.2.22 + librealsense-2.42.0
I called  rs_multiple_devices.launch  
then set the correct serial ID 

then i set the 
/camera1/stereo_module/intel_cam_sync_mode: 2
/camera2/stereo_module/intel_cam_sync_mode: 2

I connect the pin 5 to of D455 to pin 10 of arduino. and all ground are connected together. 

Below is my setup and my results

[image]

[image]

[image]
[image]

For one case, it has different hz

For another case, it is not sync

can someone suggest a working solution?

MartyG-RealSense, 2022-04-16:
Hi @snakehaihai  I have seen the error **mode doesn't support option #42** occur in a past case when inter_cam_sync_mode was set **after** launch during runtime, and the problem did not occur if defining the inter_cam_sync_mode values before launch.

When you set inter_cam_sync mode for each camera using the instructions below, are you doing it before launch (in the launch file or in a roslaunch instruction) or after launch, please?

```
/camera1/stereo_module/intel_cam_sync_mode: 2
/camera2/stereo_module/intel_cam_sync_mode: 2
```

snakehaihai, 2022-04-16:
Hi @MartyG-RealSense 

I tried both. I do add 

/camera1/stereo_module/intel_cam_sync_mode: 2
/camera2/stereo_module/intel_cam_sync_mode: 2

in rs_multiple_devices.launch.  Then i check the frequency, and the freqnecy is not the same

Then I tried to check in rqt_reconfigure and found it wasn`t set to correct mode. then I set manually

But none of the method works well

MartyG-RealSense, 2022-04-16:
Let's address first the **Invalid md size: bytes used =  0 ,start offset=10** that appears in your log before the inter_cam_sync_mode error occurs.  This will help to determine whether or not inter_cam_sync_mode is failing becayse of the invalid md size error that precedes it.

The invalid md size error has also occurred with librealsense 2.42.0 in the past case https://github.com/IntelRealSense/realsense-ros/issues/1713#issuecomment-783588112 

In that case, the log informed the user that they had built the wrapper with SDK 2.41.0 but were running it with 2.42.0.  Do you have a similar message in your log?

[image]

snakehaihai, 2022-04-16:
Hi @MartyG-RealSense 

I dont have the issue. I know the version has to match.

[image]

> ... logging to /home/nuc/.ros/log/cb1db09c-bd87-11ec-a71b-a46bb62701f5/roslaunch-nuc-6347.log
> Checking log directory for disk usage. This may take a while.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> ]2;/home/nuc/catkin_ws_drivers/src/realsense-ros-2.2.22/realsense2_camera/launch/rs_camera.launch
> [1mstarted roslaunch server http://192.168.1.5:45323/[0m
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /camera/realsense2_camera/accel_fps: -1
>  * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
>  * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
>  * /camera/realsense2_camera/align_depth: False
>  * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
>  * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
>  * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
>  * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
>  * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
>  * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
>  * /camera/realsense2_camera/allow_no_texture_points: False
>  * /camera/realsense2_camera/base_frame_id: camera_link
>  * /camera/realsense2_camera/calib_odom_file: 
>  * /camera/realsense2_camera/clip_distance: -2.0
>  * /camera/realsense2_camera/color_fps: 30
>  * /camera/realsense2_camera/color_frame_id: camera_color_frame
>  * /camera/realsense2_camera/color_height: 480
>  * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
>  * /camera/realsense2_camera/color_width: 640
>  * /camera/realsense2_camera/confidence_fps: -1
>  * /camera/realsense2_camera/confidence_height: -1
>  * /camera/realsense2_camera/confidence_width: -1
>  * /camera/realsense2_camera/depth_fps: 30
>  * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
>  * /camera/realsense2_camera/depth_height: 480
>  * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
>  * /camera/realsense2_camera/depth_width: 640
>  * /camera/realsense2_camera/device_type: 
>  * /camera/realsense2_camera/enable_accel: True
>  * /camera/realsense2_camera/enable_color: True
>  * /camera/realsense2_camera/enable_confidence: False
>  * /camera/realsense2_camera/enable_depth: True
>  * /camera/realsense2_camera/enable_fisheye1: False
>  * /camera/realsense2_camera/enable_fisheye2: False
>  * /camera/realsense2_camera/enable_fisheye: False
>  * /camera/realsense2_camera/enable_gyro: True
>  * /camera/realsense2_camera/enable_infra1: True
>  * /camera/realsense2_camera/enable_infra2: True
>  * /camera/realsense2_camera/enable_infra: False
>  * /camera/realsense2_camera/enable_pointcloud: True
>  * /camera/realsense2_camera/enable_pose: False
>  * /camera/realsense2_camera/enable_sync: False
>  * /camera/realsense2_camera/filters: 
>  * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
>  * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
>  * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
>  * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
>  * /camera/realsense2_camera/fisheye_fps: -1
>  * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
>  * /camera/realsense2_camera/fisheye_height: -1
>  * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
>  * /camera/realsense2_camera/fisheye_width: -1
>  * /camera/realsense2_camera/gyro_fps: -1
[... 88 lines cut ...]
>  * /camera/realsense2_camera/publish_tf: True
>  * /camera/realsense2_camera/rosbag_filename: 
>  * /camera/realsense2_camera/serial_no: 
>  * /camera/realsense2_camera/tf_publish_rate: 0.0
>  * /camera/realsense2_camera/topic_odom_in: odom_in
>  * /camera/realsense2_camera/unite_imu_method: linear_interpolation
>  * /camera/realsense2_camera/usb_port_id: 
>  * /rosdistro: melodic
>  * /rosversion: 1.14.12
> 
> NODES
>   /camera/
>     realsense2_camera (nodelet/nodelet)
>     realsense2_camera_manager (nodelet/nodelet)
> 
> [1mROS_MASTER_URI=http://localhost:11311[0m
> ]2;/home/nuc/catkin_ws_drivers/src/realsense-ros-2.2.22/realsense2_camera/launch/rs_camera.launch http://localhost:11311
> [1mprocess[camera/realsense2_camera_manager-1]: started with pid [6369][0m
> [1mprocess[camera/realsense2_camera-2]: started with pid [6370][0m
> [camera/realsense2_camera-2] killing on exit
> [camera/realsense2_camera_manager-1] killing on exit
> shutting down processing monitor...
> ... shutting down processing monitor complete
> [1mdone[0m
>

MartyG-RealSense, 2022-04-16:
And the problems still occur if you add **initial_reset:=true** to the multiple_devices.launch roslaunch instruction to reset the cameras at launch?

snakehaihai, 2022-04-16:
I added reset and reset works. but the frequency is still wrong. Below are the log and rostopic hz of 2 camera

`ROS_PACKAGE_PATH (CATKIN) is setting.
/home/nuc/dev_ws/src/camera_model_virals:/home/nuc/dev_ws/src/livox_ros_driver/livox_ros_driver:/home/nuc/dev_ws/src/serial:/home/nuc/dev_ws/src/uwb_driver:/home/nuc/dev_ws/src/myloc:/home/nuc/dev_ws/src/uwb_as:/home/nuc/dev_ws/src/virals:/opt/ros/melodic/share
ROS_INFO_TIME: INVISIBLE
Wi-Fi IP Address: 192.168.1.5
Wi-Fi AP SSID: NETGEAR29-5G-1
ROS_IP: 192.168.1.5
ROS_MASTER_URI: http://localhost:11311
nuc ~ $ source ./catkin_ws_drivers/
bash: source: ./catkin_ws_drivers/: is a directory
nuc ~ $ source ./catkin_ws_drivers/devel/setup.bash 
nuc ~ $ roslaunch realsense2_camera rs_multiple_devices.launch 
... logging to /home/nuc/.ros/log/9db2e87a-bd96-11ec-9aed-a46bb62701f5/roslaunch-nuc-2230.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.5:44653/

SUMMARY
========

PARAMETERS
 * /camera1/realsense2_camera/accel_fps: 0
 * /camera1/realsense2_camera/accel_frame_id: camera1_accel_frame
 * /camera1/realsense2_camera/accel_optical_frame_id: camera1_accel_opt...
 * /camera1/realsense2_camera/align_depth: False
 * /camera1/realsense2_camera/aligned_depth_to_color_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/allow_no_texture_points: False
 * /camera1/realsense2_camera/base_frame_id: camera1_link
 * /camera1/realsense2_camera/calib_odom_file: 
 * /camera1/realsense2_camera/clip_distance: -1.0
 * /camera1/realsense2_camera/color_fps: 30
 * /camera1/realsense2_camera/color_frame_id: camera1_color_frame
 * /camera1/realsense2_camera/color_height: 480
 * /camera1/realsense2_camera/color_optical_frame_id: camera1_color_opt...
 * /camera1/realsense2_camera/color_width: 640
 * /camera1/realsense2_camera/confidence_fps: 30
 * /camera1/realsense2_camera/confidence_height: 480
 * /camera1/realsense2_camera/confidence_width: 640
 * /camera1/realsense2_camera/depth_fps: 30
 * /camera1/realsense2_camera/depth_frame_id: camera1_depth_frame
 * /camera1/realsense2_camera/depth_height: 480
 * /camera1/realsense2_camera/depth_optical_frame_id: camera1_depth_opt...
 * /camera1/realsense2_camera/depth_width: 640
 * /camera1/realsense2_camera/device_type: 
 * /camera1/realsense2_camera/enable_accel: False
 * /camera1/realsense2_camera/enable_color: True
 * /camera1/realsense2_camera/enable_confidence: True
 * /camera1/realsense2_camera/enable_depth: True
 * /camera1/realsense2_camera/enable_fisheye1: False
 * /camera1/realsense2_camera/enable_fisheye2: False
 * /camera1/realsense2_camera/enable_fisheye: False
 * /camera1/realsense2_camera/enable_gyro: False
 * /camera1/realsense2_camera/enable_infra1: False
 * /camera1/realsense2_camera/enable_infra2: False
 * /camera1/realsense2_camera/enable_infra: False
 * /camera1/realsense2_camera/enable_pointcloud: False
 * /camera1/realsense2_camera/enable_pose: False
 * /camera1/realsense2_camera/enable_sync: False
 * /camera1/realsense2_camera/filters: 
 * /camera1/realsense2_camera/fisheye1_frame_id: camera1_fisheye1_...
 * /camera1/realsense2_camera/fisheye1_optical_frame_id: camera1_fisheye1_...
 * /camera1/realsense2_camera/fisheye2_frame_id: camera1_fisheye2_...
[... 386 lines cut ...]
[ INFO]: Sync Mode: Off
[ INFO]: Device Sensors: 
[ INFO]: Stereo Module was found.
[ INFO]: RGB Camera was found.
 16/04 23:05:26,263 WARNING [139971641997056] (backend-v4l2.cpp:1357) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
[ INFO]: Motion Module was found.
[ INFO]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO]: num_filters: 0
[ INFO]: Setting Dynamic reconfig parameters.
[ WARN]: Param '/camera1/rgb_camera/power_line_frequency' has value 3 that is not in the enum { {50Hz: 1} {60Hz: 2} {Disabled: 0} }. Removing this parameter from dynamic reconfigure options.
[ INFO]: Done Setting Dynamic reconfig parameters.
[ INFO]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO]: setupPublishers...
[ INFO]: Expected frequency for depth = 30.00000
[ INFO]: Expected frequency for color = 30.00000
[ INFO]: setupStreams...
[ INFO]: insert Depth to Stereo Module
[ INFO]: insert Color to RGB Camera
[ INFO]: SELECTED BASE:Depth, 0
[ INFO]: RealSense Node Is Up!
[ WARN]: 
 16/04 23:05:27,401 ERROR [139971465815808] (ds5-options.cpp:86) Asic Temperature value is not valid!

`

snakehaihai, 2022-04-16:
This is my current launch file

...

<launch>
  <arg name="serial_no_camera1"    			default="146222250676"/> 			
  <arg name="serial_no_camera2"    			default="043422251647"/> 			
  
  <arg name="camera1"              			default="camera1"/>		
  <arg name="camera2"              			default="camera2"/>		
  
  <arg name="tf_prefix_camera1"         default="$(arg camera1)"/>
  <arg name="tf_prefix_camera2"         default="$(arg camera2)"/>

  <arg name="initial_reset"             default="true"/>

  <group ns="$(arg camera1)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="serial_no"             value="$(arg serial_no_camera1)"/>
      <arg name="tf_prefix"         		value="$(arg tf_prefix_camera1)"/>
      <arg name="initial_reset"         value="$(arg initial_reset)"/>
    </include>
  </group>

  <group ns="$(arg camera2)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="serial_no"             value="$(arg serial_no_camera2)"/>
      <arg name="tf_prefix"		          value="$(arg tf_prefix_camera2)"/>
      <arg name="initial_reset"         value="$(arg initial_reset)"/>
    </include>
  </group>

  <rosparam>
    /camera1/stereo_module/intel_cam_sync_mode: 2
    /camera2/stereo_module/intel_cam_sync_mode: 2
  </rosparam>

</launch>
...

MartyG-RealSense, 2022-04-16:
Your launch log image shows a **likely requires patch for fourcc code RW16!** message.  This message can indicate that the Linux kernel has not been patched before the librealsense SDK was installed if the SDK was built from source code.  Patching the kernel adds support for camera **hardware metadata** in a source-cide build.

Metadata support is automatically included when building librealsense from Debian packages or building from source code with CMake and including the **-DFORCE_RSUSB_BACKEND=true** build flag.

Could you tell me whether you built librealsense from source code without using -DFORCE_RSUSB_BACKEND=true* please?

snakehaihai, 2022-04-16:
Hi I do patched using the provided script.

as shown here 

[image]

-DFORCE_RSUSB_BACKEND=true  this is called during the linux librealsense cmake phase??

snakehaihai, 2022-04-16:
Ok i`m re-patching it again , then use the Dconfig flag . Its going to take a while
I`ll update here once it is done

Thanks

Regards
SHenghai Yuan

MartyG-RealSense, 2022-04-16:
It looks as though you are performing a basic librealsense build with **cmake ../ -DCMAKE_BUILD_TYPE=Release**.  This builds librealsense in Release mode (compiled with optimizations) instead of in Debug mode.  This instruction will build librealsense in **V4L backend** mode (not RSUSB), meaning that the patch script needs to be run before librealsense is built with CMake.  It looks as though this is what you are doing.

May I confirm if you are building on a PC and not on an Nvidia Jetson board?  Jetson boards must use a different patch script.

snakehaihai, 2022-04-16:
> It looks as though you are performing a basic librealsense build with **cmake ../ -DCMAKE_BUILD_TYPE=Release**. This builds librealsense in Release mode (compiled with optimizations) instead of in [...]

Its a intel 11 gen NUC. I know different embedded PC need different patch. 

I`ve done the patching. now configuring

...
nuc librealsense-2.42.0 $ sudo ./scripts/patch-realsense-ubuntu-lts.sh
[sudo] password for nuc: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
bc is already the newest version (1.07.1-2).
build-essential is already the newest version (12.4ubuntu1).
git is already the newest version (1:2.17.1-1ubuntu0.10).
linux-headers-generic is already the newest version (4.15.0.175.164).
The following packages were automatically installed and are no longer required:
  fltk1.3-doc fluid hddtemp libfltk-cairo1.3 libfltk-forms1.3 libfltk-gl1.3 libfltk-images1.3 libfltk1.3
  libfltk1.3-dev libfwup1 libjs-jquery-ui libllvm9 linux-hwe-5.4-headers-5.4.0-100 linux-hwe-5.4-headers-5.4.0-86
  linux-hwe-5.4-headers-5.4.0-87 linux-hwe-5.4-headers-5.4.0-89 linux-hwe-5.4-headers-5.4.0-90
  linux-hwe-5.4-headers-5.4.0-91 linux-hwe-5.4-headers-5.4.0-92 linux-hwe-5.4-headers-5.4.0-94
  linux-hwe-5.4-headers-5.4.0-96 linux-hwe-5.4-headers-5.4.0-97 linux-hwe-5.4-headers-5.4.0-99
  python-backports.functools-lru-cache python-cycler python-matplotlib python-matplotlib-data python-opengl
  python-pyqt5.qtopengl python-pyqt5.qtwebkit python-subprocess32 python-tz ros-melodic-common-msgs
  ros-melodic-gl-dependency ros-melodic-image-geometry ros-melodic-qt-gui-py-common ros-melodic-qwt-dependency
  ros-melodic-rosboost-cfg ros-melodic-roscpp-core ros-melodic-roscpp-tutorials ros-melodic-roslang
  ros-melodic-rosmake ros-melodic-self-test ros-melodic-smach ros-melodic-smach-msgs ros-melodic-stage
  ros-melodic-urdf-parser-plugin ros-melodic-vision-opencv ros-melodic-visualization-marker-tutorials
  ros-melodic-webkit-dependency sbcl
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 136 not upgraded.
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
Create patches workspace in ubuntu-bionic-hwe-5.4 folder

Package required libelf-dev:  - found
Package required elfutils:  - found
Package required bison:  - found
Package required flex:  - found
fatal: 'origin' does not appear to be a git repository
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
Fetching Ubuntu LTS tag   to the local kernel sources folder
fatal: unsafe repository ('/home/nuc/library/librealsense-2.42.0/ubuntu-bionic-hwe-5.4' is owned by someone else)
To add an exception for this directory, call:

	git config --global --add safe.directory /home/nuc/library/librealsense-2.42.0/ubuntu-bionic-hwe-5.4
nuc librealsense-2.42.0 $ git config --global --add safe.directory /home/nuc/library/librealsense-2.42.0/ubuntu-bionic-hwe-5.4
nuc librealsense-2.42.0 $ sudo ./scripts/patch-realsense-ubuntu-lts.sh
Reading package lists... Done
Building dependency tree       
Reading state information... Done
bc is already the newest version (1.07.1-2).
build-essential is already the newest version (12.4ubuntu1).
git is already the newest version (1:2.17.1-1ubuntu0.10).
linux-headers-generic is already the newest version (4.15.0.175.164).
The following packages were automatically installed and are no longer required:
  fltk1.3-doc fluid hddtemp libfltk-cairo1.3 libfltk-forms1.3 libfltk-gl1.3 libfltk-images1.3 libfltk1.3
  libfltk1.3-dev libfwup1 libjs-jquery-ui libllvm9 linux-hwe-5.4-headers-5.4.0-100 linux-hwe-5.4-headers-5.4.0-86
  linux-hwe-5.4-headers-5.4.0-87 linux-hwe-5.4-headers-5.4.0-89 linux-hwe-5.4-headers-5.4.0-90
  linux-hwe-5.4-headers-5.4.0-91 linux-hwe-5.4-headers-5.4.0-92 linux-hwe-5.4-headers-5.4.0-94
  linux-hwe-5.4-headers-5.4.0-96 linux-hwe-5.4-headers-5.4.0-97 linux-hwe-5.4-headers-5.4.0-99
  python-backports.functools-lru-cache python-cycler python-matplotlib python-matplotlib-data python-opengl
  python-pyqt5.qtopengl python-pyqt5.qtwebkit python-subprocess32 python-tz ros-melodic-common-msgs
  ros-melodic-gl-dependency ros-melodic-image-geometry ros-melodic-qt-gui-py-common ros-melodic-qwt-dependency
  ros-melodic-rosboost-cfg ros-melodic-roscpp-core ros-melodic-roscpp-tutorials ros-melodic-roslang
  ros-melodic-rosmake ros-melodic-self-test ros-melodic-smach ros-melodic-smach-msgs ros-melodic-stage
  ros-melodic-urdf-parser-plugin ros-melodic-vision-opencv ros-melodic-visualization-marker-tutorials
[... 370 lines cut ...]
  CC [M]  /home/nuc/library/librealsense-2.42.0/ubuntu-bionic-hwe-5.4/drivers/media/v4l2-core/v4l2-flash-led-class.mod.o
  CC [M]  /home/nuc/library/librealsense-2.42.0/ubuntu-bionic-hwe-5.4/drivers/media/v4l2-core/v4l2-fwnode.mod.o
  CC [M]  /home/nuc/library/librealsense-2.42.0/ubuntu-bionic-hwe-5.4/drivers/media/v4l2-core/v4l2-mem2mem.mod.o
  CC [M]  /home/nuc/library/librealsense-2.42.0/ubuntu-bionic-hwe-5.4/drivers/media/v4l2-core/videobuf-core.mod.o
  CC [M]  /home/nuc/library/librealsense-2.42.0/ubuntu-bionic-hwe-5.4/drivers/media/v4l2-core/videobuf-dma-sg.mod.o
  CC [M]  /home/nuc/library/librealsense-2.42.0/ubuntu-bionic-hwe-5.4/drivers/media/v4l2-core/videobuf-vmalloc.mod.o
  CC [M]  /home/nuc/library/librealsense-2.42.0/ubuntu-bionic-hwe-5.4/drivers/media/v4l2-core/videodev.mod.o
[... cut ...]
  LD [M]  /home/nuc/library/librealsense-2.42.0/ubuntu-bionic-hwe-5.4/drivers/media/v4l2-core/v4l2-flash-led-class.ko
  LD [M]  /home/nuc/library/librealsense-2.42.0/ubuntu-bionic-hwe-5.4/drivers/media/v4l2-core/videodev.ko
make: Leaving directory '/home/nuc/library/librealsense-2.42.0/ubuntu-bionic-hwe-5.4'
Patched kernels modules were created successfully

build_usbcore_modules=0
Replacing videodev  -
	Applying the patched module ...  succeeded
Replacing videobuf2-common  -
cp: cannot stat '/home/nuc/5.4.0-107-generic-videobuf2-common.ko': No such file or directory
...

snakehaihai, 2022-04-16:
after adding the new flag. The rs_multicamera.launch end with red error. 

Below is the cmake output which looks suspicious 

```
nuc build $ cmake ../ -DFORCE_RSUSB_BACKEND=true
-- The CXX compiler identification is GNU 7.5.0
-- The C compiler identification is GNU 7.5.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Checking internet connection...
-- Failed to identify Internet connection
CMake Warning at CMakeLists.txt:16 (message):
  No internet connection, disabling BUILD_WITH_TM2

CMake Warning at CMakeLists.txt:22 (message):
  No internet connection, disabling IMPORT_DEPTH_CAM_FW

-- Info: REALSENSE_VERSION_STRING=2.42.0
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Setting Unix configurations
-- using RS2_USE_LIBUVC_BACKEND
-- Found usb: /usr/lib/x86_64-linux-gnu/libusb-1.0.so  
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE  
-- Could NOT find apriltag (missing: APRILTAG_INC APRILTAG_LIB) 
-- Unable to find apriltag library, skipping pose-apriltag example
-- Configuring done
-- Generating done
-- Build files have been written to: /home/nuc/library/librealsense-2.42.0/build

cp: cannot stat '/home/nuc/5.4.0-107-generic-videobuf2-common.ko': No such file or directory
```

Below is the launch error

[image]

here is the full log

```
nuc catkin_ws_drivers $ roslaunch realsense2_
realsense2_camera       realsense2_description  
nuc catkin_ws_drivers $ roslaunch realsense2_camera rs_multiple_devices.launch 
... logging to /home/nuc/.ros/log/9db2e87a-bd96-11ec-9aed-a46bb62701f5/roslaunch-nuc-26991.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.5:40381/

SUMMARY
========
[... 240 lines cut ...]
[ INFO]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO]: setupPublishers...
[ INFO]: Expected frequency for depth = 30.00000
[ INFO]: Expected frequency for color = 30.00000
[ INFO]: setupStreams...
[ INFO]: insert Depth to Stereo Module
[ INFO]: insert Color to RGB Camera
[ INFO]: SELECTED BASE:Depth, 0
[ INFO]: RealSense Node Is Up!
[ WARN]: 
 16/04 23:44:46,322 WARNING [140557477201664] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 16/04 23:44:46,373 WARNING [140557477201664] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 16/04 23:44:46,524 WARNING [140557477201664] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 16/04 23:44:46,612 WARNING [140557477201664] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 16/04 23:44:46,662 WARNING [140557477201664] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 16/04 23:44:46,713 WARNING [140557477201664] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 16/04 23:44:47,207 ERROR [140556965508864] (ds5-options.cpp:86) Asic Temperature value is not valid!
 16/04 23:44:48,208 ERROR [140556965508864] (ds5-options.cpp:86) Asic Temperature value is not valid!
[camera1/realsense2_camera-2] process has finished cleanly
log file: /home/nuc/.ros/log/9db2e87a-bd96-11ec-9aed-a46bb62701f5/camera1-realsense2_camera-2*.log
[camera1/realsense2_camera_manager-1] process has died [pid 27013, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/nuc/.ros/log/9db2e87a-bd96-11ec-9aed-a46bb62701f5/camera1-realsense2_camera_manager-1.log].
log file: /home/nuc/.ros/log/9db2e87a-bd96-11ec-9aed-a46bb62701f5/camera1-realsense2_camera_manager-1*.log

```

MartyG-RealSense, 2022-04-16:
Is the **control_transfer returned** warning generating continuously after launch or does it repeat a small number of times and stop?

snakehaihai, 2022-04-16:
For the first instance. The camera node crashes.

I tried to launch couple more times. The node lives.  But the error control_transfer returned error repeats forever. 

The good news is the hz rate of 2 camera seems to be the same. both at 7 hz. although I command it to output at 0.5hz. not big deal if it just scaling issue. 

[image]

The bad news is the data is still out of sync

[image]

MartyG-RealSense, 2022-04-16:
The depth and color streams are meant to be publishing at 30 FPS / 30 Hz each.  A rate of 7 Hz suggests and the **No data available** messages suggest that the computer is struggling to process the frames. 

After launch, could you open the rqt_reconfigure interface with **rosrun rqt_reconfigure rqt_reconfigure** and then go to the **rgb_camera** options and untick an option called **auto_exposure_priority** please.  Doing so to disable auto_exposure_priority should force the streams to try to be maintained at a constant rate instead of being permitted to vary.

[image]

snakehaihai, 2022-04-16:
I will try it tmr. It is near 1 am in Singapore time.

I preset the Arduino to output pulse at 0.5hz. I did lower the frequency so that 4-6 cameras can be fed concurrently for street view capturing and image-based localization study. 

So you mean the camera only can be set at 30hz? It can not vary between 10-20hz based on Arduino pulse frequency? Then in this case, the PC cant handle that kind of bandwidth right?

MartyG-RealSense, 2022-04-16:
Once you have defined a particular FPS frequency for a stream and launched then you should not be able to manually alter the frequency after launch.  However, the FPS is permitted to vary if auto_exposure = true and auto_exposure_priority = true (i.e you do not disable auto_exposure_priority).

You can set the FPS at whatever rate is supported by a particular camera model for a particular resolution.  On D455 the supported FPS rates for depth and color will typically be 5, 10, 15, 30 and 60.

Generally, a RealSense hardware sync setup will not try to control the FPS rate of the cameras.  Instead it is aiming to sync the stream **timestamps** with the master trigger signal.  

Using an external signal generator such as SparkFun instead of generating the trigger pulse from a RealSense camera set to an Inter Cam Sync Mode of '1' can be extremely difficult though because the trigger signal has to be so precisely matched to the Hz rate that the slave cameras are using if the slaves are set to '2'.  Also, when '2' is used, you cannot sync RGB - only depth sync is supported.

It is for this reason that Intel released an **External Synchronization (Genlock)** system to make it much easier to sync slave cameras to an external signal generator.  The genlock system also supports an Inter Cam Sync Mode of '3' (Full Slave) in which RGB can be synced.  

https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras 

I should emphasize that the Genlock system (supporting modes 3+) is considered unvalidated and experimental by Intel, compared to the 'validated and mature' original hardware sync system (modes 1 and 2)

snakehaihai, 2022-04-17:
Hi After read thought your post. I change the ciricult so that one camera is working as master and one camera is working as slave.
Here is my setup

[image]

this is my launch file now

```

<launch>
  <arg name="serial_no_camera1"    			default="146222250676"/> 			<!-- Note: Replace with actual serial number -->
  <arg name="serial_no_camera2"    			default="043422251647"/> 			<!-- Note: Replace with actual serial number -->
  
  <arg name="camera1"              			default="camera1"/>		<!-- Note: Replace with camera name -->
  <arg name="camera2"              			default="camera2"/>		<!-- Note: Replace with camera name -->
  
  <arg name="tf_prefix_camera1"         default="$(arg camera1)"/>
  <arg name="tf_prefix_camera2"         default="$(arg camera2)"/>

  <arg name="initial_reset"             default="true"/>

  <group ns="$(arg camera1)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="serial_no"             value="$(arg serial_no_camera1)"/>
      <arg name="tf_prefix"         		value="$(arg tf_prefix_camera1)"/>
      <arg name="initial_reset"         value="$(arg initial_reset)"/>
    </include>
  </group>

  <group ns="$(arg camera2)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="serial_no"             value="$(arg serial_no_camera2)"/>
      <arg name="tf_prefix"		          value="$(arg tf_prefix_camera2)"/>
      <arg name="initial_reset"         value="$(arg initial_reset)"/>
    </include>
  </group>

  <rosparam>
    /camera1/stereo_module/intel_cam_sync_mode: 1
    /camera2/stereo_module/intel_cam_sync_mode: 2
    /camera2/rgb_camera/auto_exposure_priority: false
    /camera2/rgb_camera/auto_exposure_priority: false
    /camera1/stereo_module/emitter_enabled: 0
    /camera2/stereo_module/emitter_enabled: 0
  </rosparam>

</launch>
```

here is the result
[image]

below is the log

```

... logging to /home/nuc/.ros/log/9db2e87a-bd96-11ec-9aed-a46bb62701f5/roslaunch-nuc-29649.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
]2;/home/nuc/catkin_ws_drivers/src/realsense-ros-2.2.22/realsense2_camera/launch/rs_multiple_devices.launch
[1mstarted roslaunch server http://192.168.1.5:40653/[0m

SUMMARY
========

PARAMETERS
 * /camera1/realsense2_camera/accel_fps: 0
 * /camera1/realsense2_camera/accel_frame_id: camera1_accel_frame
 * /camera1/realsense2_camera/accel_optical_frame_id: camera1_accel_opt...
[... 262 lines cut ...]
[0m[ INFO]:  [0m
 17/04 14:37:17,929 ERROR [139788518987520] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 17/04 14:37:17,929 ERROR [139788544165632] (sensor.cpp:523) acquire_power failed: failed to set power state
 17/04 14:37:17,930 WARNING [139788544165632] (rs.cpp:306) null pointer passed for argument "device"
 17/04 14:37:17,934 WARNING [139788090861312] (messenger-libusb.cpp:92) usb_request_queue returned error, endpoint: 135 error: No such device, number: 19
 17/04 14:37:17,934 WARNING [139788090861312] (uvc-device.cpp:183) rs_uvc_device exception in listen_to_interrupts method: failed to submit interrupt request, error: RS2_USB_STATUS_OTHER
 17/04 14:37:17,934 WARNING [139788090861312] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No such device, number: 19
 17/04 14:37:17,934 WARNING [139788544165632] (rs.cpp:306) null pointer passed for argument "device"
[0m[ INFO]:  [0m
 17/04 14:37:23,952 ERROR [139788099254016] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 17/04 14:37:23,952 ERROR [139788544165632] (sensor.cpp:523) acquire_power failed: failed to set power state
 17/04 14:37:23,953 WARNING [139788544165632] (rs.cpp:306) null pointer passed for argument "device"
 17/04 14:37:23,957 WARNING [139788518987520] (messenger-libusb.cpp:92) usb_request_queue returned error, endpoint: 135 error: No such device, number: 19
 17/04 14:37:23,957 WARNING [139788518987520] (uvc-device.cpp:183) rs_uvc_device exception in listen_to_interrupts method: failed to submit interrupt request, error: RS2_USB_STATUS_OTHER
 17/04 14:37:23,957 WARNING [139788518987520] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No such device, number: 19
 17/04 14:37:23,957 WARNING [139788544165632] (rs.cpp:306) null pointer passed for argument "device"
[0m[ INFO]:  [0m
 17/04 14:37:29,971 ERROR [139788510594816] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 17/04 14:37:29,971 ERROR [139788544165632] (sensor.cpp:523) acquire_power failed: failed to set power state
 17/04 14:37:29,972 WARNING [139788544165632] (rs.cpp:306) null pointer passed for argument "device"
 17/04 14:37:29,976 WARNING [139788099254016] (messenger-libusb.cpp:92) usb_request_queue returned error, endpoint: 135 error: No such device, number: 19
 17/04 14:37:29,976 WARNING [139788099254016] (uvc-device.cpp:183) rs_uvc_device exception in listen_to_interrupts method: failed to submit interrupt request, error: RS2_USB_STATUS_OTHER
 17/04 14:37:29,976 WARNING [139788099254016] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No such device, number: 19
 17/04 14:37:29,977 WARNING [139788544165632] (rs.cpp:306) null pointer passed for argument "device"
```

MartyG-RealSense, 2022-04-17:
Instead of using multiple_devices.launch, are both cameras able to launch if you open two separate ROS terminals and perform an **rs_camera.launch** for camera 1 in terminal 1 and a launch for camera 2 in terminal 2 using the instructions below.

**Terminal 1**
roslaunch realsense2_camera rs_camera.launch initial_reset:=true serial_no:=146222250676

**Terminal 2**
roslaunch realsense2_camera rs_camera.launch initial_reset:=true serial_no:=043422251647

snakehaihai, 2022-04-17:
you command has conflict as two of the camera is using same namespace.
I created 2 launch file and launch them individually. with only modification to initial reset and serial_no

below is the output

[image]

snakehaihai, 2022-04-17:
This method did not work 

below is the output
[image]

here is the 2 launch file
````

<launch>

  <arg name="serial_no"           default="146222250676"/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="camera1"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
  <arg name="output"              default="screen"/>

  <arg name="fisheye_width"       default="-1"/>
  <arg name="fisheye_height"      default="-1"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="640"/>
  <arg name="depth_height"        default="480"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="confidence_width"    default="-1"/>
  <arg name="confidence_height"   default="-1"/>
  <arg name="enable_confidence"   default="false"/>
  <arg name="confidence_fps"      default="-1"/>

  <arg name="infra_width"         default="640"/>
  <arg name="infra_height"        default="480"/>
  <arg name="enable_infra"        default="false"/>
  <arg name="enable_infra1"       default="true"/>
  <arg name="enable_infra2"       default="true"/>
  <arg name="infra_rgb"           default="false"/>

  <arg name="color_width"         default="640"/>
  <arg name="color_height"        default="480"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="-1"/>
  <arg name="depth_fps"           default="30"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="30"/>
  <arg name="gyro_fps"            default="-1"/>
  <arg name="accel_fps"           default="-1"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>

  <arg name="enable_pointcloud"         default="true"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>
  <arg name="allow_no_texture_points"   default="false"/>
  <arg name="ordered_pc"                default="false"/>

  <arg name="enable_sync"               default="false"/>
  <arg name="align_depth"               default="false"/>

  <arg name="publish_tf"                default="true"/>
  <arg name="tf_publish_rate"           default="0"/>

  <arg name="filters"                   default=""/>
  <arg name="clip_distance"             default="-2"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="true"/>
  <arg name="unite_imu_method"          default="linear_interpolation"/>
[... 199 lines cut ...]
      <arg name="publish_tf"               value="$(arg publish_tf)"/>
      <arg name="tf_publish_rate"          value="$(arg tf_publish_rate)"/>

      <arg name="filters"                  value="$(arg filters)"/>
      <arg name="clip_distance"            value="$(arg clip_distance)"/>
      <arg name="linear_accel_cov"         value="$(arg linear_accel_cov)"/>
      <arg name="initial_reset"            value="$(arg initial_reset)"/>
      <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>
      <arg name="topic_odom_in"            value="$(arg topic_odom_in)"/>
      <arg name="calib_odom_file"          value="$(arg calib_odom_file)"/>
      <arg name="publish_odom_tf"          value="$(arg publish_odom_tf)"/>
      <arg name="allow_no_texture_points"  value="$(arg allow_no_texture_points)"/>
      <arg name="ordered_pc"               value="$(arg ordered_pc)"/>
     
<rosparam>
  /camera2/stereo_module/emitter_enabled:false
  /camera2/stereo_module/intel_cam_sync_mode: 2
</rosparam> 

    </include>
  </group>

</launch>

```

MartyG-RealSense, 2022-04-17:
You can distinguish cameras by name by adding a **camera:=** instruction, like below.

Terminal 1
roslaunch realsense2_camera rs_camera.launch initial_reset:=true camera:=camera1 serial_no:=146222250676

Terminal 2
roslaunch realsense2_camera rs_camera.launch initial_reset:=true camera:=camera2 serial_no:=043422251647

snakehaihai, 2022-04-17:
Oh  just now forgot to add auto_exposure_priority

after adding the auto_exposure_priority

the resulting frequency becomes really weird. 

[image]

snakehaihai, 2022-04-17:
> You can distinguish cameras by name by adding a **camera:=** instruction, like below.
>
> **Terminal 1** roslaunch realsense2_camera rs_camera.launch initial_reset:=true camera:=camera1 [...]

I pwr cycled PC. and run this command

here is the output

[image]

MartyG-RealSense, 2022-04-17:
And you get similar results if you only launch one camera instead of two?

snakehaihai, 2022-04-17:
Hi MartyG
One of them constantly stuck at 14.6 hz and one of them is at 30 hz. even after I disconnected sync cable

snakehaihai, 2022-04-17:
Let me reload the firmware and perform hardware set first. Liekly previous config stuck inside

snakehaihai, 2022-04-17:
after reflash the firmware and power reset. both of them can get back to 30 hz.  So what should I do now to get them sync？  still master-slave trigger with auto_exposure_priority set to false?

MartyG-RealSense, 2022-04-17:
If you are now using camera 1 as Master and camera 2 as Slave then set the Master camera as inter_cam_sync_mode '1' and the Slave camera as '2'.  Setting auto_exposure_priority to false is not a requirement for multiple camera hardware sync but is useful as an optional choice for keeping the FPS rate constant.

snakehaihai, 2022-04-17:
My student borrowed them for some presentation tmr. I`ll continue sync test after I get them back tmr.

MartyG-RealSense, 2022-04-17:
Okay, thanks very much for the updates.  I look forward to continuing this case with you.

snakehaihai, 2022-04-18:
Here is the result.

With initial reset set to true.  1 camera set to master. 1 camera set to slave and auto_exposure_priority set to false

The result still have large offsets 
[image]

here is the script 1

```

<launch>

  <arg name="serial_no"           default="146222250676"/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="camera1"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
  <arg name="output"              default="screen"/>

  <arg name="fisheye_width"       default="-1"/>
  <arg name="fisheye_height"      default="-1"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="640"/>
  <arg name="depth_height"        default="480"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="confidence_width"    default="-1"/>
  <arg name="confidence_height"   default="-1"/>
  <arg name="enable_confidence"   default="false"/>
  <arg name="confidence_fps"      default="-1"/>

  <arg name="infra_width"         default="640"/>
  <arg name="infra_height"        default="480"/>
  <arg name="enable_infra"        default="false"/>
  <arg name="enable_infra1"       default="true"/>
  <arg name="enable_infra2"       default="true"/>
  <arg name="infra_rgb"           default="false"/>

  <arg name="color_width"         default="640"/>
  <arg name="color_height"        default="480"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="-1"/>
  <arg name="depth_fps"           default="30"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="30"/>
  <arg name="gyro_fps"            default="-1"/>
  <arg name="accel_fps"           default="-1"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>

  <arg name="enable_pointcloud"         default="true"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>
  <arg name="allow_no_texture_points"   default="false"/>
  <arg name="ordered_pc"                default="false"/>

  <arg name="enable_sync"               default="false"/>
  <arg name="align_depth"               default="false"/>

  <arg name="publish_tf"                default="true"/>
  <arg name="tf_publish_rate"           default="0"/>

  <arg name="filters"                   default=""/>
  <arg name="clip_distance"             default="-2"/>
[... 206 lines cut ...]
      <arg name="publish_tf"               value="$(arg publish_tf)"/>
      <arg name="tf_publish_rate"          value="$(arg tf_publish_rate)"/>

      <arg name="filters"                  value="$(arg filters)"/>
      <arg name="clip_distance"            value="$(arg clip_distance)"/>
      <arg name="linear_accel_cov"         value="$(arg linear_accel_cov)"/>
      <arg name="initial_reset"            value="$(arg initial_reset)"/>
      <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>
      <arg name="topic_odom_in"            value="$(arg topic_odom_in)"/>
      <arg name="calib_odom_file"          value="$(arg calib_odom_file)"/>
      <arg name="publish_odom_tf"          value="$(arg publish_odom_tf)"/>
      <arg name="allow_no_texture_points"  value="$(arg allow_no_texture_points)"/>
      <arg name="ordered_pc"               value="$(arg ordered_pc)"/>

    </include>
  </group>

  <rosparam>
    /camera2/stereo_module/intel_cam_sync_mode: 2
    /camera2/rgb_camera/auto_exposure_priority: false
    /camera2/stereo_module/emitter_enabled: 0
  </rosparam>
</launch>

```

snakehaihai, 2022-04-18:
After turn off the auto exposure and set fix exposure to both camera. I notice that the time delay between 2 camera become fixed

[image]

So I think the master-slave sync wosk. But because it is trigger by external source and could have latency issue for 2nd camera

MartyG-RealSense, 2022-04-18:
If you are setting one camera to Master ('1') instead of using an external signal generator, setting the slave as '2' will only sync Depth timestamps.  If you are aiming to sync color timestamps, as suggested by your chart above, then the slave camera should be set to '3' (Full Slave mode). 

Also bear in mind that if the timestamps of slave and master are perfectly, permanently synced then that indicates that hardware sync is **not** working.  With hardware sync, if the timestamps slowly drift apart over a long period of time then this indicates that sync is working correctly.  This is discussed in the section of the original hardware sync paper linked to below, in the paragraph titled **Now to the somewhat counter intuitive aspect of time stamps**.

https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration#section-3-multi-camera-programming

snakehaihai, 2022-04-18:
Ok i get it. now it is 1 d455 set to 1 and another 1 set to 3. The result looks consistent. I`ll try other both 3 later and see the result

[image]

MartyG-RealSense, 2022-04-18:
It's excellent to hear that you made apparent progress!  I look forward to hearing about your further test results.  :)

snakehaihai, 2022-04-19:
right now im running both in mode 3 and using an arduino to trigger them. 
I set all exposure to be manual with all exposure = 100

```
  <rosparam>
    /camera1/stereo_module/intel_cam_sync_mode: 3
    /camera1/rgb_camera/auto_exposure_priority: false
    /camera1/rgb_camera/enable_auto_exposure: false
    /camera1/rgb_camera/exposure: 100
    /camera1/stereo_module/emitter_enabled: 0
    /camera1/stereo_module/enable_auto_exposure: false
    /camera1/stereo_module/exposure: 100
  </rosparam>
```
I encounter this case that initial timestamp of data could align
[image]

But at time goes, the time stamp offset goes wider. 
[image]

could it be the global time issue? I`m going to test global time variable

snakehaihai, 2022-04-19:
Does master-slave trigger really triggers the image? If so? can i read the master trigger rising edge and force the next master and slave camera to have the same trigger timestamp?

The timestamp drift is caused similar to this issue right? https://github.com/IntelRealSense/realsense-ros/issues/796

snakehaihai, 2022-04-19:
So can i confirm that its the same trigger just that the timestamp there is a bug?

[image]

[image]

MartyG-RealSense, 2022-04-19:
Mode 3 for achieving RGB sync as Full Slave has very little documentation compared to genlock modes 4+.  My understanding from the External Synchronization paper is that Mode 3 will behave the same as slave mode '2' except for the addition of RGB sync support.  This implies (not stated in the paper) that in '3' mode, the Slave will listen for a trigger signal on each frame and then after a certain time period has passed, it will give up trying to sync to the master on that particular frame and initiate unsynced capture independently.  This 'wait and then capture the frame unsynced' is how slave mode '2' works.

In mode 4+ (genlock), the slave does not trigger after a certain wait period but instead waits indefinitely for a trigger signal and then only initiates capture once the trigger is received.

It is normal for timestamps to drift apart when hardware sync is working correctly, though the rate at which yours are driting seems to be very fast, as the original multiple camera hardware sync paper states that "you might expect this drift to be on the order of less than 1ms/minute".

https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration#section-3-multi-camera-programming

Ultimately, the very limited amount of documentation for mode '3' makes it difficult to advise about this problem, unfortunately.

snakehaihai, 2022-04-20:
> Mode 3 for achieving RGB sync as Full Slave has very little documentation compared to genlock modes 4+. My understanding from the External Synchronization paper is that Mode 3 will behave the same [...]

OK I got it. The hardware dev team forgot to put a memo. Bad boys. 

I`ll try to read the sync pulse and send the pulse into ROS topic system and see how is the timing correlated to each other.  If I guess it right, both images are triggered at that pulse rising edge time, span for exposure time to form the image. So by right, I can modify the rs driver to take that pulse time as the reference and set both camera timestamps to be at the same time.

MartyG-RealSense, 2022-04-20:
Thanks very much, @snakehaihai - please do update here with the results of your next tests when you have them.

snakehaihai, 2022-04-20:
somehow i read cant find pulse after set arduino pin in input mode. 
[image]
```

#include <ros.h>
#include <std_msgs/Bool.h>

ros::NodeHandle nh;

std_msgs::Bool pushed_msg;
ros::Publisher pub_sync_pulse("/sync_pulse", &pushed_msg);

const int input_pin = 10;

bool last_reading;
long last_debounce_time=0;
long debounce_delay=1;
bool published = true;

void setup()
{
  nh.initNode();
  nh.advertise(pub_sync_pulse);
  

  pinMode(input_pin, INPUT);
  

  digitalWrite(input_pin, HIGH);
  

  last_reading = ! digitalRead(input_pin);
 
}

void loop()
{
  
  bool reading = ! digitalRead(input_pin);

  published = false;

  if ( !published ) {
    pushed_msg.data = reading;
    pub_sync_pulse.publish(&pushed_msg);
    published = true;
  }

  last_reading = reading;
  
  nh.spinOnce();
}
```
The sync data is active high all the way.  My best guess is that during digital pin input mode. the pin is actually being pull up at logic 1. 

I think I need to find an oscilloscope to better identify the issue. Some1 steal the scope from my lab

MartyG-RealSense, 2022-04-21:
Good luck with finding your oscilloscope or a replacement!

MartyG-RealSense, 2022-04-28:
Hi @snakehaihai  Do you have an update about this case that you can provide, please?  Thanks!

snakehaihai, 2022-04-28:
Hi currently i`m confusing about what happens to the sync.
i tried both arduino powered on or arduino powered off
With 1 camera set to be master sync and 1 camera set to be 3 slave
I`m not able to pick up any pulse no matter Arduino state. 

I`m not even know how the camera appears to be in sync

I`m doing more testing and will update here later

[image]

[image]

snakehaihai, 2022-04-28:
with both camera set to mode 3
now i`m able to get synced image by external arduino trigger. But the timestamp difference is still quite big. tmr i`ll try to get the sync pulse time 

[image]
[image]

MartyG-RealSense, 2022-04-28:
The purpose of using Arduino in a multi-camera setup is typically to provide a trigger pulse for the Slave camera (3).  If one of the cameras is set to '1' to designate it as Master then that camera is generating the trigger pulse instead of Arduino.  So I would not expect the Arduino being on or off to influence the outcome of the sync.

There is very little information about using Full Slave mode (3) and virtually no information about using Full Slave in ROS other than your own investigation, unfortunately.  The External Synchronization system that introduced mode 3 was also an experimental one that did not receive further development.  So you are a pioneer in this regard.

I recall that there was a past case in 2020 where depth-RGB sync in the External Synchronization system was apparently working for D435i but not for the D455.  A RealSense team member suggested that a fix for providing RGB-depth sync for D455 was being worked on.  It was not confirmed later whether this work was completed though and as mentioned, the External Synchronization system has not received further updates.

MartyG-RealSense, 2022-04-29:
It's great to hear that you made significant progress.  Thanks so much for continuing to share your methods with the RealSense community!

snakehaihai, 2022-05-09:
Hi, I`ve managed to rewrite the package and sync all time according to the trigger. thx,I `ll close this ticket

closed 2022-05-09 by snakehaihai

MartyG-RealSense, 2022-05-09:
Hi @snakehaihai  That is great news!  Thanks very much for the update. :)