realsenseai/realsense-ros #2327: How to change fps of color image and IMU of my D455?

Fllying, 2022-04-22:
I want to change fps of color image to 4hz and fps of color image to 400hz.
I use` roslaunch realsense2_camera rs_camera_d455.launch color_width:=640 color_height:=480 color_fps:=4`, but there is something error .
```
SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: `camera_aligned_de...`
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 4
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: False
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_rgb: False
 * /camera/realsense2_camera/infra_width: 848
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/ordered_pc: False
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/publish_odom_tf: True
 * /camera/realsense2_camera/publish_tf: True
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /camera/realsense2_camera/stereo_module/exposure/1: 7500
 * /camera/realsense2_camera/stereo_module/exposure/2: 1
 * /camera/realsense2_camera/stereo_module/gain/1: 16
 * /camera/realsense2_camera/stereo_module/gain/2: 16
 * /camera/realsense2_camera/tf_publish_rate: 0.0
 * /camera/realsense2_camera/topic_odom_in: odom_in
 * /camera/realsense2_camera/unite_imu_method: linear_interpolation
 * /camera/realsense2_camera/usb_port_id: 
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [19564]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 52e20c70-c211-11ec-ad38-00e04ce92461
process[rosout-1]: started with pid [19575]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [19582]
process[camera/realsense2_camera-3]: started with pid [19583]
[ INFO] [1650614021.027170217]: Initializing nodelet with 8 worker threads.
[ INFO] [1650614021.182374706]: RealSense ROS v2.3.0
[ INFO] [1650614021.182466956]: Built with LibRealSense v2.50.0
[ INFO] [1650614021.182515858]: Running with LibRealSense v2.50.0
[ INFO] [1650614021.223082577]:  
[ INFO] [1650614021.235388448]: Device with serial number 037522250554 was found.

[ INFO] [1650614021.235457471]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video0 was found.
[ INFO] [1650614021.235488737]: Device with name Intel RealSense D455 was found.
[ INFO] [1650614021.236184855]: Device with port number 2-1 was found.
[ INFO] [1650614021.236231776]: Device USB type: 3.2
[ INFO] [1650614021.239088207]: getParameters...
[ INFO] [1650614021.317359493]: setupDevice...
[ INFO] [1650614021.317400444]: JSON file is not provided
[ INFO] [1650614021.317421908]: ROS Node Namespace: camera
[ INFO] [1650614021.317444965]: Device Name: Intel RealSense D455
[ INFO] [1650614021.317464487]: Device Serial No: 037522250554
[ INFO] [1650614021.317515329]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1:1.0/video4linux/video0
[ INFO] [1650614021.317550934]: Device FW version: 05.13.00.50
[ INFO] [1650614021.317573385]: Device Product ID: 0x0B5C
[ INFO] [1650614021.317588526]: Enable PointCloud: Off
[ INFO] [1650614021.317602867]: Align Depth: Off
[ INFO] [1650614021.317617961]: Sync Mode: On
[ INFO] [1650614021.317672330]: Device Sensors: 
[ INFO] [1650614021.325425327]: Stereo Module was found.
[ INFO] [1650614021.330968313]: RGB Camera was found.
[ INFO] [1650614021.331127044]: Motion Module was found.
[ INFO] [1650614021.331168087]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1650614021.331193505]: num_filters: 0
[ INFO] [1650614021.331207256]: Setting Dynamic reconfig parameters.
[ INFO] [1650614021.475611853]: Done Setting Dynamic reconfig parameters.
[ WARN] [1650614021.475975587]: Given stream configuration is not supported by the device!  Stream: Color, Stream Index: 0, Width: 640, Height: 480, FPS: 4, Format: None
[ WARN] [1650614021.476017107]: Using default profile instead.
[ INFO] [1650614021.478131459]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1650614021.478198013]: gyro stream is enabled - fps: 200
[ INFO] [1650614021.478230571]: accel stream is enabled - fps: 63
[ INFO] [1650614021.478258815]: setupPublishers...
[ INFO] [1650614021.482987401]: Expected frequency for color = 30.00000
[ INFO] [1650614021.526968029]: Start publisher IMU
[ INFO] [1650614021.527703026]: setupStreams...
[ INFO] [1650614021.528332689]: insert Color to RGB Camera
[ INFO] [1650614021.528393140]: insert Gyro to Motion Module
[ INFO] [1650614021.528428506]: insert Accel to Motion Module
[ INFO] [1650614021.538187249]: SELECTED BASE:Depth, 0
[ INFO] [1650614021.546911964]: RealSense Node Is Up!
```

as you can see, `[ WARN] [1650614021.475975587]: Given stream configuration is not supported by the device!  Stream: Color, Stream Index: 0, Width: 640, Height: 480, FPS: 4, Format: None`, color image seems not supported 4HZ.
I also want to change fps of IMU to 400HZ, how to do it?

Fllying, 2022-04-22:
I try `rosrun topic_tools throttle messages /camera/color/image_raw 4.0 /color` to save image to a new topic, but the image is black.

MartyG-RealSense, 2022-04-22:
Hi @Fllying  The minimum FPS for depth and RGB that is supported on RealSense D455 is 5 FPS, not 4.

The maximum supported FPS for RGB on D455 is:

90 FPS at 640x360
60 FPS at 848x480
30 FPS at 1280x720 

If you wish to set the IMU to 400 Hz then I recommend adding the instructions below to your roslaunch instruction:

**enable_gyro:=true gyro_fps:=400**

closed 2022-04-23 by Fllying