realsenseai/realsense-ros #2328: why is the rs2::error??

Z-star1, 2022-04-22:


Z-star1, 2022-04-22:
[image]

MartyG-RealSense, 2022-04-22:
Hi @Z-star1  It looks as though you are using the **ORB-SLAM3** SLAM library at the link below.  Is that correct, please?

https://github.com/UZ-SLAMLab/ORB_SLAM3

The example that you are launching was designed for use with the RealSense D435i camera model and needs to have the .yaml file edited if it is not a D435i.  Are you using a D435i or a different RealSense model?

The **failed to set power state** error usually means that the camera was not able to be accessed, perhaps because it could not be detected or it was already busy when the access attempt was made (for example, the camera was already currently being used for streaming in another program).

Z-star1, 2022-04-22:
> Hi @Z-star1 It looks as though you are using the **ORB-SLAM3** SLAM library at the link below. Is that correct, please?
>
> https://github.com/UZ-SLAMLab/ORB_SLAM3
>
> The example that you are launching [...]

yes,i am runing ORBSLAM3 with D435i.but  it made a rs2::error.   and I've located the error here.    can you help me?because not only one have this problem.
[image]

MartyG-RealSense, 2022-04-22:
Does the problem still occur if you use the script below?

```
  rs2::context ctx;// Create librealsense context for managing devices
    // Declare RealSense pipeline, encapsulating the actual device and sensors
    std::vector<rs2::pipeline> pipelines;

    // Start a streaming pipe per each connected device
    for (auto&& dev : ctx.query_devices())
    {
        rs2::pipeline pipe(ctx);
        rs2::config cfg;
        //uncomment this will rise a `failed to set power state`bug
        //cfg.enable_device(dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER));
        pipe.start(cfg);
        pipelines.emplace_back(pipe);
    }
```

This script is from an Nvidia Jetson user who was experiencing the **Failed to set power state** error when building a list of camera devices.

https://forums.developer.nvidia.com/t/failed-to-set-power-state-error-for-multiple-realsense/121105

closed 2022-05-04 by MartyG-RealSense