realsenseai/realsense-ros #2329: NO compressed topic

Zhuomin-Zhou, 2022-04-24:
Hi,
When I run ros2-foxy wraps on raspberry 4, it jumped the errors which show blow. In the topic list, there are no compressed topics. 
The realsense-viewer works well, and I have tried the initial_reset:=true.

Version LibRealSense2: 2.50
system: Ubuntu 20.04
camera: D435i and D435

[realsense2_camera_node-1]  24/04 10:26:07,463 WARNING [281473013762464] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

The topic list is shown below.
Topic list:
/camera/aligned_depth_to_color/camera_info
/camera/aligned_depth_to_color/image_raw
/camera/color/camera_info
/camera/color/image_raw
/camera/color/metadata
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth/metadata
/camera/extrinsics/depth_to_color
/camera/imu
/parameter_events
/rosout
/tf_stati

Thanks in advance!

MartyG-RealSense, 2022-04-24:
Hi @Zhuomin-Zhou  As you state that you are using the **ros2-foxy** wrapper, may I confirm which wrapper version number you are using please?  The RealSense ROS2 wrapper stopped being called 'ros2-foxy' after version 3.2.0.

https://github.com/IntelRealSense/realsense-ros/releases/tag/3.2.0

The most recent 'ros2-foxy' version (3.2.0) was designed for matching with librealsense SDK version **2.45.0**, whilst you are using librealsense 2.50.0.  RealSense ROS wrapper versions should be matched as closely as possible with a specific librealsense version listed in the ROS wrapper release notes for that version.  

If you are using librealsense 2.50.0 then it should be matched with ROS2 wrapper version **3.2.3** on the ros2 branch or **4.0.4** on the ros2-beta branch.  
  
Can you confirm which RealSense ROS2 wrapper version you are using with librealsense 2.50.0 please?

It is also worth bearing in mind that the RealSense Viewer does not perform depth to color alignment in its **2D** mode, whilst you have configured the RealSense ROS wrapper to align depth and color.  Alignment is a processing-intensive function that is processed on the computer and not on the camera hardware.  

So it is possible that whilst the Pi 4 is able to run the Viewer well because it is not using alignment, the Pi may be struggling to process alignment in the ROS wrapper, resulting in the **resource temporarily unavailable, number 11** type of control_transfer returned warning.  This message can indicate that the computer is unable to keep up with processing the volume of data that is arriving on the USB port.

Zhuomin-Zhou, 2022-04-25:
Hi,

I am sorry, I was taking the ROS2 wrapper.
I have test the latest version of ROS2 and -beta Wrapper. But either didn't show the compressed topics. The error is the one I mentioned.

Thanks!
Min

Zhuomin-Zhou, 2022-04-25:
Hi @MartyG-RealSense ,

Would it be caused by the parameters setting? as the CPU of raspberry pi 4 is not enough.
Do you have any recommendations for parameters setting, to get the depth, color-comperssed images, and point cloud?

Thanks in advance!
Min

MartyG-RealSense, 2022-04-25:
Which method are you using to start the ROS2 wrapper, please - a launch file or ros2 run

There was a recent case where a RealSense ROS2 user had problems when starting with **ros2 launch realsense2_camera rs_launch.py** that did not occur if they started with **ros2 run realsense2_camera realsense2_camera_node** instead.

Zhuomin-Zhou, 2022-04-26:
Hi @MartyG-RealSense 

I tried to start with **ros2 run realsense2_camera realsense2_camera_node**, but it was not working. 
I also changed the depth_module and rgb_modle to 640×480×15， which was not working.
The compressed topics still disappeared.

Thanks!

MartyG-RealSense, 2022-04-26:
I have not seen any other reports of compressed topics being missing on ROS2.

Would your project be able to make use of uncompressed **/camera/color/image_raw** if '/camera/color/image_raw/compressed' is not published?

Zhuomin-Zhou, 2022-04-26:
> I have not seen any other reports of compressed topics being missing on ROS2.
> 
> 
> 
> Would your project be able to make use of uncompressed **/camera/color/image_raw** if '/camera/color/image_raw/compressed' is not published? 

Hi @MartyG-RealSense，

Thanks for your help!
Because the bandwidth is not enough, the uncompressed images cannot be used.
But， I have found the reason that causes the situation. As the **images_transport_plugins** were not installed, which is why there were compressed topics!

Regards,
Min

MartyG-RealSense, 2022-05-04:
Case closed due to solution achieved and no further comments received.

closed 2022-05-04 by MartyG-RealSense

pengzhenghao, 2024-05-28:
For those found this issue, install the image transport plugin with:

```
# For this at your robot
sudo apt-get install ros-foxy*-image-transport*

# Install this at your PC
sudo apt-get install ros-humble*-image-transport*
mamba install ros-humble*-image-transport*
mamba install ros-${ROS_DISTRO}-image-transport-plugins
```

shtern, 2024-06-24:
@pengzhenghao you're a lifesaver, thank you very much! worked for me (ROS Galactic, 4.0.4, 2.50)