realsenseai/realsense-ros #2330: Error when run "roslaunch realsense2_camera rs_camera.launch" with realsense D455

canglangzhige, 2022-04-26:
I have done "catkin_make" successfully. But when I run "roslaunch realsense2_camera rs_camera.launch" with realsense D455, the errors occur in the terminal as follows:
[ INFO] [1650962486.250530086]: setupPublishers...
[ INFO] [1650962486.253286841]: Expected frequency for depth = 30.00000
[ INFO] [1650962486.273225919]: Expected frequency for color = 30.00000
[ INFO] [1650962486.288825662]: setupStreams...
[ INFO] [1650962486.318041076]: SELECTED BASE:Depth, 0
[ INFO] [1650962486.324823780]: RealSense Node Is Up!
[ WARN] [1650962486.577524282]: 
[ERROR] [1650962486.865507906]: hwmon command 0x75( 0 2cf 0 4ff ) failed (response -6= Invalid parameter)

Have can I solve it?

MartyG-RealSense, 2022-04-26:
Hi @canglangzhige  In a couple of past cases where I have seen this error, it has occurred when the launch file has been edited to define RGB auto epxosure settings.  For example:

```
<rosparam>
/camera/rgb_camera/enable_auto_exposure: false
/camera/rgb_camera/enable_white_balance: false
</rosparam>
```

Have you altered your rs_camera.launch file in any way, please?

canglangzhige, 2022-04-27:
I have solved the error when I reboot my PC. But I do not know why.

 

 <rosparam&gt; /camera/rgb_camera/enable_auto_exposure: false /camera/rgb_camera/enable_white_balance: false </rosparam&gt;

MartyG-RealSense, 2022-04-27:
There are some problems that are not corrected by a camera reset or an unplug-replug of the USB cable and are only cleared by a complete shutdown to off state and then 'cold reboot' of the computer with the power button.

MartyG-RealSense, 2022-05-04:
Hi @canglangzhige  Do you require further assistance with this case, please?  Thanks!

nevermore0127, 2022-05-10:
Still have this problem on ubuntu 20.04 + d435i , and i just tried all the reply you have done in realsense and realsense-ros's issues about 0x75 . but not work for me.  the cam works well for somedays ，then the problem ocurs ，i think i did nothing changed. so what can i do now . or what should i provide to you ? @MartyG-RealSense

MartyG-RealSense, 2022-05-10:
Which method did you use to build the librealsense SDK. please?  From Debian packages or from source code?

Which librealsense and RealSense ROS wrapper versions are you using?  When building the wrapper from source code it is important to match a particular wrapper version as closely as possible with a specific librealsense version listed in the release notes for that wrapper version.

Would it be possible to paste your full launch log from the ROS terminal in a comment below?

MartyG-RealSense, 2022-05-16:
Hi @nevermore0127   Do you require further assistance with your problem, please?  Thanks!

nevermore0127, 2022-05-16:
i've changed a new  d435i camera . and it works. thanks！

MartyG-RealSense, 2022-05-16:
That's great to hear, @nevermore0127 - thanks very much for the update!

As you have solved your problem with a camera change and the creator of this case @canglangzhige solved their problem with a reboot, I will close this case.  Thanks again!

closed 2022-05-16 by MartyG-RealSense