realsenseai/realsense-ros #2331: [ERROR] Regions-of-interest is not implemented for this device

imrojo, 2022-04-26:
I am using a IntelRealSense D435 with Noetic and I get this error: [image]
How can I fix it?

MartyG-RealSense, 2022-04-26:
Hi @imrojo  Could you provide the following information, please:

- librealsense SDK version
- RealSense ROS wrapper version
- camera firmware version

All of the information should be in your roslaunch log in the ROS terminal.

imrojo, 2022-04-26:
[image]

MartyG-RealSense, 2022-04-26:
Your D435's firmware version 5.9.2.0 dates back to February 2018, not long after the camera was first launched in late 2017.  There have been many firmware updates since then.

As you are using librealsense 2.50.0 your firmware version should be a minimum of **5.12.8.200**, with the recommended firmware for use with 2.50.0 being **5.13.0.50**.  

This is because from firmware version 5.12.8.200 there were internal changes in the firmware that required use of librealsense version 2.39.0 or newer.

If you have access to the RealSense Viewer tool then you can update your camera's firmware to 5.13.0.50 by following the steps below.

1.  Download the firmware file named **Signed_Image_UVC_5_13_0_50.bin** from the RealSense firmware page at the link below. 

https://dev.intelrealsense.com/docs/firmware-releases#section-d400-series-firmware-downloads

[image]

2.  Launch the RealSense Viewer and left-click on the **More** option near the top of the Viewer's options side-panel and select the **Update Firmware** menu option.

[image]

3.  A file selector box will pop up.  Nevaigate to the location on your computer that you downloaded **Signed_Image_UVC_5_13_0_50.bin** to and select the file.  The updating of your camera firmwaere driver will now automatically begin.

***********

Alternatively, if you do not have access to the RealSense Viewer tool then the firmware can be updated with the **rs-firmware-update** tool.

https://github.com/IntelRealSense/librealsense/tree/master/tools/fw-update

imrojo, 2022-05-03:
I've managed to fix that but now I get the following error: 

```
[ INFO] [1651563724.668349901]: setupStreams...
 03/05 09:42:05,143 WARNING [139770634163968] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1651563725.194775688]: SELECTED BASE:Depth, 0
 03/05 09:42:05,196 WARNING [139770634163968] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1651563725.268872887]: RealSense Node Is Up!
 03/05 09:42:05,468 WARNING [139770634163968] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 03/05 09:42:05,524 WARNING [139770634163968] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
[ WARN] [1651563725.915891703]: 
 03/05 09:42:05,921 WARNING [139770634163968] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/05 09:42:06,069 WARNING [139770634163968] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/05 09:42:06,230 WARNING [139770634163968] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/05 09:42:06,551 ERROR [139770512815872] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 82
 03/05 09:42:06,580 ERROR [139770545415936] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 84
 03/05 09:42:06,581 WARNING [139770634163968] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 03/05 09:42:06,635 WARNING [139770634163968] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/05 09:42:07,008 WARNING [139770634163968] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/05 09:42:07,060 WARNING [139770634163968] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

```

Does anyone know how to fix this??

MartyG-RealSense, 2022-05-03:
The **Resource temporarily unavailable** error indicates that your computer / computing device may be struggling to keep up with processing the amount of camera data coming in from the USB port.

Which computer / computing device are you using, please (Raspberry Pi, Nvidia Jetson, PC laptop / desktop, etc).

Also, what roslaunch instruction are you using?

closed 2022-05-16 by MartyG-RealSense