realsenseai/realsense-ros #2334: Poor performances on embedded devices with align_depth and pointcloud

MatCapLynx, 2022-04-27:
I'm using a D435i camera on a Jetson Xavier NX computer and I have noticed low fps on a few topics that I need to use.
I've checked in realsense-viewer and everything is working properly. When I launch `roslaunch realsense2_camera rs_camera.launch`, I get high fps (30) on both `/camera/depth/image_rect_raw` and `/camera/color/image_raw`.

The problem is when I launch `roslaunch realsense2_camera rs_camera.launch align_depth:=true` then on the topic `/camera/aligned_depth_to_color/image_raw` I only get 2-5 fps which is not enough to run my application. A similar problem occurs when I launch `roslaunch realsense2_camera rs_camera.launch filters:=pointclound`. On `/camera/depth/color/points` I get no message or juste a few frames at startup. I also get this problem with IMU topics.

I've checked on my computer with the same camera and cable and everything works as expected with ~30 fps on `/camera/aligned_depth_to_color/image_raw`.

I originally thought this was a compatibility issue between Jetpack and ROS wrapper/librealsense version so I've tried a few differents version without success. Here is the list:
* On Jetson Xavier NX Jetpack 4.6 (Ubuntu 18.04)
  * Debian ROS package: ros-melodic-realsense2-camera: ROS Wrapper 2.3.2 | LibRealSense v2.50.0
  * ROS wrapper built from source, librealsense installed from debian packages: ROS Wrapper 2.3.2 | LibRealSense v2.50.0
  * ROS wrapper built from source, librealsense built from source with both RSUSB and native backend and with CUDA support: ROS Wrapper 2.3.2 | LibRealSense v2.50.0
  * ROS wrapper built from source, librealsense built from source with RSUSB backend and CUDA support: ROS Wrapper 2.2.23 | LibRealSense v2.43.0 based on this: https://github.com/IntelRealSense/realsense-ros/issues/1967#issuecomment-954814317
* On Jetson Xavier NX Jetpack 5.0 (Ubuntu 20.04)
  * Debian ROS package: ros-noetic-realsense2-camera: ROS Wrapper 2.3.2 | LibRealSense v2.50.0
  On Ubuntu 20/Noetic I only noticed that IMU topics were now publishing normally.
* On RaspberryPi 4B (Ubuntu server 20.04)
  * Debian ROS package: ros-noetic-realsense2-camera: ROS Wrapper 2.3.2 | LibRealSense v2.50.0

I don't think it is a performance issue as Xavier NX is a very capable device and it doesn't seem to be maxed out during use. I have no  clue about the problem origin so any help would be appreciated. Thanks !

MatCapLynx, 2022-04-27:
Hi @MartyG-RealSense, thanks for the quick answer.

Is there some known performances upgrades with ROS2 package on Jetson boards ? I might be able to test this latter.

Thanks for the advice for arm-based devices. As "Method 1" was partially working I thought it was still the prefered installation method. I will now only build packages.

MatCapLynx, 2022-04-27:
I've retried SDK 2.43.0 build and everything works, thanks !

gonultasbu, 2023-04-10:
Hello @MartyG-RealSense, in the issue you mentioned https://github.com/IntelRealSense/realsense-ros/issues/1967 that older versions should not be installed anymore and rgbd-launch should be used. Is that recommendation applicable for this problem as well?

MartyG-RealSense, 2023-04-10:
hi @gonultasbu  The rs_rgbd.launch file sets align_depth to true, so if you already have slow performance when align_depth is true then an rs_rgbd launch may not be helpful.  

However, it is certainly worth trying rs_rgbd if you are using ROS1 in order to see whether this launch file improves performance, as all you need to do is first install RGBD support for the ROS1 version you are using (Kinetic, Melodic or Noetic) and then initiate rs_rgbd.launch - see https://github.com/IntelRealSense/librealsense/issues/10223#issuecomment-1152864261

gonultasbu, 2023-04-10:
Thanks for the quick answer. I tried the referred launchfile, it did not improve the slow performance. 

Another issue I have with the downgrade is that after running rs_camera.launch, the file outputs to terminal:
```
Built with LibRealSense v2.50.0
Running with LibRealSense v2.50.0
```
which should not be the case, because I downgraded the local librealsense2 to 2.41.0 via apt-get and verified that via realsense-viewer and compiled realsense-ros 2.2.21. I am wondering why the launchfile still finds 2.50.0 somehow, would you be able to suggest any solution to that? Could it be the camera firmware?

MartyG-RealSense, 2023-04-11:
It's excellent news that you achieved a solution, @gonultasbu - thanks very much for the update!

gonultasbu, 2023-09-10:
Hello @MartyG-RealSense, due to other dependency requirements, it looks like our research group is about to merge everything to ROS2 and realsense is a pretty fundamental part of our research. Do you know whether this low FPS issue on Jetson boards is still observable with the latest librealsense and SDK? Thanks.

MartyG-RealSense, 2023-09-10:
Hi @gonultasbu  It does still happen occasionally in ROS2 but is not reported very often.

gonultasbu, 2023-09-12:
I have just tested and confirm that this issue is fixed on ros 2 foxy with librealsense 2.53 for future reference. Thanks!

MartyG-RealSense, 2023-09-13:
You are very welcome, @gonultasbu - thanks very much for the confirmation!