realsenseai/realsense-ros #2335: Configure parameters

psanzgalar, 2022-04-27:
Hi, i have de next problem:
From the launch command bar i launch **ros2 run realsense2_camera realsense2_camera_node --ros-args -p initial_reset:=true -p enable_accel:=true -p enable_gyro:=true -p gyro_fps:=200.0 -p accel_fps:=63.0 -p enable_pointcloud:=true** and it works perfectly but when I try to run it from the roslaunch file I can not configure the parameters well, any idea?

MartyG-RealSense, 2022-04-27:
Hi @psanzgalar  There was a recent case at https://github.com/IntelRealSense/realsense-ros/issues/2305#issuecomment-1095102857 where a RealSense ROS2 user could launch correctly with **ros2 run** but not with a roslaunch file (they had missing infrared topics when using roslaunch).  Although ros2 run worked perfectly, it had the disadvantage of the launch instruction being very lengthy due to not being able to configure settings within the launch file.

**********

If you are using the **ros2-beta** branch (the 4.x series of RealSense ROS2 wrappers) then the ros2-beta wrapper documentation explains that there are some differences in the parameters.  

https://github.com/IntelRealSense/realsense-ros/tree/ros2-beta#please-notice-if-you-are-moving-from-realsense-ros2-branch-to-ros2-beta

For example, instead of using **enable_pointcloud:=true**, the new format of this instruction for the 4.x wrappers is **pointcloud.enable:=true**

For example:

**ros2 launch realsense2_camera rs_launch.py depth_module.profile:=1280x720x30 pointcloud.enable:=true**

psanzgalar, 2022-04-27:
Hello, thanks for the response so fast:
I have made changes to the file launch.py (we have changed the default values) however these changes have not been applied

psanzgalar, 2022-04-27:
I'm trying to activate the point cloud (exit off) the infrared stream (exit off) and change the values of the fps

MartyG-RealSense, 2022-04-27:
My interpretation of the updated instructions in the ros2-beta wrapper is that instead of setting resolution and FPS as three different parameters (stream height, width, FPS), it is configured with a **profile**.  For example, if **depth_module.profile:=640x480x30** is used then both the depth and infrared streams are set to a resolution of 640x480 and an FPS of 30.  You then likely have the option of disabling individual stream types if you do not want them.

For example, if you wanted to disable infrared and enable the pointcloud, the instruction in the 4.x wrapper might be something like this:

**ros2 launch realsense2_camera rs_launch.py depth_module.profile:=640x480x30 enable_infra1:=false enable_infra2:=false pointcloud.enable:=true**

psanzgalar, 2022-04-28:
Good morning, try yesterday with interpreting the instruction, however still does not make changes in the configuration any other idea?

MartyG-RealSense, 2022-04-28:
Can you confirm for me please which ROS2 wrapper version you are using - 3.2.3 or 4.0.4

psanzgalar, 2022-04-28:
We have already solved it, thank you very much for your ideas

closed 2022-04-28 by psanzgalar

MartyG-RealSense, 2022-04-28:
You are very welcome.  It's excellent to hear that you achieved a solution, @psanzgalar  - thanks very much for the update!