realsenseai/realsense-ros #2339: How to have /camera/rgb_camera/parameter_descriptions topic?

miku54, 2022-05-05:
I use a D455 camera, which is connected to my device via USB3.0,my system is foxy,and I am using the ROS2-beta branch.
ros2 launch realsense2_camera rs_launch.py

Tried to start like this, there is no /camera/rgb_camera/parameter_descriptions topic
：ros2 run realsense2_camera realsense2_camera_node

miku54, 2022-05-05:
When I start the camera with this command：
ros2 launch realsense2_camera rs_launch.py​​ align_depth:=true pointcloud.enable:=true
Still no topic of pointcloud2；
https://github.com/IntelRealSense/realsense-ros/issues/2295
This thread has a similar situation to mine, but I can't use this workaround to solve my problem

miku54, 2022-05-05:
[image]

MartyG-RealSense, 2022-05-06:
Hi @miku54  The **parameter_descriptions** topic was listed as supported in the previous ROS2 wrapper (up to version 3.2.3) but is not included in the topics list of the ros2_beta (4.x versions) wrapper documentation.  Instead it adds a topic called **parameter_events**

https://github.com/IntelRealSense/realsense-ros/tree/ros2-beta#published-topics

Are you able to achieve a point cloud with the launch instruction below?

`ros2 launch realsense2_camera rs_launch.py depth_module.profile:=1280x720x30 pointcloud.enable:=true`

miku54, 2022-05-06:
Hi, @MartyG-RealSense <https://github.com/MartyG-RealSense> thanks for your
reply!
I used to download the realsense package of ros2 using binary instructions,
and now I use ros2_beta to start the camera;
ros2 launch realsense2_camera rs_launch.py
depth_module.profile:=1280x720x30 pointcloud.enable:=true
It is not possible to implement point clouds, but I found that there is a
launch (
https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/demo_pointcloud.launch)
file in the ROS2 branch that can open point clouds and get topics data, but
it only runs for a few seconds after opening the file, and then the image
freezes (it looks more like the node died) but the terminal doesn't report
any errors, which is too weird

miku54, 2022-05-06:
Hi, @MartyG-RealSense thanks for your reply!
I used to download the realsense package of ros2 using binary instructions, and now I use ros2_beta to start the camera;
ros2 launch realsense2_camera rs_launch.py depth_module.profile:=1280x720x30 pointcloud.enable:=true
It is not possible to implement point clouds, but I found that there is a launch (https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/demo_pointcloud.launch) file in the ROS2 branch that can open point clouds and get topics data, but it only runs for a few seconds after opening the file, and then the image freezes (it looks more like the node died) but the terminal doesn't report any errors, which is too weird

miku54, 2022-05-06:
[image]

MartyG-RealSense, 2022-05-06:
It looks as though your demo_pointcloud link goes to the ROS1 version of the launch file.  The **ros2** branch's version is here:

https://github.com/IntelRealSense/realsense-ros/blob/ros2/realsense2_camera/launch/demo_pointcloud_launch.py

miku54, 2022-05-07:
Sorry for the wrong link, I am running the official launch file.[https://github.com/IntelRealSense/realsense-ros/blob/ros2/realsense2_camera/launch/demo_pointcloud_launch.py](url)

MartyG-RealSense, 2022-05-07:
A couple of users of the ros2_beta wrapper recently found that using a launch file did not work correctly for them but **ros2 run** did.  For example to enable the pointcloud filter by that method:

`ros2 run realsense2_camera realsense2_camera_node --ros-args -p pointcloud.enable:=true`

miku54, 2022-05-07:
Thanks, this instruction can implement point cloud. I tried to use depth_image_proc to complete the conversion, but the effect is not very good, can realsense2_camera_node produce XYZRGB point cloud? like this:
[image]

MartyG-RealSense, 2022-05-07:
In the ROS1 wrapper you could get an organized point cloud with **depth_image_proc** by using the 'rs_rgbd.launch'.  In ros2-beta you can generate an organized point cloud with the parameter **pointcloud.ordered_pc**.  By default the pointcloud is unorganized.

I would therefore speculate that the ros2 run instruction would look like the one below.

`ros2 run realsense2_camera realsense2_camera_node --ros-args -p pointcloud.enable:=true -p pointcloud.ordered_pc:=true`

Does that launch instruction provide you with access to depth_image_proc please?

MartyG-RealSense, 2022-05-19:
Case closed due to no further comments received.

manhha1402, 2022-08-08:
Hi @MartyG-RealSense,
I launch the d405 camera with the command: `ros2 launch realsense2_camera rs_launch.py  pointcloud.enable:=true pointcloud.ordered_pc:=true`, but the still get unorganized pointcloud
[image]

MartyG-RealSense, 2022-08-08:
Hi @manhha1402   I wonder whether this may be because the D405 camera model does not have a separate RGB sensor like the other 400 Series models but instead obtains RGB from its depth sensor.

Is the cloud still unordered if you add **align_depth.enable:=true** to your launch instruction, please?

manhha1402, 2022-08-08:
Hi @MartyG-RealSense, the result is still the same. I had a look on the code `https://github.com/IntelRealSense/realsense-ros/blob/ros2-beta/realsense2_camera/src/base_realsense_node.cpp`.There is no keyword `ordered_pc`. I think it's not `ordered_pc` integrated into `ros-beta` branch.

MartyG-RealSense, 2022-08-08:
I will refer the issue to my Intel RealSense colleagues on the ROS wrapper development team.

MartyG-RealSense, 2022-08-21:
Hi @manhha1402  As the issue that you reported will be investigated by my Intel RealSense colleagues, I have added an **Enhancement** label to this case to signify that it should be kept open for the duration of the investigation.  Thanks again for your report!

SamerKhshiboun, 2022-09-20:
Hi @manhha1402, 

As I saw, there is something wrong in the calculation of ordered point cloud, and it looks it is not related to the camera model.
I'm investigating the issue and will update you asap.

Thanks,
Samer

manhha1402, 2022-09-21:
Hi @SamerKhshiboun
Thank you for fixing the bug!

MartyG-RealSense, 2022-09-21:
Yes, thanks so much, @SamerKhshiboun   👍

MartyG-RealSense, 2022-09-27:
Hi @miku54 and @manhha1402  Do you require further assistance with this case now that a fix has been implemented, please?  Thanks!

MartyG-RealSense, 2022-10-03:
Case closed due to fix implemented and no further comments received.

closed 2022-10-03 by MartyG-RealSense