realsenseai/realsense-ros #2341: D455 frames not published

MrOCW, 2022-05-06:
Hi, initially installed librealsense via debian but built librealsense the manual way (https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) as I encountered 
```
_[ WARN] [1651861782.645117261]: Param '/camera/rgb_camera/power_line_frequency' has value 3 that is not in the enum { {50Hz: 1} {60Hz: 2} {Disabled: 0} }. Removing this parameter from dynamic reconfigure options._
[ INFO] [1651861782.650532277]: Done Setting Dynamic reconfig parameters.
[ INFO] [1651861782.650915955]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1651861782.651775001]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1651861782.652533017]: setupPublishers...
[ INFO] [1651861782.654235890]: Expected frequency for depth = 30.00000
[ INFO] [1651861782.676849288]: Expected frequency for color = 30.00000
[ INFO] [1651861782.693968442]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1651861782.711620033]: setupStreams...
[ INFO] [1651861782.721937388]: rtabmap 0.20.18 started...
[ INFO] [1651861782.918697137]: SELECTED BASE:Depth, 0
[ INFO] [1651861782.925413048]: RealSense Node Is Up!
_[ WARN] [1651861783.650855109]: Hardware Notification:Depth stream start failure,1.65186e+12,Error,Hardware Error
(backend-v4l2.cpp:1127) Frames didn't arrived within 5 seconds_
```
but failed at ./scripts/patch-realsense-ubuntu-lts.sh
```
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
Unsupported kernel version 5.13.0-40-generic . The Focal patches are maintained for Ubuntu LTS with kernel 5.4, 5.8, 5.11 only
```
carried on anyway, built from source and realsense-viewer works but i am still getting the same errors.
Also, the IMU doesnt seem to be publishing any information as well.

initial run of realsense-viewer works but when i disable and enable the RGB or stereo module, I get
[image]

MartyG-RealSense, 2022-05-06:
Hi @MrOCW  Installing librealsense from both Debian packages and source code on the same computer should be avoided.  An example of the conflicts that can arise is the warning **Multiple realsense udev-rules were found!** that is caused by having two sets of udev device handling rules in two different folder locations.

In regard to the kernel version, although librealsense can work with unsupported kernels it increases the risk of unforseen consequences in regard to stability.  If you are unable to change your kernel version then building librealsense from source code with the **RSUSB backend** installation method may help.  An RSUSB build of librealsense bypasses the kernel.  It is therefore not dependent on Linux versions of kernel versions and does not require patching.

Another point to bear in mind if you have installed librealsense more than once is that each time you update librealsense, the RealSense ROS wrapper has to be built again afterwards.

MrOCW, 2022-05-07:
@MartyG-RealSense Yup I got that so i removed the udev rule from the debian package and removed librealsense before building from source again. Not sure if I missed out on removing anything else.

Will try with the RSUSB backend and get back with the results! I've built the ROS wrapper from source

MartyG-RealSense, 2022-05-07:
Thanks very much @MrOCW - I look forward to your next results.  Good luck!

MrOCW, 2022-05-07:
@MartyG-RealSense 
I have built librealsense with 
cmake -DFORCE_RSUSB_BACKEND -DCMAKE_BUILD_TYPE=Release ..
sudo make uninstall && make clean && make && sudo make install
When i run realsense-viewer, I get
```
 08/05 00:15:52,385 WARNING [139944798451456] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 08/05 00:15:54,953 ERROR [139944437729024] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 08/05 00:15:54,953 ERROR [139944897644288] (sensor.cpp:572) acquire_power failed: failed to set power state
 08/05 00:15:54,953 WARNING [139944897644288] (sensor.cpp:1173) Failed to add Backlight Compensation control for RGB Camera
 08/05 00:15:54,953 ERROR [139944437729024] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 08/05 00:15:54,953 ERROR [139944897644288] (sensor.cpp:572) acquire_power failed: failed to set power state
 08/05 00:15:54,954 WARNING [139944897644288] (sensor.cpp:1173) Failed to add Brightness control for RGB Camera
 08/05 00:15:54,954 ERROR [139944437729024] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 08/05 00:15:54,954 ERROR [139944897644288] (sensor.cpp:572) acquire_power failed: failed to set power state
 08/05 00:15:54,954 WARNING [139944897644288] (sensor.cpp:1173) Failed to add Contrast control for RGB Camera
 08/05 00:15:54,954 ERROR [139944437729024] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 08/05 00:15:54,954 ERROR [139944897644288] (sensor.cpp:572) acquire_power failed: failed to set power state
 08/05 00:15:54,954 WARNING [139944897644288] (sensor.cpp:1173) Failed to add Exposure control for RGB Camera
 08/05 00:15:54,954 ERROR [139944437729024] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 08/05 00:15:54,954 ERROR [139944897644288] (sensor.cpp:572) acquire_power failed: failed to set power state
 08/05 00:15:54,954 WARNING [139944897644288] (sensor.cpp:1173) Failed to add Gamma control for RGB Camera
 08/05 00:15:54,954 ERROR [139944437729024] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 08/05 00:15:54,954 ERROR [139944897644288] (sensor.cpp:572) acquire_power failed: failed to set power state
 08/05 00:15:54,954 WARNING [139944897644288] (sensor.cpp:1173) Failed to add Hue control for RGB Camera
 08/05 00:15:54,954 ERROR [139944437729024] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 08/05 00:15:54,954 ERROR [139944897644288] (sensor.cpp:572) acquire_power failed: failed to set power state
 08/05 00:15:54,954 WARNING [139944897644288] (sensor.cpp:1173) Failed to add Saturation control for RGB Camera
 08/05 00:15:54,954 ERROR [139944437729024] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 08/05 00:15:54,954 ERROR [139944897644288] (sensor.cpp:572) acquire_power failed: failed to set power state
 08/05 00:15:54,954 WARNING [139944897644288] (sensor.cpp:1173) Failed to add Sharpness control for RGB Camera
 08/05 00:15:54,954 ERROR [139944437729024] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 08/05 00:15:54,954 ERROR [139944897644288] (sensor.cpp:572) acquire_power failed: failed to set power state
 08/05 00:15:54,954 WARNING [139944897644288] (sensor.cpp:1173) Failed to add White Balance control for RGB Camera
 08/05 00:15:54,954 ERROR [139944437729024] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 08/05 00:15:54,954 ERROR [139944897644288] (sensor.cpp:572) acquire_power failed: failed to set power state
 08/05 00:15:54,954 WARNING [139944897644288] (sensor.cpp:1173) Failed to add Enable Auto Exposure control for RGB Camera
 08/05 00:15:54,954 ERROR [139944437729024] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 08/05 00:15:54,954 ERROR [139944897644288] (sensor.cpp:572) acquire_power failed: failed to set power state
 08/05 00:15:54,954 WARNING [139944897644288] (sensor.cpp:1173) Failed to add Enable Auto White Balance control for RGB Camera
```
the realsense-viewer hangs

MrOCW, 2022-05-07:
ok sometimes it works after retrying a few times.. will make do with this for now. however, another problem encountered was that after running roslaunch realsense2_camera opensource_tracking.launch, rostopic hz imu/data or rostopic hz camera/imu both shows "no new messages"

Running roslaunch realsense2_camera rs_camera.launch enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation gets me:

```
 08/05 01:07:36,424 WARNING [140570741667584] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[ WARN] [1651943256.490673327]: 
[ WARN] [1651943256.490729587]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
 08/05 01:07:36,562 WARNING [140570513106688] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11..... (repeats many times)
[ WARN] [1651943257.140252255]: Hardware Notification:Motion Module failure,1.65194e+12,Error,Hardware Error
 08/05 01:07:37,137 WARNING [140570513106688] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11..... (repeats many times)
```
and rostopic echo /camera/imu works

closed 2022-05-07 by MrOCW

MartyG-RealSense, 2022-05-07:
Thanks very much. @MrOCW 

When you build from source with CMake, you should include the build flags below to build the graphical examples and tools, including realsense-viewer and support in the SDK for those graphical examples / tools.

**-DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**

For example:

**cmake ../  -DFORCE_RSUSB_BACKEND -DCMAKE_BUILD_TYPE=Release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=false**