realsenseai/realsense-ros #2343: D455 and SLAM

MrMoe830, 2022-05-07:
Hi dude, I am using the "D435i and SLAM" function, I installed the dependency package according to the official instructions in [wIki](https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i), 
and successfully run it through the command `roslaunch realsense2_camera opensource_tracking.launch` , I recorded my data in the `my_bagfile_1.bag` through this line of command `rosbag record -O my_bagfile_1.bag /camera/aligned_depth_to_color/camera_info  camera/aligned_depth_to_color/image_raw /camera/color/camera_info /camera/color/image_raw /camera/imu /camera/imu_info /tf_static`, the next step is to get the inside of the exported rosbag data package. point cloud information and perform obstacle detection, but I found that in the exported rosbag data package, there is no topic about point cloud, which makes me a little confused
[image]

MrMoe830, 2022-05-07:
I seem to have solved the problem

MartyG-RealSense, 2022-05-07:
It's excellent to hear that you found a solution, @MrMoe830 - thanks very much for the update!

MrMoe830, 2022-05-07:
> It's excellent to hear that you found a solution, @MrMoe830 - thanks very much for the update!

Thanks very much

MartyG-RealSense, 2022-05-14:
Case closed due to solution achieved and no further comments received.

closed 2022-05-14 by MartyG-RealSense

MrMoe830, 2022-05-14:
这是来自QQ邮箱的假期自动回复邮件。你好，我最近正在休假中，无法亲自回复你的邮件。我将在假期结束后，尽快给你回复。