realsenseai/realsense-ros #2344: Post-processing filter params edited in ROS

w407022008, 2022-05-08:
Hello, 
Could we edit[ those post-processing filter params](https://dev.intelrealsense.com/docs/post-processing-filters) such as[ the scale factor of the decimation filter ](https://github.com/IntelRealSense/librealsense/blob/5ff27fca5aaeec4736d6bb3bfb958fee82ee198b/src/proc/decimation-filter.cpp#L205)in ROS, for example in rqt_reconfigure?
The sourse code ([src/proc/decimation-filter.cpp](https://github.com/IntelRealSense/librealsense/blob/5ff27fca5aaeec4736d6bb3bfb958fee82ee198b/src/proc/decimation-filter.cpp#L205)) that it always seems to be the default value of 2 and is not controlled. But it''s just my guess.

MartyG-RealSense, 2022-05-08:
Hello, are you using the RealSense ROS1 wrapper for Kinetic, Melodic and Noetic?  If so then you can define the Decimation filter and its **magnitude** value in a couple of different ways.

-. Visit https://github.com/IntelRealSense/realsense-ros/issues/1964#issuecomment-874286605 and scroll down to the section headed **Decimation filter performace test** for an example of launch file code for setting up the filter and magnitude.  I have quoted the code below for your reading convenience.

	```
<arg name="filters"                   default="pointcloud,decimation"/>
	<arg name="decimation_filter_magnitude"  default="2"/>
 
  <group ns="$(arg camera)">
    <param name="decimation/filter_magnitude" value="$(arg decimation_filter_magnitude)"/>

    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
   
    </include>
  </group>
```

-. Alternatively, define the Decimation filter in a roslaunch instruction and then set its magnitude in a **rosrun** instruction during runtime after the launch has completed, as described at https://github.com/IntelRealSense/realsense-ros/issues/583#issue-401683755

w407022008, 2022-05-09:
OK, got it! Let me have a try.

closed 2022-05-09 by w407022008