realsenseai/realsense-ros #2345: When i try to launch realsense depth camera D435i........i'm getting this issue

LOKAKSHAI, 2022-05-09:
[ WARN] [1652095474.637614892]: Given stream configuration is not supported by the device!  Stream: Color, Stream Index: 0, Width: -1, Height: -1, FPS: 15, Format: None
[ WARN] [1652095474.637644114]: Using default profile instead.

MartyG-RealSense, 2022-05-09:
Hi @LOKAKSHAI  It looks as though you have tried to set a custom stream configuration for FPS only.  If you are using the RealSense ROS1 wrapper or the 3.x versions of the ROS2 wrapper then a custom configuration needs to provide three pieces of information for each stream type being defined - width, height and FPS.  If this is not done then the launch decides that the custom configuration is invalid and uses the stream defaults of the particular RealSense camera model that is being used.

For example:

**ROS1**
roslaunch realsense2_camera rs_camera.launch color_width:=640 color_height:=480 color_fps:=15

**ROS2**
ros2 launch realsense2_camera rs_launch.py color_width:=640 color_height:=480 color_fps:=15.0

LOKAKSHAI, 2022-05-10:
> Hi @LOKAKSHAI It looks as though you have tried to set a custom stream configuration for FPS only. If you are using the RealSense ROS1 wrapper or the 3.x versions of the ROS2 wrapper then a custom configuration needs to provide three pieces of information for each stream type being defined - width, height and FPS. If this is not done then the launch decides that the custom configuration is invalid and instead uses the stream defaults of the particular RealSense camera model that is being used.
> 
> For example:
> 
> **ROS1** roslaunch realsense2_camera rs_camera.launch color_width:=640 color_height:=480 color_fps:=15
> 
> **ROS2 (3.x wrapper versions)** ros2 launch realsense2_camera rs_launch.py color_width:=640 color_height:=480 color_fps:=15.0

Yes got it..will check

closed 2022-05-22 by MartyG-RealSense