realsenseai/realsense-ros #2347: D435I no show image on ros2 galactic

sunil-gitcode, 2022-05-10:
the warning log:
Could not set param: pointcloud.stream_index_filter with -1 Range: [-1, -2147483648]: parameter 'pointcloud.stream_index_filter' could not be set: Parameter {} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1652153992.874715809] [camera.camera]: Start Sensor: Stereo Module

use the cmd = ros2 launch realsense2_camera rs_launch.py align_depth:=true filters:=pointcloud

MartyG-RealSense, 2022-05-10:
Hi @sunil-gitcode  Are you using the ros2_beta wrapper, such as version 4.0.4 please?  If you are then some of the parameters have changed.  To enable the pointcloud in that wrapper, **pointcloud.enable:=true** is used instead of filters:=pointcloud and **align_depth.enable:=true** instead of align_depth:=true.  For example:

`ros2 launch realsense2_camera rs_launch.py align_depth.enable:=true pointcloud.enable:=true`

sunil-gitcode, 2022-05-10:
> Hi @sunil-gitcode Are you using the ros2_beta wrapper, such as version 4.0.4 please? If you are then some of the parameters have changed from those in the previous version 3 ROS2 wrappers. To enable the pointcloud in the version 4 wrappers, **pointcloud.enable:=true** is used instead of filters:=pointcloud and **align_depth.enable:=true** instead of align_depth:=true. For example:
> 
> `ros2 launch realsense2_camera rs_launch.py align_depth.enable:=true pointcloud.enable:=true`

log is:
[realsense2_camera_node-1] [INFO] [1652161008.305298013] [camera.camera]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1652161010.880925665] [camera.camera]: Set ROS param depth_module.profile to default: 848x480x30
[realsense2_camera_node-1] [INFO] [1652161011.397687954] [camera.camera]: Set ROS param rgb_camera.profile to default: 1280x720x30
[realsense2_camera_node-1] [INFO] [1652161011.398902525] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1652161011.399142809] [camera.camera]: Set ROS param accel_fps to default: 100
[realsense2_camera_node-1] [WARN] [1652161011.445761961] [camera.camera]: Could not set param: pointcloud.stream_index_filter with -1 Range: [-1, -2147483648]: parameter 'pointcloud.stream_index_filter' could not be set: Parameter {} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1652161011.448645878] [camera.camera]: Start Sensor: Stereo Module
[realsense2_camera_node-1]  09/05 22:36:51,750 WARNING [140628889425664] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

The ending is the same.

sunil-gitcode, 2022-05-10:
<package format="3">
  <name>realsense2_camera</name>
  <version>4.0.4</version>
  <description>RealSense camera package allowing access to Intel T265 Tracking module and SR300 and D400 3D cameras</description>
  <maintainer email="librs.ros@intel.com">LibRealSense ROS Team</maintainer>
  <license>Apache License 2.0</license>
  
  <url type="website">http://www.ros.org/wiki/RealSense</url>
  <url type="bugtracker">https://github.com/intel-ros/realsense/issues</url>
version is 4.04

sunil-gitcode, 2022-05-11:
Sorry, I'm already using power. I cloned realsense's ROS2 branch from git and recompiled it to fix this problem.  It seems that using Method 1 to install will cause this error

sunil-gitcode, 2022-05-11:
> Sorry, I'm already using power. I cloned realsense's ROS2 branch from git and recompiled it to fix this problem. It seems that using Method 1 to install will cause this error

sunil-gitcode, 2022-05-11:
> A RealSense ROS user who was using wrapper 4.0.4 and experienced the same error **Could not set param: pointcloud.stream_index_filter** found that RViz2 was responsible for the error in their particular case and provided their solution at [#2295 (comment)](https://github.com/IntelRealSense/realsense-ros/issues/2295#issuecomment-1082660267)

sunil-gitcode, 2022-05-11:
> A RealSense ROS user who was using wrapper 4.0.4 and experienced the same error **Could not set param: pointcloud.stream_index_filter** found that RViz2 was responsible for the error in their [...]

Sorry, I'm already using power. I cloned realsense's ROS2 branch from git and recompiled it to fix this problem. It seems that using Method 1 to install will cause this error

MartyG-RealSense, 2022-05-11:
It's excellent news that you achieved a solution, @sunil-gitcode - thanks very much for the update and for sharing your solution with the RealSense community!

MartyG-RealSense, 2022-05-17:
Hi @sunil-gitcode  Do you require further assistance with this case, please?  Thanks!

sunil-gitcode, 2022-05-17:
> Hi @sunil-gitcode Do you require further assistance with this case, please? Thanks

No,please close this issue,Thanks

MartyG-RealSense, 2022-05-17:
Thanks very much for the update!

closed 2022-05-17 by MartyG-RealSense

youngday, 2022-08-11:
i build and run source code ,but D455 did not run properly .
same problem:
Could not set param: pointcloud.stream_index_filter with -1 Range: [-1, -2147483648]: parameter 'pointcloud.stream_index_filter' could not be set: Parameter {} doesn't comply with integer range.

MartyG-RealSense, 2022-08-11:
Hi @youngday  Have you tried the solution that worked for @sunil-gitcode in this discussion at https://github.com/IntelRealSense/realsense-ros/issues/2347#issuecomment-1123042027 please?

youngday, 2022-08-12:
I compiled librealsensense with version 2.51 and librealsensense-ros with version 4.04,point cloud and image could get from rviz2, but the warn still print out ,I can use it first,waiting for the reason of this warn ,thanks .

MartyG-RealSense, 2022-08-12:
If you receive a warning message but the performance of what you are doing does not appear to be negatively affected then the warning can likely be safely ignored.  

I do not have a meaning available for the **Could not set param: pointcloud.stream_index_filter with -1 range** message though

Does the warning go away if you set pointcloud.index_stream_filter to '0' in your launch instruction?

**ros2 launch realsense2_camera rs_launch.py pointcloud.stream_index_filter:=0**