realsenseai/realsense-ros #2348: roslaunch realsense2_camera rs_rgbd.launch ERROR!

Xue21, 2022-05-12:
when I roslaunch realsense2_camera rs_rgbd.launch color_width:=640 color_height:=480 color_fps:=15 depth_width:=640 depth_height:=480 depth_fps:=15,there are some questions.
[ WARN] [1652346309.385772917]: Hardware Notification:Depth stream start failure,1.65235e+12,Error,Hardware Error

MartyG-RealSense, 2022-05-12:
Hi @Xue21  Have you installed support for an RGBD launch please, as it it is not included by default in the RealSense ROS wrapper.  You need to first install the support using the appropriate instruction below for the ROS1 version that you are using.

**Kinetic**
sudo apt-get install ros-kinetic-rgbd-launch

**Melodic**
sudo apt-get install ros-melodic-rgbd-launch

**Noetic**
sudo apt-get install ros-noetic-rgbd-launch

*********

Also. which RealSense camera model are you using, please?  If it is an L515 then firmware version **1.5.8.1** should be installed in order for 640x480 color resolution to be supported.

Xue21, 2022-05-12:
@MartyG-RealSense,Thank you.it is well.

MartyG-RealSense, 2022-05-12:
That is great to hear - thanks for the update :)

closed 2022-05-19 by MartyG-RealSense