realsenseai/realsense-ros #2349: Images are available from only one camera in case of multiple D415 are attached

ravijo, 2022-05-13:
I am connecting 2 D415 in my system and using the following launch file:

```xml
<launch>
  <include file="$(find realsense2_camera)/launch/rs_camera.launch">
    <arg name="serial_no" value="112233445566" />
    <arg name="camera" value="camera1" />
    <arg name="initial_reset" default="true" />
  </include>

  <include file="$(find realsense2_camera)/launch/rs_camera.launch">
    <arg name="serial_no" value="112233445567" />
    <arg name="camera" value="camera2" />
    <arg name="initial_reset" default="true" />
  </include>
</launch>
```

It turned out that images are available only from camera 2. Surprisingly, the camera mentioned inside the second `<include> ...  </include>` block works. So if I move the second `<include> ...  </include>` block to the above first one, the bottom one, which is now camera 1 starts working. Weird!

I also used `rs_multiple_devices.launch` but no luck!
```console
$ roslaunch realsense2_camera rs_multiple_devices.launch 
```

I tried `rs_camera.launch` but no success!
```console
$ roslaunch realsense2_camera rs_camera.launch camera:=camera1 serial_no:=112233445566
$ roslaunch realsense2_camera rs_camera.launch camera:=camera2 serial_no:=112233445567
```

The `rs_camera.launch` reported the following version information of RealSense packages:
```
[ INFO] [1652435664.859818951]: RealSense ROS v2.3.2
[ INFO] [1652435664.859890701]: Built with LibRealSense v2.50.0
[ INFO] [1652435664.859928885]: Running with LibRealSense v2.50.0
[ INFO] [1652435664.861644135]: RealSense ROS v2.3.2
[ INFO] [1652435664.861713704]: Built with LibRealSense v2.50.0
[ INFO] [1652435664.861756794]: Running with LibRealSense v2.50.0
```
As per the official documentation, [RealSense ROS v2.3.2 is compatible with LibRealSense v2.50.0](https://github.com/IntelRealSense/realsense-ros/tree/2.3.2). So it seems no issue.

Furthermore, both of the cameras are available and work (both are listed as USB 3.2) inside the `realsense-viewer`. The `rs-enumerate-devices` can detect both of them as well as shown below:

```
$ rs-enumerate-devices -s
Device Name                   Serial Number       Firmware Version    
Intel RealSense D415          112233445566        05.13.00.50         
Intel RealSense D415          112233445567        05.13.00.50
```

I increase the `usbfs_memory_mb` to 256 but it didn't help.

The `dmesg` reported below info:
```
[ 3107.206191] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 3107.337902] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 3107.388175] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 3107.539001] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 3107.668650] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 3107.718930] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 3107.869804] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 3108.020159] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
```

Below is the more information:

* Kernel Info:
    ```console
    $ uname -a
    Linux dell 5.4.0-110-generic #124-Ubuntu SMP Thu Apr 14 19:46:19 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
    ```
* OS Info:
    ```console
    $ lsb_release -a
    No LSB modules are available.
    Distributor ID:	Ubuntu
    Description:	Ubuntu 20.04.4 LTS
    Release:	20.04
    Codename:	focal
    ```
* librealsense Package Info:
    ```console
    $ dpkg -l | grep librealsense
    ii  librealsense2:amd64                                 2.50.0-0~realsense0.6128                   amd64        Intel(R) RealSense(tm) Cross Platform API - runtime
    ii  librealsense2-dbg:amd64                             2.50.0-0~realsense0.6128                   amd64        Intel(R) RealSense(tm) Camera Capture API - debug symbols
    ii  librealsense2-dev:amd64                             2.50.0-0~realsense0.6128                   amd64        Intel(R) RealSense(tm) Camera Capture API - development files
    ii  librealsense2-dkms                                  1.3.18-0ubuntu1                            all          Modified kernel modules for librealsense2
    ii  librealsense2-gl:amd64                              2.50.0-0~realsense0.6128                   amd64        Intel(R) RealSense(tm) - GLSL-enabled extensions
    ii  librealsense2-gl-dbg:amd64                          2.50.0-0~realsense0.6128                   amd64        Intel(R) RealSense(tm) GLSL extensions - debug symbols
    ii  librealsense2-gl-dev:amd64                          2.50.0-0~realsense0.6128                   amd64        Intel(R) RealSense(tm) Camera Capture API - development files
    ii  librealsense2-net:amd64                             2.50.0-0~realsense0.6128                   amd64        Intel(R) RealSense(tm) - Network Bridge extension
    ii  librealsense2-net-dbg:amd64                         2.50.0-0~realsense0.6128                   amd64        Intel(R) RealSense(tm) Network Bridge extension - debug symbols
    ii  librealsense2-net-dev:amd64                         2.50.0-0~realsense0.6128                   amd64        Intel(R) RealSense(tm) Camera Capture API - Network Bridge extension development files
    ii  librealsense2-udev-rules:amd64                      2.50.0-0~realsense0.6128                   amd64        Intel(R) RealSense(tm) Camera Capture API - udev rules
    ii  librealsense2-utils:amd64                           2.50.0-0~realsense0.6128                   amd64        Intel(R) RealSense(tm) Camera Capture API - utils and demos
    ```
* GCC Info:
    ```console
    $ gcc --version
    gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0
    Copyright (C) 2019 Free Software Foundation, Inc.
    This is free software; see the source for copying conditions.  There is NO
    warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
    ```
* G++ Info:
    ```console
    $ g++ --version
    g++ (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0
    Copyright (C) 2019 Free Software Foundation, Inc.
    This is free software; see the source for copying conditions.  There is NO
    warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
    ```
* Python Info:
    ```console
    $ python --version
    Python 3.8.10
    ```

MartyG-RealSense, 2022-05-14:
Hi @ravijo  When you tried launching the two cameras with rs_camera.launch, can you confirm whether you launched them in two separate ROS terminals as the wrapper's multiple camera documentation advises, please?

https://github.com/IntelRealSense/realsense-ros#work-with-multiple-cameras

The documentation also advises providing the serial numbers within the launch instruction when using rs_multiple_devices.launch.  For example:

`roslaunch realsense2_camera rs_multiple_devices.launch serial_no_camera1:=112233445566 serial_no_camera2:=112233445567`

ravijo, 2022-05-14:
@MartyG-RealSense 

> ```
> <launch>
>   <include file="$(find realsense2_camera)/launch/rs_camera.launch">
>     <arg name="serial_no" value="112233445566" />
>     <arg name="camera" value="camera1" />
>     <arg [...]

I used the above launch file on ROS Noetic.

MartyG-RealSense, 2022-05-15:
If rs_camera.launch should be launched in separate ROS terminals when usng multiple cameras as per Intel's recommendation and you are definng the camera names and serial numbers in the launch file, aren't you setting the camera name and serial number twice (once for each terminal launch) when in fact you should only be setting the camera name and serial number for the particular camera used in a specific terminal by setting serial and name in the roslaunch instruction?

ravijo, 2022-05-15:
@MartyG-RealSense 

Sorry, I couldn't understand what you mean.

I have one launch file with the following content:

```xml
<launch>
  <include file="$(find realsense2_camera)/launch/rs_camera.launch">
    <arg name="serial_no" value="112233445566" />
    <arg name="camera" value="camera1" />
    <arg name="initial_reset" default="true" />
  </include>

  <include file="$(find realsense2_camera)/launch/rs_camera.launch">
    <arg name="serial_no" value="112233445567" />
    <arg name="camera" value="camera2" />
    <arg name="initial_reset" default="true" />
  </include>
</launch>
```

Furthermore, I also tried running it separately as well:

Terminal 1
```console
$ roslaunch realsense2_camera rs_camera.launch camera:=camera1 serial_no:=112233445566
```
Terminal 2
```console
$ roslaunch realsense2_camera rs_camera.launch camera:=camera2 serial_no:=112233445567
```

> Can you also confirm please that the serial numbers are correct? I have never before seen cameras with serials with only a 1-digit difference.

These serial numbers are replaced in the post. These two serial numbers are totally different in reality. Serial numbers were detected by the `rs-enumerate-devices -s` command.

MartyG-RealSense, 2022-05-15:
What I was suggesting is not to define args for the camera name and serial number within the 'rs_camera.launch' launch file, and do the configuration in the individual roslaunch instructions for each terminal instead.

ravijo, 2022-05-15:
@MartyG-RealSense 

> What I was suggesting is not to define args for the camera name and serial number within the 'rs_camera.launch' launch file

I am not modifying the `rs_camera.launch`. I have created a separate file, say `my_cameras.launch` with the following content:

```xml
<launch>
  <include file="$(find realsense2_camera)/launch/rs_camera.launch">
    <arg name="serial_no" value="112233445566" />
    <arg name="camera" value="camera1" />
    <arg name="initial_reset" default="true" />
  </include>

  <include file="$(find realsense2_camera)/launch/rs_camera.launch">
    <arg name="serial_no" value="112233445567" />
    <arg name="camera" value="camera2" />
    <arg name="initial_reset" default="true" />
  </include>
</launch>
```

> do the configuration in the individual roslaunch instructions for each terminal instead.

I tried it already. But no success. However, in reality, I don't want to have two terminals. I want to run `my_cameras.launch`.

MartyG-RealSense, 2022-05-15:
How about using the contents of rs_multiple_devices.launch in your custom launch file and only editing lines 2 and 3 to add your two serial numbers, and not changing any other details.

https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/rs_multiple_devices.launch#L2-L3

ravijo, 2022-05-16:
@MartyG-RealSense 

Thank you very much. I spent some time and found the issue.

**If the `initial_reset` is set to true for both the cameras, only one camera turns up.**

This behavior can be seen by setting the `initial_reset` to true in the following files:

- [rs_multiple_devices.launch](https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/rs_multiple_devices.launch)
- my_cameras.launch
    ```xml
    <launch>
      <include file="$(find realsense2_camera)/launch/rs_camera.launch">
        <arg name="serial_no" value="112233445566" />
        <arg name="camera" value="camera1" />
        <arg name="initial_reset" default="true" />
      </include>
    
      <include file="$(find realsense2_camera)/launch/rs_camera.launch">
        <arg name="serial_no" value="112233445567" />
        <arg name="camera" value="camera2" />
        <arg name="initial_reset" default="true" />
      </include>
    </launch>
    ```

Please note that the `initial_reset` is set to `false` in [rs_multiple_devices.launch#L2-L3](https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/rs_multiple_devices.launch#L2-L3)

MartyG-RealSense, 2022-05-16:
It's excellent to hear that you found a solution for your problem!  Do you require further assistance with this case, please?  Thanks again!

ravijo, 2022-05-16:
My observation brings us to a question: why only one camera turns up when the initial_reset is set to true for both the cameras?

I would appreciate greatly, if you can replicate the issue in your side and confirm my observations, please.

MartyG-RealSense, 2022-05-16:
There was a case that report problems with multiple camera initial_reset in January 2022 at https://github.com/IntelRealSense/realsense-ros/issues/2188 and the RealSense ROS wrapper developer provided feedback and identified a bug.  They implemented a fix for ROS2 but apparently not for ROS1.

Another RealSense ROS user implemented a fix themselves on their GitHub fork of the RealSense ROS1 wrapper.

https://github.com/dfautomation/realsense-ros/commit/a1e8a9b20d3dc1233409046e4e6d4c82724e8f80

A fix was apparently not implemented on the official ROS1 wrapper though.  RealSense ROS wrapper development is now focused on the ROS2 **ros2_beta** wrapper and further updates to the ROS1 wrapper are not planned.  So you may have to recreate the edit to the wrapper source code file **realsense_node_factory.cpp** on your own wrapper build detailed in the commit link above in order to achieve a solution, unfortunately.  Or you could build the forked version of the wrapper.

https://github.com/dfautomation/realsense-ros/

ravijo, 2022-05-17:
Thanks for the wonderful response. Now I can close this issue! Thanks again.

closed 2022-05-17 by ravijo

MartyG-RealSense, 2022-05-17:
You are very welcome.  I'm pleased that I could be of help.  Thanks for the update!