realsenseai/realsense-ros #2350: D455 not publishing imu on a Jetson Nano

MrOCW, 2022-05-14:
Hi,
Using Jetson Nano running Jetpack 4.6.1
installed librealsense2 from debian packages and built realsense-ros from source but unable to get the IMU from camera/imu topic, no messages received with rostopic hz camera/imu and rostopic echo
attempted to built librealsense from source with

./scripts/patch-realsense-ubuntu-L4T.sh  
which returns:
```
Jetson Board (proc/device-tree/model): NVIDIA Jetson Nano Developer Kit
Jetson L4T version: 32.7.1
Unsupported JetPack revision 32.7.1 aborting script
```
so i just went on to:
```
sudo apt-get install git libssl-dev libusb-1.0-0-dev libudev-dev pkg-config libgtk-3-dev -y
./scripts/setup_udev_rules.sh  
mkdir build && cd build  
cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=release -DFORCE_RSUSB_BACKEND=false -DBUILD_WITH_CUDA=true && make -j$(($(nproc)-1)) && sudo make install
```

then built realsense-ros from source.
Still had the same issue.
exactly the same issue as #2212 but my jetpack version is supposedly supported by librealsense v2.50....?

MrOCW, 2022-05-15:
Tried building librealsense v2.48.0 and realsense-ros 2.3.1 from source and that solved this issue but would still prefer to use a newer version

MartyG-RealSense, 2022-05-15:
Hi @MrOCW  The focus of the RealSense ROS wrapper developers has switched to the ROS2 beta wrapper (the 4.x series) and further updates are not planned to for the ROS1 wrapper (the 2.x series).  

So if you are concerned about the possibility of upgrading to newer versions then sticking with SDK 2.48.0 and wrapper 2.3.1 is likely to be the best option, as wrapper 2.3.2 / SDK 2.50.0 are not likely to provide a tangible benefit to your ROS setup unless you need support for metadata information in ROS which was added in wrapper 2.3.2.

closed 2022-05-21 by MrOCW