realsenseai/realsense-ros #2352: [ros2-beta] Inconsistent profile names depth_module.profile vs rgb_camera.profile

ravijo, 2022-05-18:
The ros2-beta branch has inconsistent profile names such as:

- depth_module.profile
- rgb_camera.profile

Source:

https://github.com/IntelRealSense/realsense-ros/blob/451147de3aaa9c5b05b1fb1f7a70f3f3f43c66f6/realsense2_camera/launch/rs_launch.py#L23-L30

Proposal: 
The rgb_camera.profile can be renamed to either on from below:

1. rgb_module.profile
2. color_module.profile

MartyG-RealSense, 2022-05-18:
Hi @ravijo  I would speculate (as I am not involved in wrapper development) that the **rgb_camera** naming is used by the ros2_beta wrapper's profile system because it has been the familiar term that has been used for a long period of time to describe the RGB camera in RealSense ROS wrappers.

[image]

ravijo, 2022-05-18:
@MartyG-RealSense 

I understand your opinion. 

However, this inconsistency made me check the source code of [rs_launch.py](https://github.com/IntelRealSense/realsense-ros/blob/451147de3aaa9c5b05b1fb1f7a70f3f3f43c66f6/realsense2_camera/launch/rs_launch.py#L23-L30). Before that, I was using the **wrong** command shown below:

```
ros2 launch realsense2_camera rs_launch.py depth_module.profile:=1280x720x15 rgb_module.profile:=1280x720x15
```

MartyG-RealSense, 2022-05-18:
I will forward your suggestion to the wrapper developers.

ravijo, 2022-05-18:
Thank you very much

MartyG-RealSense, 2022-05-30:
Hi @ravijo  I just wanted to update you that your request about a color_module.profile command is still being discussed by RealSense team members.  I will update again when I have further information to report.  Thanks very much for your patience!

ravijo, 2022-05-30:
@MartyG-RealSense Thank you very much. I appreciate your kind support.

MartyG-RealSense, 2022-06-24:
Hi @ravijo  Following discussions within Intel about creating a color_module.profile command, an official change request has been created so that Intel can investigate further regarding whether to implement this feature at a future date.  Thanks very much for your suggestion!

SamerKhshiboun, 2024-03-27:
@ravijo 

Please follow the latest Readme which include all information about these parameters, with new changes regarding there profiles
https://github.com/IntelRealSense/realsense-ros?tab=readme-ov-file#parameters-that-can-be-modified-during-runtime

The names of `rgb_module` and `depth_module` are coming from RealSense SDK, and representing the physical modules (Sensors) themselves, RGB and Depth.

@MartyG-RealSense Let me know if this enhancement still needed.

MartyG-RealSense, 2024-03-27:
Hi @SamerKhshiboun  We have not received any further requests for the creation of a color_module profile since the June 2022 one in this issue, so it does not seem to be a priority for RealSense ROS users.

SamerKhshiboun, 2024-03-29:
@MartyG-RealSense  Thanks for the update. 
I will close our internal ticket as I don't see any reason to implement this for now.
Let me know if we should reopen it in the future.

MartyG-RealSense, 2024-03-30:
Thanks very much @SamerKhshiboun  I will close this issue as it is not going to be implemented.

closed 2024-03-30 by MartyG-RealSense