realsenseai/realsense-ros #2353: Zero-copy point cloud subscriber in ROS2

ravijo, 2022-05-18:
The components in ROS2 provide Zero-copy communication. Luckily, the realsense package comes with this option. 

Quoting the following from [Efficient intra-process communication](https://github.com/IntelRealSense/realsense-ros/tree/ros2-beta#efficient-intra-process-communication) section:
> Our ROS2 Wrapper node supports zero-copy communications if loaded in the same process as a subscriber node. This can reduce copy times on image topics (not point-cloud or others), especially with big frame resolutions and high FPS.

**Does it mean that point clouds are not supported for zero-copy if loaded in the same process as a subscriber node?**

MartyG-RealSense, 2022-05-18:
Hi @ravijo  Another way to interpret that statement would be that copy times are not reduced for pointclouds.

ravijo, 2022-05-18:
Hi @MartyG-RealSense 

Sorry for the confusion. Does that mean intra-process communication isn't helpful in the case of point cloud???  How to achieve zero-copy communication for point clouds then?

MartyG-RealSense, 2022-05-18:
That would be my interpretation that zero-copy would not be useful for pointclouds.  

Someone in the link below wrote a ROS2 module that was tested with ROS2 Galactic that supports pointclouds with zero-copy, though I cannot guarantee how well it will work with the RealSense ROS2 wrapper.

https://github.com/ZhenshengLee/ros2_shm_msgs

MartyG-RealSense, 2022-05-25:
Hi @ravijo  Do you require further assistance with this case, please?  Thanks!

ravijo, 2022-05-25:
@MartyG-RealSense: I am not dealing with it at prersent. So it is better to close this issue. Thanks for the kind support.

MartyG-RealSense, 2022-05-25:
You are very welcome.  Thanks very much for the update!

closed 2022-05-25 by MartyG-RealSense

ZhenshengLee, 2022-06-28:
> Someone in the link below wrote a ROS2 module that was tested with ROS2 Galactic that supports pointclouds with zero-copy, though I cannot guarantee how well it will work with the RealSense ROS2 wrapper.

@MartyG-RealSense  @ravijo 

Hi, this lib https://github.com/ZhenshengLee/ros2_shm_msgs has been used in the custom version of ros2_v4l2_camera 
 and rslidar_sdk, and the performance improvement is awesome!

see https://github.com/ZhenshengLee/ros2_v4l2_camera
and https://github.com/ros2driver/rslidar_sdk/tree/outdoor/dev_opt_shm

if you need any help to merge this module into the RealSense ROS2 , feel free to ping me.

MartyG-RealSense, 2022-06-28:
Thanks so much @ZhenshengLee for sharing this with the RealSense ROS community!

ZhenshengLee, 2022-12-05:
@toan70411hd91 

> I just wonder whether some modules in ros2_shm_msgs are merged into Realsense ROS2 or not? 

No, nothing has been merged.

> Or Type adaption is considered in ROS2 Realsense Wrapper (Humble) or not?

It's unknown for me, because I didn't track this feature.

ZhenshengLee, 2022-12-05:
@toan70411hd91 
You can use zero-copy through intra-process comm which is provided officially, check it in https://github.com/IntelRealSense/realsense-ros#efficient-intra-process-communication

toan70411hd91, 2022-12-06:
@ZhenshengLee thank you so much for your information and quick helps. May I ask you some questions related to ros2_shm_msgs?
1) In folder "intra", i found that you added intra type adaption within cpu. Currently, ros2_shm_msgs only support type adaption (within CPU) for image only, but not for pointcloud?
2) Is it correct if "zero-copy" in ros2_shm_msgs called "minimal copy" when we publish topics?  
3) In your opinion, whether the performance (regarding to speed and CPU consumption) much better if we use type adaption?

ZhenshengLee, 2022-12-06:
> @ZhenshengLee thank you so much for your information and quick helps. May I ask you some questions related to ros2_shm_msgs?
> 
> 1. In folder "intra", i found that you added intra type adaption within cpu. Currently, ros2_shm_msgs only support type adaption (within CPU) for image only, but not for pointcloud?
> 2. Is it correct if "zero-copy" in ros2_shm_msgs called "minimal copy" when we publish topics?
> 3. In your opinion, whether the performance (regarding to speed and CPU consumption) much better if we use type adaption?
> 
> Even I did not use https://github.com/IntelRealSense/realsense-ros#efficient-intra-process-communication, the performance of ROS2 Realsense Wrapper (regarding to speed and CPU consumption) is almost same with ros2_shm-msgs. I am not sure the hints inside since i am quite new in this topic. Maybe, my tested point-cloud data is small (camera resolution 640x480, with decimation filter).

@toan70411hd91 please post this issue in https://github.com/ZhenshengLee/ros2_shm_msgs

I will find time to answer this.

Thanks!

toan70411hd91, 2022-12-06:
@ZhenshengLee Thank you so much for your active and quick helps. I posted this issue in https://github.com/ZhenshengLee/ros2_shm_msgs

ZhenshengLee, 2022-12-09:
> Thank you so much for your active and quick helps. I posted this issue in

anyone interested would follow this
https://github.com/ZhenshengLee/ros2_shm_msgs/issues/25