realsenseai/realsense-ros #2354: symbol lookup error ubuntu18.04 and realsense D435

yyy8Bb, 2022-05-18:
I get an error as follows:
[ INFO] [1652888152.312185349]: Setting Dynamic reconfig parameters.
/opt/ros/melodic/lib/nodelet/nodelet: symbol lookup error: /home/young8/mobile_manipulater/real_ws/devel/lib//librealsense2_camera.so: undefined symbol: _ZN20ddynamic_reconfigure19DDynamicReconfigureC1ERKN3ros10NodeHandleEb

[camera/realsense2_camera_manager-2] process has died [pid 11892, exit code 127, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/young8/.ros/log/327a4254-d6c0-11ec-904d-38002590fcd6/camera-realsense2_camera_manager-2.log].
log file: /home/young8/.ros/log/327a4254-d6c0-11ec-904d-38002590fcd6/camera-realsense2_camera_manager-2*.log

yyy8Bb, 2022-05-18:
> Hi @yyy8Bb This error typically occurs if dynamic_reconfigure has been installed more than once on the same computer (perhaps because there is an old ROS wrapper build on it). This error is discussed at #838

much thanks!!

yyy8Bb, 2022-05-19:
> Hi @yyy8Bb This error typically occurs if dynamic_reconfigure has been installed more than once on the same computer (perhaps because there is an old ROS wrapper build on it). This error is discussed at #838

New error comes as follows:

[ INFO] [1652953273.200301986]: Done Setting Dynamic reconfig parameters.
/opt/ros/melodic/lib/nodelet/nodelet: symbol lookup error: /home/young8/mobile_manipulater/real_ws/devel/lib//librealsense2_camera.so: undefined symbol: _ZN2cv3MatC1Ev

It looks like lack opencv lib.So I add OpenCV.LIB in the cmakelist of realsense2_camera.It seems don't work.

MartyG-RealSense, 2022-05-19:
A RealSense ROS user who experienced the **_ZN2cv3MatC1Ev** error a month ago at https://github.com/IntelRealSense/realsense-ros/issues/2326 found that CMakeLists edits suggested in that discussion at https://github.com/IntelRealSense/realsense-ros/issues/2326#issuecomment-1107658481 resolved their problem.

yyy8Bb, 2022-05-19:
> A RealSense ROS user who experienced the **_ZN2cv3MatC1Ev** error a month ago at #2326 found that CMakeLists edits suggested in that discussion at [#2326 (comment)](https://github.com/IntelRealSense/realsense-ros/issues/2326#issuecomment-1107658481) resolved their problem.

perfectly solved!!!!
I use  ${OpenCV_LIBRARIES} to replace ${OpenCV_LIB}.It works.
Thanks for your patience.One more question,is there a official document or guide to help me,a rookie,with something like  how to change fps of rgb image or depth image and so on

yyy8Bb, 2022-05-19:
> A RealSense ROS user who experienced the **_ZN2cv3MatC1Ev** error a month ago at #2326 found that CMakeLists edits suggested in that discussion at [#2326 (comment)](https://github.com/IntelRealSense/realsense-ros/issues/2326#issuecomment-1107658481) resolved their problem.

To set configuration ,what i need to do is just change the launch file, yes?

MartyG-RealSense, 2022-05-19:
On the ROS1 wrapper, you can edit the launch file with custom values but you can also define a custom configuration in the roslaunch instruction without having to change the launch file.

For each stream type that you are using, you should provide three factors - stream width, height and FPS.  For example to enable depth and color streams at 640x480 with 30 FPS:

`roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:-30 color_width:=640 color_height:=480 color_fps:=30`

MartyG-RealSense, 2022-05-25:
Hi @yyy8Bb  Do you require further assistance with this case, please?  Thanks!

yyy8Bb, 2022-05-28:
> Hi @yyy8Bb Do you require further assistance with this case, please? Thanks!

Hi.After add the target to opencv lib in cmakelist in realsense2_camera,I can run rs_camera.launch and check depth and rgb image in rviz now.
But when I use  roslaunch realsense2_camera rs_camera.launch filters:=pointcloud  or run demo.pointcloud.launch .
The undefined symbol error comes again:

[ INFO] [1653718682.587429643]: Done Setting Dynamic reconfig parameters.
/opt/ros/melodic/lib/nodelet/nodelet: symbol lookup error: /home/young8/mobile_manipulater/real_ws/devel/lib//librealsense2_camera.so: undefined symbol: _ZN2cv3MatC1Ev
[camera/realsense2_camera_manager-2] process has died [pid 4976, exit code 127, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/young8/.ros/log/eb2ea3b4-de4d-11ec-943c-38002590fcd6/camera-realsense2_camera_manager-2.log].
log file: /home/young8/.ros/log/eb2ea3b4-de4d-11ec-943c-38002590fcd6/camera-realsense2_camera_manager-2*.log

After that,I run rs_camera.launch which can work before,still error like that.
I tried remove the devel build and install in the workspace,then catkin_make again.But I can't run rs_camera.launch.
If I make another workspace,git 、add something about opencv in cmakelist and build again.I can run rs_camera.launch.
But still, I can not use  roslaunch realsense2_camera rs_camera.launch filters:=pointcloud  or run demo.pointcloud.launch  just same as before.
It seems strange for me.Thanks for your patience.

MartyG-RealSense, 2022-05-28:
I researched the problem again from the beginning.  The **_ZN2cv3MatC1Ev** type of symbol lookup error is so rare in RealSense that there is almost no information about how to deal with it though, unfortunately.  In these circumstances where there is no references about how to proceed, I would typically recommend a complete wipe and reinstall of the computer, including the operating system, to try to clear any hidden problems that might be causing the error.

I note that you found a method for removing the error if you make another workspace though.

If you are removing a previous wrapper build, I would recommend deleting the whole **catkin_ws/src** workspace folder and not just the files inside it if you have not tried a complete workspace folder deletion already.  Then create a new **catkin_ws/src** folder and build the wrapper again.

We can also try an **RGBD** type launch to produce a pointcloud.  first, install support for the RGBD launch file with the command below for Melodic:

**sudo apt-get install ros-melodic-rgbd-launch**

Then use this roslaunch instruction:

**roslaunch realsense2_camera rs_rgbd.launch**

This launch should produce an ordered pointcloud.

closed 2022-06-09 by MartyG-RealSense