realsenseai/realsense-ros #2356: [ WARN] [1653074716.457914845]: No RealSense devices were found!

Shawn207, 2022-05-20:
System distribution: ubuntu 20.04 WSL2
Kernel version:  5.10.16.3-microsoft-standard-WSL2
ros version: Noetic
camera device: D435i
I followed the instruction method1: install realsense2_camera from the Ros distribution by running:
`sudo apt-get install ros-noetic-realsense2-camera`
But I got the following message:
`... logging to /root/.ros/log/02351cd6-d86e-11ec-9d60-00155dafeee6/roslaunch-DESKTOP-KCE729Q-17165.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://DESKTOP-KCE729Q:45267/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file:
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type:
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters:
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_rgb: False
 * /camera/realsense2_camera/infra_width: 848
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path:
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/ordered_pc: False
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/publish_odom_tf: True
 * /camera/realsense2_camera/publish_tf: True
 * /camera/realsense2_camera/reconnect_timeout: 6.0
 * /camera/realsense2_camera/rosbag_filename:
 * /camera/realsense2_camera/serial_no:
 * /camera/realsense2_camera/stereo_module/exposure/1: 7500
 * /camera/realsense2_camera/stereo_module/exposure/2: 1
 * /camera/realsense2_camera/stereo_module/gain/1: 16
 * /camera/realsense2_camera/stereo_module/gain/2: 16
 * /camera/realsense2_camera/tf_publish_rate: 0.0
 * /camera/realsense2_camera/topic_odom_in: odom_in
 * /camera/realsense2_camera/unite_imu_method:
 * /camera/realsense2_camera/usb_port_id:
 * /camera/realsense2_camera/wait_for_device_timeout: -1.0
 * /rosdistro: noetic
 * /rosversion: 1.15.14

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

process[camera/realsense2_camera_manager-1]: started with pid [17179]
process[camera/realsense2_camera-2]: started with pid [17180]
[ INFO] [1653074710.346352585]: Initializing nodelet with 8 worker threads.
[ INFO] [1653074710.434581765]: RealSense ROS v2.3.2
[ INFO] [1653074710.434713045]: Built with LibRealSense v2.50.0
[ INFO] [1653074710.434767724]: Running with LibRealSense v2.50.0
[ WARN] [1653074710.453861312]: No RealSense devices were found!
[ WARN] [1653074716.457914845]: No RealSense devices were found!
^C[camera/realsense2_camera-2] killing on exit
[camera/realsense2_camera_manager-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done`

I also tried to install the realsense SDK 2.0 and run realsense-viewer, but no deviced can be detected in realsense viewer. When I was installing realsense SDK 2.0, I noticed that kernel version 5.10 seems not to be supported. Is that where the problem lies?
Thanks !

MartyG-RealSense, 2022-05-20:
Hi @Shawn207  As a starting point in investigating your issue, I recommend reading a case at https://github.com/IntelRealSense/librealsense/issues/10506 from a week ago with similar conditions.  The RealSense user in that case also had camera detection problems with WSL2 on Ubuntu 20.04 with kernel 5.10.

Shawn207, 2022-05-20:
Thanks for your quick reply!
Unfortunately, the situations are different. That user is able to run realsense-viewer with sudo, but I also failed to do so. I also tried to build libsense SDK with libusb, but the problem keeps.

Shawn207, 2022-05-20:
Also, I tried to download realsense-viewer windows version, and it works. I don't know still why it fails on ubuntu

MartyG-RealSense, 2022-05-27:
Hi @Shawn207  Do you require further assistance with this case, please?  Thanks!

ngdyes, 2022-06-01:
A workable solution for me  is  "sudo service udev restart",and replug the camera.

Shawn207, 2022-06-01:
Hi, thank you for your reply. I didn’t solve the problem, but I changed to
run it on windows and recorded a rosbag as a substitute

MartyG-RealSense, 2022-06-01:
@ngdyes  Thanks very much for your suggestion to @Shawn207 

@Shawn207  I appreciate the update.  Thanks again!

closed 2022-06-12 by MartyG-RealSense