realsenseai/realsense-ros #2358: Depth units not loaded from json with param-depthunits or param-zunits, limits min depth step to 1mm

J-C-Haley, 2022-05-26:
Hello, I'm trying for a close up/higher accuracy scan. Loading param-disparityshift from a .json works fine, along with other items. However, I see 1mm steps in the data corresponding to the Y16 discretization, and resetting param-depthunits or param-zunits  does not allow for smaller units as in the viewer. Any suggestions? Thank you!

[image]

MartyG-RealSense, 2022-05-27:
Hi @J-C-Haley  A RealSense ROS wrapper developer states at https://github.com/IntelRealSense/realsense-ros/issues/277#issuecomment-525676873 explains why the wrapper does not support changing the depth scale in the same way that it can be changed in the librealsense SDK.

Instead of changing the depth units and other items, you could try applying the SDK's built-in **High Accuracy** preset.  This can be done by adding a rosparam instruction to the launch file, as described by the same wrapper developer at https://github.com/IntelRealSense/realsense-ros/issues/2055#issuecomment-968278545

J-C-Haley, 2022-05-27:
> developer states at #277 (comment)
> wrapper developer at #2055 (comment)

Thank you, good find, I was worried that was the case. This is probably
going to be an issue for other people too with the new D405. This probably
has a pretty bad impact on normals estimation too. I'll see if I can
implement the scaling myself, no promises yet, but 1mm quantization isn't
going to do.

MartyG-RealSense, 2022-05-27:
The D405 uses a different depth unit scale to the rest of the 400 Series models.  As it is designed for very close range depth sensing (ideal range 7 cm to 50 cm), its default scale is **0.01** instead of the 0.001 default that the other 400 Series models use.  

The D405 model is only be supported in the ros2_beta wrapper though and will not be supported in older wrappers such as ROS1 and the previous generation ROS2 wrapper.

In librealsense there are two separate depth scale options, the Advanced Mode one associated with the json that is set to '1000' and another that is '0.001' by default, as mentioned above.  In the Viewer, this alternative option is under the **Stereo Module > Controls** section.

[image]

I wonder whether an alternative to setting the Advanced Mode depth unit scale in json would be to use a **node script** in ROS such as a Python one to change from within ROS the other 'Depth Units' option with the librealsense instruction **rs.option.depth_units**, as described at https://github.com/IntelRealSense/librealsense/issues/1905

MartyG-RealSense, 2022-05-27:
My knowledge of node scripts is limited, but typically I think that the camera is first roslaunched in the ROS termnal in the usual way and then a node script is run that contains functions that do not involve a pipeline start (since the pipeline has already been started by the roslaunch).  This is shown in the wrapper's set of official node script examples at the link below, which do not contain a pipe start instruction.

https://github.com/IntelRealSense/realsense-ros/tree/development/realsense2_camera/scripts

J-C-Haley, 2022-05-31:
So, to update on what I've figured out on this:
- The node script examples do not access the SDK directly, they only set options through the driver, so they won't be able to set depth scale without implementation in the driver.
- I can reset the depth scale if I test hard-coding the setting into the driver:
sensor.set_option(RS2_OPTION_DEPTH_UNITS, 0.0001f);
(in base_realsense_node.cpp, setupStreams() right before reading _depth_scale_meters)

This actually seems to not downscale the units, **however** it's oddly unstable, sometimes I get 0.1mm quantization + real units in ros, and sometimes I get 1mm real-scale quantization + 1/10th scale units in ROS:
 
Lower 0.1mm real scale quantization, projected into ROS at real scale:
[image]
1mm real-scale quantization, projected into ROS at 1/10 scale:
[image]

So, I'm still confused, going to keep attempting to debug. If/when I find a solution, I'll add it as a rosparam and make a PR. Thanks!

MartyG-RealSense, 2022-05-31:
Thanks very much for the update and the sharing of your thought process so far!

MartyG-RealSense, 2022-06-18:
Hi @J-C-Haley  Do you require further assistance with this case, please?  Thanks!

J-C-Haley, 2022-06-19:
Hello @MartyG-RealSense , I can update on this - there's already a hardcoded parameter in constants.h - ROS_DEPTH_SCALE  = 0.001. My testing was racing with this being set. It looks like there's already rescaling happening to correct for the unit change, so just adjusting this to 0.0001 worked for me. I may fork and add as a ros parameter if I find time, but I think we can close this (anyone can build from source and adjust directly if they need).

MartyG-RealSense, 2022-06-20:
Thanks very much @J-C-Haley for the update and the sharing of your findings about ROS_DEPTH_SCALE.  As you suggest, I will close this issue.  Thanks again!

closed 2022-06-20 by MartyG-RealSense