realsenseai/realsense-ros #2363: ROS-realsense with D405 errors

yuzej, 2022-06-02:
I'm now using Realsense D405 camera. The SDK 2.0 in windows and ubuntu is realised to seceived both 2D and 3D image.
Recently I want to employ D405 in ROS melodic with Ubuntu18.

I have tried the 2 methods in the README file. The errors are listed below:
1. sudo apt-get install ros-melodic-realsense2-camera
    E: unable to locate the package 
    when input as ros-melodic-re , press "tab" button, only the ros-melodic-realtime-tools and ros-melodic-resource-retriever show.

2. Install from source
   The workspace is created and catkin_make success. When running "roslaunch realsense2_camera rs_camera.launch", the errors is below:
[ INFO] [1654135956.501689406]: Initializing nodelet with 1 worker threads.
[ INFO] [1654135956.575161138]: RealSense ROS v2.3.2
[ INFO] [1654135956.575297787]: Built with LibRealSense v2.50.0
[ INFO] [1654135956.575356184]: Running with LibRealSense v2.50.0
[ INFO] [1654135956.594469388]:  
[ INFO] [1654135956.608103916]: Device with serial number 123622270967 was found.

[ INFO] [1654135956.608234424]: Device with physical ID /sys/devices/pci0000:00/0000:00:15.0/0000:03:00.0/usb4/4-1/4-1:1.0/video4linux/video0 was found.
[ INFO] [1654135956.608288499]: Device with name Intel RealSense D405 was found.
[ INFO] [1654135956.608730154]: Device with port number 4-1 was found.
[ INFO] [1654135956.608805337]: Device USB type: 3.2
[FATAL] [1654135956.608897888]: Unsupported device! Product ID: 0x0B5B
[camera/realsense2_camera_manager-2] process has died [pid 55678, exit code 1, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/jyz/.ros/log/7818498c-e219-11ec-8035-000c297c12bd/camera-realsense2_camera_manager-2.log].
log file: /home/jyz/.ros/log/7818498c-e219-11ec-8035-000c297c12bd/camera-realsense2_camera_manager-2*.log

So, what should I do to use my D405 in ROS? Which configuration should I change ?

MartyG-RealSense, 2022-06-02:
Hi @yuzej  The D405 camera model is only supported in the **ros2_beta** wrapper for ROS2 (currently version 4.0.4).  There are no plans to add D405 support to the ROS1 wrapper or the previous generation of ROS2 wrapper.

MartyG-RealSense, 2022-06-09:
Hi @yuzej  A RealSense ROS user at https://github.com/IntelRealSense/realsense-ros/pull/2297 has created a custom version of the ROS1 wrapper that contains D405 support.

https://github.com/rjwb1/realsense-ros

MartyG-RealSense, 2022-06-14:
Thanks very much.  As you are satisfied with the outcome of the case, I will close it.  Thanks again!

closed 2022-06-14 by MartyG-RealSense

yuzej, 2022-10-11:
font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    
    Thank you for the reply.
    It's my first time to know that the D405 has the requirement of ROS versions.
        My question is, shall I use D405 in openCV or python to locate the objects, and consider the locations as input of ROS scripts, then control the robot?
    
         
        
            font{
            line-height: 1.6;
        }Yuze ***@***.***            ***@***.***            
        
    
     

  
    ---- Replied Message ----
  
  
    
      
        
         From 
        
        
            ***@***.***>
            
        
      
      
        
         Date 
        
        
        6/2/2022 13:51
        
      
      
        
         To 
        
        
         
          
            ***@***.***>
            
          
        
      
      
        
         Cc 
        
        
          
            ***@***.***>
            ,
          
          
            ***@***.***>
            
          
        
      
      
        
         Subject 
        
        
              Re: [IntelRealSense/realsense-ros] ROS-realsense with D405 errors (Issue #2363)
        
      
    
  
  
Hi @yuzej  The D405 camera model is only supported in the ros2_beta wrapper for ROS2 (currently version 4.0.4).  There are no plans to add D405 support to the ROS1 wrapper or the previous generation of ROS2 wrapper.

yuzej, 2022-10-11:
font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    
    
        Thanks god!
        Pretty excellent job!I have run the demo scripts of ROS-realsense wrapper in Ubuntu18, ROS melodic, with D405 camera.The issue is solved perfect.
    
        
            
        
    
    
         
        
            font{
            line-height: 1.6;
        }
        
    
     

  
    ---- Replied Message ----
  
  
    
      
        
         From 
        
        
            ***@***.***>
            
        
      
      
        
         Date 
        
        
        6/9/2022 14:21
        
      
      
        
         To 
        
        
         
          
            ***@***.***>
            
          
        
      
      
        
         Cc 
        
        
          
            ***@***.***>
            ,
          
          
            ***@***.***>
            
          
        
      
      
        
         Subject 
        
        
              Re: [IntelRealSense/realsense-ros] ROS-realsense with D405 errors (Issue #2363)
        
      
    
  
  
Hi @yuzej  A RealSense ROS user at #2297 has created a custom version of the ROS1 wrapper that contains D405 support.
https://github.com/rjwb1/realsense-ros

MartyG-RealSense, 2022-10-11:
Thanks very much @yuzej - it's great to hear that you were successful  :)