realsenseai/realsense-ros #2364: D435i - Odomquality=0, RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided

SerdarAbali, 2022-06-02:
Dear community, 
what does it take it to make this nearly 400 EUR device work properly when I send the command:

`roslaunch realsense2_camera opensource_tracking.launch`

How do I troubleshoot the below errors?
`[ INFO] [1654169704.476895759]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.103005s
[ERROR] (2022-06-02 14:35:04.480) Rtabmap.cpp:1278::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 167 is ignored!
`
It is just plain and simple question.

MartyG-RealSense, 2022-06-02:
Hi @SerdarAbali  Please try using the roslaunch instruction below to enable the IMU's gyro and accelerometer, as they are disabled by default in the RealSense ROS wrapper.

`roslaunch realsense2_camera opensource_tracking.launch enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation`

SerdarAbali, 2022-06-02:
Dear @MartyG-RealSense , I am aware that it is disabled and I already enabled gyro. Just to make sure for 50th time, I reinstalled realsense2_camera seconds ago and I had exactly same error after sending the exact same command you posted above:
`roslaunch realsense2_camera opensource_tracking.launch enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation
... logging to /home/debster/.ros/log/73fbfac6-e26c-11ec-a868-30e37abd990d/roslaunch-debster-KBL-U-Pro-Cubi2-MS-B142-21097.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.103:37173/

SUMMARY
========

CLEAR PARAMETERS
 * /rtabmap/rgbd_odometry/
 * /rtabmap/rtabmap/
 * /ukf_se/

PARAMETERS
 * /ImuFilter/_publish_tf: False
 * /ImuFilter/_world_frame: enu
 * /ImuFilter/use_mag: False
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
[... 405 lines cut ...]
[ INFO] [1654171606.636778510]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.096749s
[ WARN] (2022-06-02 15:06:46.731) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=9) between -1 and 45"
[ INFO] [1654171606.732305332]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.092845s
[ WARN] (2022-06-02 15:06:46.833) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=11) between -1 and 46"
[ INFO] [1654171606.834104313]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.099079s
[ WARN] (2022-06-02 15:06:46.935) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=15) between -1 and 47"
[ INFO] [1654171606.935963016]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.095327s
[ERROR] (2022-06-02 15:06:46.939) Rtabmap.cpp:1278::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 157 is ignored!
[ INFO] [1654171606.939883905]: rtabmap (0): Rate=1.00s, Limit=0.000s, Conversion=0.0032s, RTAB-Map=0.0001s, Maps update=0.0000s pub=0.0000s (local map=0, WM=0)
^C[ukf_se-7] killing on exit
[rtabmap/rtabmap-6] killing on exit
[rtabmap/rgbd_odometry-5] killing on exit
[ImuFilter-4] killing on exit
[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[ WARN] (2022-06-02 15:06:47.053) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=14) between -1 and 48"
[ WARN] (2022-06-02 15:06:47.259) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=14) between -1 and 49"
rtabmap: Saving database/long-term memory... (located at /home/debster/.ros/rtabmap.db)
rtabmap: Saving database/long-term memory...done! (located at /home/debster/.ros/rtabmap.db, 0 MB)
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`

MartyG-RealSense, 2022-06-02:
Most of the errors that you experienced are also reported in the case https://github.com/IntelRealSense/realsense-ros/issues/1793

Do the errors occur if the camera is stationary, or do they start once the camera begins moving?

MartyG-RealSense, 2022-06-02:
Thank you.  Does the IMU publish its topics correctly if launching with rs_camera instead of rs_opensource_tracking

`roslaunch realsense2_camera rs_camera.launch enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation`

SerdarAbali, 2022-06-02:
Yes, it is published
`/camera/gyro/imu_info
/camera/imu
`
and rostopic echo shows value changes when I move the camera which proves the brand new hardware healthy

Rviz plugin tutorial set to camera/imu visualisation also work fine when I move the camera, it is very accurate.

But I get the error below:
`WARNING [140196892882688] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11`
I also tried with initial_reset:=true

MartyG-RealSense, 2022-06-02:
The developer of the RealSense ROS1 wrapper has said in the past that the **Resource temporarily unavailable error** message can be related to lack of system resources - the application can't keep up with the incoming flow of images.  

What computer / computing device are you using, please?

SerdarAbali, 2022-06-02:
I use an MSI Cubi 2, sort of Intel NUC equivalent version of MSI.

4G ram - DDR4 2133MHz 
Intel(R) Core(TM) i3-7100U CPU @ 2.40GHz

SerdarAbali, 2022-06-02:
hold on for 30 minutes please, I am builting rtabmap from source instead of packages

MartyG-RealSense, 2022-06-02:
It sounds like a good specification.  Are you using the official 1 meter USB cable supplied with the camera or a longer cable of your own choice?

SerdarAbali, 2022-06-02:
> It sounds like a good specification. Are you using the official 1 meter USB cable supplied with the camera or a longer cable of your own choice?

I will report back after I build rtabmap ros in 30 mins. I suspect rtabmap package that comes with melodic. I will report back shortly for the sake of next guy that purchases this camera and make it work with Ros1

MartyG-RealSense, 2022-06-02:
That's excellent news  :)  Thanks very much for the update!

closed 2022-06-08 by MartyG-RealSense