realsenseai/realsense-ros #2365: Optical filters

hakunaMatataHub, 2022-06-03:
I am using D435i in my indoor application. The scene is mostly about 1-1.5 meters. I read the white paper on optical filters and how they can be used to improve depth map estimation in case of glare due to specular reflection from indoor lighting. 
I am interested to know what is the wavelength range of the IR spectrum used in the projector which would help me select appropriate filter.

Thank you.

MartyG-RealSense, 2022-06-03:
Hi @hakunaMatataHub  The recommended IR wavelength of projectors used with RealSense cameras is 850 nm.  The cameras can also see the 940 nm wavelength (though with reduced sensitivity), as described at https://github.com/IntelRealSense/librealsense/issues/3370

**Projector Parameters**

[image]

If you are interested in using a **thin film linear polariztion filter** to greatly negate the effects of glare from reflections then any polarizing filter that is linear should be sufficient (though not the circular filters used in 3D glassses).  You can search stores such as Amazon for the term **linear polarizing filter sheet**

You may also be interested in the new RealSense D435f camera model launched yesterday that features a light filter on its left and right sensors.  More information can be found at https://github.com/IntelRealSense/librealsense/issues/10560

hakunaMatataHub, 2022-06-03:
@MartyG-RealSense 
Thank you so much. I am checked the new release, would love to see it with the inertial unit.

If i get the white paper right, the infrared stereo cameras are also sensitive to visible light, right? So would not a long pass infrared filter work well as well?

Or does long pass polarising infrared filters exist?
Please help me, I am very new to these filters and don't have much idea.

MartyG-RealSense, 2022-06-03:
The left and right IR sensors on most 400 Series stereo camera models can see all visible and near-infrared light frequencies. as they are not equipped with filters.  

An exception is the **D405** short-range depth sensing model, which is equipped with an **IR-Cut** filter on its left and right sensors and so can see visible light frequencies but not infrared and also cannot see IR dot pattern projections (it is not equipped with an IR dot pattern projector anyway).

In section **3.  Types of optical filter** in the optical filter white-paper, it describes that a shortpass filter lets through wavelengths **below** a certain frequency and blocks the frequencies above that range, whilst a longpass filter allows frequencies **above** a certain wavelength.

I would therefore think that if you wanted to block visible light but allow near-IR and IR light to be perceived then a longpass filter should be used, and if you wanted to allow visible light but block IR frequencies then a shortpass filter should be used.

A senior RealSense team member recommended a couple of examples of purchasing sources for filters (Astra Products and Lee Filters) at https://github.com/IntelRealSense/librealsense/issues/2875#issuecomment-446793681 but said that Astra do not provide the **polarizer** type of filter.

hakunaMatataHub, 2022-06-03:
@MartyG-RealSense 
Thank you for the suggestions.

MartyG-RealSense, 2022-06-10:
Hi @hakunaMatataHub  Do you require further assistance with this case, please?  Thanks!

hakunaMatataHub, 2022-06-10:
@MartyG-RealSense 
I have ordered NIR linear polarisers from [link](https://www.edmundoptics.in/p/15mm-dia-nir-linear-polarizing-film/45625/) for my d435i as suggested in the white paper. 
I would appreciate if you could suggest me what would be the best way to mount it?
Should I mount it on the camera casing from outside or should I mount it on the lens after removing the casing or any other way?
I know I should take care to not apply filter to the projector but is there any other guidelines which I need to keep in mind for best results?

MartyG-RealSense, 2022-06-10:
Filters can be placed on the outside of the camera without having to remove the casing's cover glass.  One person created a custom-made 3D printed housing for the D435 camera to hold the filter (see the link below), whilst another person simply held the filter on with elastic bands wrapped around the camera.

https://www.thingiverse.com/thing:5140445

If you are placing filters on the infrared sensors then put a filter on both the left and right sensor.  Also, when mounting the filters be careful that solid parts of the mount do not overlap the sensors and obscure their field of view.

hakunaMatataHub, 2022-06-10:
@MartyG-RealSense 
Thank you I will try this

closed 2022-06-22 by MartyG-RealSense