realsenseai/realsense-ros #2366: Setting config parameters from a yaml file in ROS

hakunaMatataHub, 2022-06-04:
I have enabled HDR_Merge to true, set the sequence name to 0, set the sequence size to 2 using ros yaml files.
How can I set exposure and gain values for the two frames in the subpreset sequence i.e subpreset sequence id 1 and sub preset sequenceid 2 using the yaml file.

In the real sense viewer I am able to  set the exposure and gain values for individual sequence ids when HDR Enabled is checked using the slide-bar. I also note that if HDR-enabled is checked in realsense -viewer it automatically deselects auto exposure which kind of makes sense. Should i also explicitly turn off auto exposure in ros yaml file?

Also it would be very helpful if you can guide me where can I set disparity shift , min z, max z values, and depth units in the ros wrapper. I coud not locate them in the rs_camera.launch file neither in the dynamic reconfigure gui.

MartyG-RealSense, 2022-06-04:
Hi @hakunaMatataHub  Thanks for your questions.

Whilst I do not have information about setting up sequence_id in a yaml file, the links below provide advice about setting it in launch file rosparams.  This may provide some useful insights about how to define those instructions in a yaml file instead.

**ROS1**
https://github.com/IntelRealSense/realsense-ros/issues/1657#issuecomment-981866418

**ROS2**
https://github.com/IntelRealSense/realsense-ros/issues/1657#issuecomment-968920694

Please note that the ROS2 advice in the link above was intended for the previous ROS2 wrapper (up to version 3.2.3) rather than the current ros2_beta wrapper (currently version 4.0.4).

******************

You can set the disparity shift by defining its value in a **json** camera configuration file and then loading that json file during launch, as described at https://github.com/IntelRealSense/realsense-ros/issues/981

The depth unit scale cannot be changed in the RealSense ROS wrapper for reasons explained at https://github.com/IntelRealSense/realsense-ros/issues/277#issuecomment-525676873

hakunaMatataHub, 2022-06-04:
@MartyG-RealSense 
Thank you I figured out how to set exposure for individual streams in HDR Merge configuration from the yaml file.

How do I set depth clamp min and depth clamp max in ros wrapper?
Also are these settings coupled with the disparity shift i.e if i incease the disparity shift will the depth clamp max and depth clamp min automatically adjust as per their mathematical relation (with a window of 126 disparity)? If not what benefit does these params have in overal depth quality? What are the guidelines in setting it? 

Atleast I can deduce that the clamp depth min/max values should be within the min and max depth values resulting from the depth-disparity realtion corresponding to a window of 126 disparity, right?

MartyG-RealSense, 2022-06-04:
The RealSense ROS wrapper does not support setting a minimum depth distance.  You can set a maximum distance with the **clip_distance** parameter.  However, if you use rtabmap-ros then you can set a minimum distance with its **RangeMin** parameter and a maximum with **RangeMax**.  More information about this can be found at https://github.com/introlab/rtabmap_ros/issues/413

Disparity Shift is not the same as a depth clamp, which defines the range of distance values that are allowed to be rendered on the depth image and excludes from the image the depth values that are outside that defined min-max range.  

When the disparity shift value is increased then the camera's minimum depth sensing distance decreases, enabling the camera to move closer to the surface of an object before the depth image starts breaking up when the camera moves below minimum distance.  However, the camera's maximum observable depth distance also **decreases** at the same time as disparity shift is increased.

A disparity shift of between 50 and 100 provides a good balance between being able to move the camera closer to a surface and not losing most of the longer-range background detail.  A disparity shift may only improve the depth image in regard to the area of the depth image nearest to the camera being more detailed.  

Unless you are using the min-max of rtabmap_ros's RangeMin and RangeMax though then the only clamp option available to you will be to set the maximum distance with clip_distance.

hakunaMatataHub, 2022-06-05:
@MartyG-RealSense 
If rendering is all that is controlled by the depth clamp min/max parameters, how is it different from the threshold ilter in the post processing block available in the real sense viewer?

MartyG-RealSense, 2022-06-05:
There isn't a difference.  The threshold filter is one of the methods available to implement depth clamping of the image by controlling what distance values are permitted to be included on the image.

hakunaMatataHub, 2022-06-05:
If I point my real-sense towards ceiling of my room with depth unit = 1000, depth clamp min =0 and depth clamp max = 65536 in the depth table under advanced controls, the pixels corresponding to the ceiling in the depth image shows depth of  ~ 2.2 m. If I change only the depth unit to say 4000 the depth for the same camera-scene configuration changes to almost half i.e ~1m and If I decrease it ti say 300  it is beyond 10m for same scene-camera configuration.

Doesn't the depth unit say how many micro-meters does one unit in the 16bit representation of the depth image correspond to? Why does it change the value of the estimated depth?

hakunaMatataHub, 2022-06-05:
[image]
[image]

MartyG-RealSense, 2022-06-05:
Real distance in meters can be obtained by multipying the raw 16-bit depth value by the **depth unit scale**, which on the 400 Series cameras is usually 0.001 (m) by default (except for the D405 model, where it is 0.01 by default).  So if the maximum raw depth value of 65536 is multiplied by 0.001 then that gives a real distance value of 65.536 m.

Most of the Advanced Mode features are undocumented by Intel and do not have explanations of their functions.  This is because Advanced Mode controls interact with one another in complex ways and so Intel chose to control them with machine learning algorithms (whilst allowing users to perform trial and error experimentation with them).  I would therefore recommend leaving the Advanced Mode's Depth Units setting alone and instead controlling scale with the **Depth Unit Scale** option under Stereo Module > Controls

[image]

hakunaMatataHub, 2022-06-05:
@MartyG-RealSense 
But changing either of them changes the other one as well

MartyG-RealSense, 2022-06-05:
I tested that and you are correct, changing the Controls > Depth Units does automatically change the Advanced Mode > Depth Units.  This is an example of how the undocumented nature of Advanced Mode functions can lead to surprises even for expert users.  :)

Two RealSense team members at https://github.com/IntelRealSense/librealsense/issues/5487 explain how the two modes are interconnected and like I did, they recommend using the Controls version of the setting instead of Advanced Mode.

hakunaMatataHub, 2022-06-05:
@MartyG-RealSense 
Yeah thank you it is clear now.

One last thing, I am not able to see meta data such as sequence id and exposure in the realsense -viewer?
The following image is all I get:
[image]

MartyG-RealSense, 2022-06-05:
When auto-exposure is enabled, the exposure value does not update and so the **Actual Exposure** metadata value represents the exposure value.

https://github.com/IntelRealSense/librealsense/pull/7404/commits/9ec511b60d41986c250dc6152dceeb2c4ce98c06 indicates that HDR metadata values are available for sequence size, sequence ID and sequence name.

As the Viewer's metadata overlay was programmed a long time before the HDR mode was introduced, it is likely that the selection of metadata options displayed in the overlay is fixed in the Viewer's code and so does not contain the HDR metadata, which should be able to be retrieved through scripting when creating your own program.

http://docs.ros.org/en/kinetic/api/librealsense2/html/rs__frame_8h.html#a91f19a01f5bf2abadc30959a8d3109c9a4ffc1611317e82bfa633ade0de8dc05c

hakunaMatataHub, 2022-06-05:
@MartyG-RealSense 
In this[ white paper](https://dev.intelrealsense.com/docs/high-dynamic-range-with-stereoscopic-depth-cameras) they seem to imply that it should be available in RS-viewer.
https://dev.intelrealsense.com/docs/frame-metadata
 this link in the above document talks about some patches for linux uvcvideo driver on for ubuntu 14 /16 LTS but nothing about 20.04 LTS which is my configuration.

Does this mean It is not supported in 20.04 LTS?

MartyG-RealSense, 2022-06-05:
References to frame metadata in the SDK documentation usually refer to the original metadata types that are are listed in the metadata overlay in the Viewer.

Kernel patching support for 20.04 (Focal) and kernel 5.8 was added in SDK 2.45.0 - as detailed at https://github.com/IntelRealSense/librealsense/pull/8787 - and a patch was added for Focal and kernel 5.11 in SDK 2.50.0, as detailed at https://github.com/IntelRealSense/librealsense/pull/9727

The patching process should add hardware metadata support to a source code build of librealsense.  Patching is not necessary when building from Debian packages or from source code with the RSUSB backend installation method.

hakunaMatataHub, 2022-06-05:
I have debian installation for llibrealsense sdk. If I got you right I don't need to worry about patching my kernel, correct?

I didn't get the first part of your comment, do you  mean the highlighted metadata type is not one of original metadata type?
[image]

MartyG-RealSense, 2022-06-05:
If you are installing the SDK with Debian packages then you do not need to patch, as the patch is included in the package.

What I mean about the Real Exposure metadata type is that because the exposure value does not update whilst auto-exposure is enabled, you cannot read the exposure unless you disable auto-exposure or you obtain the 'Real Exposure' metadata value instead of querying the exposure value directly.

hakunaMatataHub, 2022-06-05:
@MartyG-RealSense 
My problem is I do not have exposure meta data at all overlayed on the infrared image in realsense-viewer irrespective of whether I enable or disable auto-exposure
[image]

MartyG-RealSense, 2022-06-05:
I was able to duplicate your metadata readout with a D435i where support for hardware metadata was not enabled.

[image]

hakunaMatataHub, 2022-06-05:
I just checked my kernel is 5.13.0-44-generic.

So do you recommend building it from source for hardware metadata support?
@MartyG-RealSense

MartyG-RealSense, 2022-06-05:
Yes, I would recommend a source code build.  If you do not want to change your kernel then you could try building the SDK from source code with the **RSUSB backend** installation method.  This install method is compatible with 20.04.  It bypasses the Linux kernel and so is not dependent on Linux versions or kernel versions and does not require patching.  Hardware metadata support is provided automatically in an RSUSB build without having to patch it in.

hakunaMatataHub, 2022-06-05:
@MartyG-RealSense 
Thank you I will try this out

hakunaMatataHub, 2022-06-06:
@MartyG-RealSense 

I have the following parameters set in a yaml filew which is uploaded to rosparam server from rs_camera.launch file,
I am unable to get hdr_enabled. I say this beacuse I don't see flickering in the infrared stream which should happen because I have set the exposure to very differen values for each seq id in hdr. I have also enabled hdr_filter in the rs_camera.launch file.

-------------------------------------------------------------------------------------------------------
This is rs_camera_config.yaml

/camera/temporal/filter_smooth_alpha: 0.1
/camera/temporal/filter_smooth_delta: 100
/camera/temporal/frames_queue_size: 1
/camera/temporal/steam_filter: 1
/camera/temporal/holes_fill: 1
/camera/spatial/frames_queue_size: 10
/camera/spatial/filter_smooth_alpha: 0.28
/camera/spatial/filter_smooth_delta: 17
/camera/decimation/frames_queue_size: 15
/camera/decimation/filter_magnitude: 2
/camera/disparity/frames_queue_size: 15
/camera/stereo_module/enable_auto_exposure: false
/camera/spatial/holes_fill: 3 
/camera/stereo_module/hrd_enabled: true
/camera/stereo_module/sequence_id: 2
/camera/stereo_module/exposure/2: 1000
/camera/stereo_module/sequence_id: 1
/camera/stereo_module/exposure/1: 1
/camera/stereo_module/emitter_on_off: false
/camera/stereo_module/laser_power: 150.0

----------------------------------------------------------------------------------------------------------------------------
This is rs_camera.launch file

<launch>
  <arg name="respawn"           default="true"/>
  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default="/home/mowito/ros1_ws/src/vision_executive/config/middensitypreset.json"/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
  <arg name="output"              default="screen"/>

  <arg name="fisheye_width"       default="-1"/>
  <arg name="fisheye_height"      default="-1"/>

  <arg name="fisheye_fps"         default="50"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="848"/>
  <arg name="depth_height"        default="480"/>
  <arg name="depth_fps"           default="30"/>  
  <arg name="enable_depth"        default="true"/>
  

  <arg name="color_width"         default="848"/>
  <arg name="color_height"        default="480"/>
  <arg name="color_fps"           default="30"/>
  
  <arg name="enable_color"        default="true"/>

  <arg name="confidence_width"    default="848"/>
  <arg name="confidence_height"   default="480"/>
  <arg name="enable_confidence"   default="false"/>

  <arg name="confidence_fps"      default="15"/>

  <arg name="infra_width"         default="848"/>
  <arg name="infra_height"        default="480"/>
[... 142 lines cut ...]
    "param-regioncolorthresholdb": "0.0136986",
    "param-regioncolorthresholdg": "0.707436",
    "param-regioncolorthresholdr": "0.181996",
    "param-regionshrinku": "3",
    "param-regionshrinkv": "1",
    "param-robbinsmonrodecrement": "23",
    "param-robbinsmonroincrement": "3",
    "param-rsmdiffthreshold": "1.8125",
    "param-rsmrauslodiffthreshold": "1",
    "param-rsmremovethreshold": "0.482143",
    "param-scanlineedgetaub": "16",
    "param-scanlineedgetaug": "259",
    "param-scanlineedgetaur": "896",
    "param-scanlinep1": "132",
    "param-scanlinep1onediscon": "77",
    "param-scanlinep1twodiscon": "234",
    "param-scanlinep2": "342",
    "param-scanlinep2onediscon": "390",
    "param-scanlinep2twodiscon": "151",
    "param-secondpeakdelta": "600",
    "param-texturecountthresh": "0",
    "param-texturedifferencethresh": "0",
    "param-usersm": "1",
    "param-disparityshift": "20"
}

hakunaMatataHub, 2022-06-06:
@MartyG-RealSense 
If I unset hdr_merge filter from rs_camera.launch and I make a coomand from terminal i.e
"rosrun dynamic_reconfigure dynparam set /camera/stereo_module hdr_enabled true"  as suggested [here](https://github.com/IntelRealSense/realsense-ros/issues/1657#issuecomment-983345133)
I can see the flickering.

But I would also need the hdr_merge filter to be turned on for improving my depth.
Please help.

MartyG-RealSense, 2022-06-06:
My understanding is that in the ROS1 wrapper you can enable hdr_merge in the roslaunch instruction as a filter with **filters:=hdr_merge**

For example:

`roslaunch realsense2_camera rs_camera.launch filters:=hdr_merge`

If you needed to use more than one filter, such as pointcloud and hdr_merge, then you can separate the filters with a comma with no space between them:

`roslaunch realsense2_camera rs_camera.launch filters:=pointcloud,hdr_merge`

The hdr_merge instruction is listed within the **filters** section of the RealSense ROS wrapper documentation, confirming that it should be handled as a filter type.

**********

Based on the information in https://github.com/IntelRealSense/realsense-ros/issues/1657#issuecomment-981866418 the rosrun equivalents of the HDR settings may look something like the instructions below if you wish to apply the settings during runtime after launch has completed.

rosrun dynamic_reconfigure dynparam set /camera/stereo_module hdr_enabled true /camera//stereo_module/filter_by_sequence_id/enable: true /camera/stereo_module/sequence_id: 2 /camera/stereo_module/exposure/1 7500 /camera/stereo_module/exposure/2 8500 

I include the exposure/1 and exposure/2 commands as an example of setting values for 2 different exposures, though you may only need to define exposure/2

hakunaMatataHub, 2022-06-06:
@MartyG-RealSense 
I guess you didn't get my question correct.

I am able to turn on hdr_merge filter from within rs_camera.launch file along with spatial, temporal and decimation filters.

My issue is It does not turn on hdr beacuse I can not see any flickering when i visualisize the infrared images in rqt_image_view.

MartyG-RealSense, 2022-06-06:
The purpose of the rosrun command that I listed was to act as an example of how to define all the instructions associated with HDR in rosrun, not necessarily to use them all in the same instruction.

There is little information available about using the HDR commands in the RealSense ROS wrapper, so I have attempted to provide as much as I can of the information that is available in order to help you to work out the commands that you need to use to achieve the desired effect.  HDR configuration in the ROS wrapper is not a subject that I am familar with though, so there is a limit to the advice that I can provide on the subject unfortunately.

hakunaMatataHub, 2022-06-06:
@MartyG-RealSense 
Thank you for clarifying . Can someone else from  your team help me out please?

MartyG-RealSense, 2022-06-06:
I have referred your question to the RealSense ROS development team.

hakunaMatataHub, 2022-06-07:
@MartyG-RealSense 
Any update please?

hakunaMatataHub, 2022-06-07:
@MartyG-RealSense 
I had a debian installation for realsense-ros, so I source build it and compiled the development branch.
This time I see this warning on the terminal,

```
WARNING [140538533295872] (hdr-merge.cpp:38) HDR Merge filter cannot process frames because relevant metadata params are missing

```

MartyG-RealSense, 2022-06-07:
I have not received a response yet to my referral to Intel of your question.  I will update you when I do.  Thanks very much for your patience.

As you have performed a Debian installation and then a source code build on the same computer, this can result in conflicts such as **Multiple udev rules**.  You should be able to completely remove Debian packages related to librealsense using the command below:

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

hakunaMatataHub, 2022-06-07:
@MartyG-RealSense  I did source installation only for ros wrapper
I will be waiting for your update.

hakunaMatataHub, 2022-06-09:
@MartyG-RealSense 
can you please check for any update?

MartyG-RealSense, 2022-06-09:
I have not received a response yet.  I promise that I will update when I do.

MartyG-RealSense, 2022-06-14:
Hi @hakunaMatataHub  Intel concluded their discussion of the issue that you reported.  They decided that the non-flickering on the infrared image when visualizing the infrared images in rqt_image_view was a valid concern and an official bug report for the issue was created so that it can be investigated by Intel.

hakunaMatataHub, 2022-06-15:
@MartyG-RealSense 
Is the bug report in form of github issue which I can track? How would I know its progress or when it is fixed?

MartyG-RealSense, 2022-06-15:
No, the reports are internal and not publicly viewable or trackable.  If a fix is implemented then it is typically listed in release notes such as those for librealsense or the ROS wrapper.

I added an **Enhancement** label to this case, indicating that the case should be kept open until the bug report is concluded.  When a report is closed, the Enhancement label is removed from the case.  So you can tell at a glance if the bug report is still open by looking at the list of labels in the side panel at the top of the case.

SamerKhshiboun, 2023-01-24:
Hi @MartyG-RealSense @hakunaMatataHub,

I've added hdr_merge.enable and depth_module.hdr_enabled to the rs_launch.py file (ros2-development branch)
You can take a look on the launch file in this PR #2603 
Now, the filter should work correctly, and you can see the IR is flicking in the rqt_image_view

`ros2 launch realsense2_camera rs_launch.py depth_module.hdr_enabled:=true enable_infra1:=true hdr_merge.enable:=true`

Let me know if you need further assistance.

MartyG-RealSense, 2023-01-24:
Excellent, thanks so much @SamerKhshiboun  :)

@hakunaMatataHub  Can you confirm please whether the amended  rs_launch.py file resolves the issue for you, please?  Thanks!

MartyG-RealSense, 2023-02-05:
As the hdr_merge filter change has been merged into the ROS wrapper, I will close this case.  Thanks!

closed 2023-02-05 by MartyG-RealSense