realsenseai/realsense-ros #2367: Min Z  specifications for D405

hakunaMatataHub, 2022-06-05:
if we go by the formulae

Min Z = (focal length (in pixel) * baseline )/126

Min Z for D405 at 480p resolution would be 0.036 m or 3.6 cm.

But the [specification](specification) says it is 7cm at 480p resolution does that mean the disparity window is about ~65 pixels instead of 126 .

MartyG-RealSense, 2022-06-05:
The values for the D401 (the camera board inside the D405 camera model) are baseline **18 mm** (from the data sheet) and the rs-enumerate-devices tool gives focal length fx / fy values at 640x480 (480p) of around **382.259** on my particular D405.  Are these the approximate values that you used in your calculation, please?

[image]

I have previously performed tests with the D405 that confirmed that 7 cm was the approximate minimum distance that could be achieved before the depth image began breaking up when moving the camera closer to a surface than that, and that using disparity shift to attempt to reduce minimum distance further made no difference on that camera model.

On the D405, the default depth unit scale value is 0.01 instead of the 0.001 that is the default on other 400 Series models.

A RealSense team member provides advice about disparity, including the formula that you quoted, at https://github.com/IntelRealSense/librealsense/issues/3039#issuecomment-456975023

hakunaMatataHub, 2022-06-05:
@MartyG-RealSense 
I used HFOV as 87 degrees as said [here ](https://www.intelrealsense.com/depth-camera-d405/)
and I mistakenly used 480 pixels instead of 640 pixels to calculae focal length using,
f = Xres/(2*tan(HFOV/2))
But even if i plug in your values i.e focal length of 382.259 for 640 pixels Xres and 79.87 degrees HFOV, I do not get 7cm as Minz but 5.4 cm approx.

hakunaMatataHub, 2022-06-05:
@MartyG-RealSense 
Min Z comes to 7cm if I use the following

Min Z = (focal length (in pixel) * baseline )/98

This would make sense if the disparity window in D405 would be about 100 pixels

MartyG-RealSense, 2022-06-05:
I am admittedly not familiar enough with the subject of disparity mathematics to comment on the figures.  My experience with practical tests though was that the achievable minimum in real-world testing was 7 cm.

closed 2022-06-17 by MartyG-RealSense