realsenseai/realsense-ros #2368: D435 camera strange phenomenon, a point cloud map

xiaopgaxiaopg, 2022-06-06:
Under the system of 18.04, gazebo simulation D435 camera released point cloud data, the phenomenon is as follows, where the red box is abnormal, I converted the point cloud into Anotomap, the phenomenon was completely exposed, I want to seek help, how to remove the point cloud map in the red box in the point cloud environment, very much need your help, I have been stuck for a week。
[image]
[image]
[image]

MartyG-RealSense, 2022-06-06:
Hi @xiaopgaxiaopg  Is the 'wall' that you have marked with the block an area that exists in the real world or is it false depth data representing something that is not actually there, please?

xiaopgaxiaopg, 2022-06-06:
This is the scene simulated in gazebo, not using the real d435, the scene in the picture below, this wall does not exist

---- Replied Message ----
| From | ***@***.***> |
| Date | 06/06/2022 20:27 |
| To | ***@***.***> |
| Cc | ***@***.******@***.***> |
| Subject | Re: [IntelRealSense/realsense-ros] D435 camera strange phenomenon, a point cloud map (Issue #2368) |

Hi @xiaopgaxiaopg Is the 'wall' that you have marked with the block an area that exists in the real world or is it false depth data representing something that is not actually there, please?

MartyG-RealSense, 2022-06-06:
I am not able to test customer source code on my computer, unfortunately.

If there is an area of depth detail that does not correspond to the real life scene then it may be because of a phenomenon called 'ghost noise'.  This may be because the camera may have been confused by some of the objects or surfaces in the real life scene.  Are you able to provide an RGB image of the scene so that I can see whether there are elements present that may be difficult for the camera to read accurately, please?

xiaopgaxiaopg, 2022-06-06:
> I am not able to test customer source code on my computer, unfortunately.
>
> If there is an area of depth detail that does not correspond to the real life scene then it may be because of a phenomenon [...]

I need to save the scene in this picture, the projected image in this picture is projected by the rgb camera and can be closed, I tested. I suspect that the depth camera also carried out projection, but I didn't find the place to close, which is just a guess
[image]
<sensor name="${camera_name}color" type="camera">
        <camera name="${camera_name}">
          <horizontal_fov>${69.4*deg_to_rad}</horizontal_fov>
          <image>
            <width>1920</width>
            <height>1080</height>
            <format>RGB_INT8</format>
          </image>
          <clip>
            <near>0.1</near>
            <far>100</far>
          </clip>
          <noise>
            <type>gaussian</type>
            <mean>0.0</mean>
            <stddev>0.007</stddev>
          </noise>
        </camera>
        <always_on>1</always_on>
        <update_rate>30</update_rate>
        <visualize>0</visualize>
        
      </sensor>

xiaopgaxiaopg, 2022-06-06:
> I am not able to test customer source code on my computer, unfortunately.
>
> If there is an area of depth detail that does not correspond to the real life scene then it may be because of a phenomenon [...]

In fact, the problem of reproducing is not complicated, just need to use the D435 camera to collect depth data in the gazebo environment, publish the depth topic /depth/point, and then view the topic of using pointcloud2 in Rviz

MartyG-RealSense, 2022-06-06:
Is the problem that the projected RGB image is appearing on the depth image and you are unable to close it?

xiaopgaxiaopg, 2022-06-07:
> Is the problem that the projected RGB image is appearing on the depth image and you are unable to close it?

I don't know, I can't determine the cause, I decided to change the camera

xiaopgaxiaopg, 2022-06-07:
> Is the problem that the projected RGB image is appearing on the depth image and you are unable to close it?

Dear sir, I have a new idea, I feel that it should be the point cloud within the effective detection range of the clip parameter control, the point cloud projection perpendicular to the camera lens plane, that is, this projection is parallel to the camera plane, but I don't know how to turn off this projection, clip this parameter I can only search the farthest and closest distance at the moment, but I have not found the parameter that controls the projection

xiaopgaxiaopg, 2022-06-07:
> Is the problem that the projected RGB image is appearing on the depth image and you are unable to close it?

[image]
kinect v2

MartyG-RealSense, 2022-06-07:
I researched your case again carefully.  I wonder whether your situation is similar to a MoveIt tutorial in the link below that shows how to extract a cylinder from a pointcloud and remove the points of the surface on which the pointcloud is resting.

https://ros-planning.github.io/moveit_tutorials/doc/perception_pipeline/perception_pipeline_tutorial.html#detecting-and-adding-object-as-collision-object

xiaopgaxiaopg, 2022-06-07:
> I researched your case again carefully. I wonder whether your situation is similar to a MoveIt tutorial in the link below that shows how to extract a cylinder from a pointcloud and remove the points [...]

> I researched your case again carefully. I wonder whether your situation is similar to a MoveIt tutorial in the link below that shows how to extract a cylinder from a pointcloud and remove the points [...]

Dear sir, the function I want to do is indeed the function in this tutorial, the similar to the wall in the figure I do not know how he appeared, looking at the phenomenon is indeed similar, in the previous message, I changed the camera still appeared the phenomenon of d435, I just contacted my friend, and provided the source code, please give them a test, will not appear my same phenomenon. Reply to mr. later. If it is convenient, I am willing to provide the source code of github's kinect, please sir run it personally.
https://github.com/wangxian4423/kinect_v2_udrf
The code needs to be tweaked, I have a good adjustment here, I hope Mr. provides a mailbox

MartyG-RealSense, 2022-06-07:
I am not able to run code or tests or use ROS on my computer, unfortunately.  I can only advise about how to adjust your own system where possible.  My knowledge of Gazebo is very limited though so I can only point you to other references in the hope that they will be helpful.

https://github.com/ros-planning/moveit/issues/2374 is another example of problems with extracting a cylinder from a pointcloud with MoveIt.  Do you get any error messages in your terminal like the ones in this link?

Of the few cases I have seen in my research where people were trying to remove the pointcloud, they all seem to have experienced problems with it that they could not find a solution to.

xiaopgaxiaopg, 2022-06-09:
> I am not able to run code or tests or use ROS on my computer, unfortunately. I can only advise about how to adjust your own system where possible. My knowledge of Gazebo is very limited though so in this particular subject I can only point you to other references in the hope that they will be helpful.
> 
> [ros-planning/moveit#2374](https://github.com/ros-planning/moveit/issues/2374) is another example of problems with extracting a cylinder from a pointcloud with MoveIt. Do you get any error messages in your terminal like the ones in this link?
> 
> Of the few cases I have seen in my research where people were trying to remove the pointcloud, they all seem to have experienced problems with it that they could not find a solution to.

Dear sir, my problem was solved, it should be a system problem, I reinstalled the ros system of 18.04 and everything became normal, thank you sir for your support and help these days.

closed 2022-06-15 by MartyG-RealSense