realsenseai/realsense-ros #2371: Modify depth step-size from roslaunch

Camilochiang, 2022-06-07:
Is it possible to change the depth step-size using roslaunch? Also. Is it possible to change Disparity shift using roslaunch ? . Thanks!

Camilochiang, 2022-06-07:
if im not wrong is hard coded in `base_realsense_node.cpp`, no @MartyG-RealSense ?
```
cv::Mat& BaseRealSenseNode::fix_depth_scale(const cv::Mat& from_image, cv::Mat& to_image)
{
    static const float meter_to_mm = 0.001f;
    if (fabs(_depth_scale_meters - meter_to_mm) < 1e-6)
    {
        to_image = from_image;
        return to_image;
    }
`
    if (to_image.size() != from_image.size())
    {
        to_image.create(from_image.rows, from_image.cols, from_image.type());
    }
`
    CV_Assert(from_image.depth() == _image_format[RS2_STREAM_DEPTH]);

    int nRows = from_image.rows;
    int nCols = from_image.cols;

    if (from_image.isContinuous())
    {
        nCols *= nRows;
        nRows = 1;
    }

    int i,j;
    const uint16_t* p_from;
    uint16_t* p_to;
    for( i = 0; i < nRows; ++i)
    {
        p_from = from_image.ptr<uint16_t>(i);
        p_to = to_image.ptr<uint16_t>(i);
        for ( j = 0; j < nCols; ++j)
        {
            p_to[j] = p_from[j] * _depth_scale_meters / meter_to_mm;
        }
    }
    return to_image;
}`

MartyG-RealSense, 2022-06-07:
Hi @Camilochiang  As described at https://github.com/IntelRealSense/realsense-ros/issues/981 you can set disparity shift in the RealSense ROS wrapper by defining it in a **json** camera configuration file and then loading the json file with the roslaunch instruction or by defining the path to it inside the launch file with the **json_file_path** parameter.

https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/rs_camera.launch#L5

The depth unit scale cannot be changed in the ROS wrapper for reasons described at https://github.com/IntelRealSense/realsense-ros/issues/277#issuecomment-525676873

Camilochiang, 2022-06-07:
thanks for the quickly answer. Please note that im asking for changing the step-size and not the scale of depth, as recomend it [here](https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance)

closed 2022-06-07 by Camilochiang