realsenseai/realsense-ros #2372: How to get the current exposure value of the camera auto exposure mode with code?

Zoltan3057, 2022-06-07:
Hi,
Is there a way to receive current exposure value just like tum mono dataset through ros. 
https://vision.in.tum.de/mono-dataset
Thanks in advance.

MartyG-RealSense, 2022-06-07:
Hi @Zoltan3057  When depth auto-exposure is enabled, the exposure value is not updated.  You can though obtain the "real" exposure value from camera **metadata**.  

The publishing of metadata topics was added to the RealSense ROS wrapper in ROS1 wrapper 2.3.2 and ROS2 wrapper **3.2.3**.  The ROS2 beta wrapper (currently version **4.0.4**) also publishes the metadata topics.   Please use the command **rostopic list** for ROS1 or **ros2 topic list** for ROS2 to list the rostopics that are published for your camera to check that the metadata topics are available. 

If you require RGB exposure metadata, this information is not available for the RGB stream when auto-exposure is enabled, as advised by a RealSense team member at https://github.com/IntelRealSense/librealsense/issues/2549#issuecomment-431863920

Zoltan3057, 2022-06-08:
Thanks for the reply.
Unfortunately, I need the exposure time for RGB data. Since the exposure metadata for RGB is not available, I tried to set auto-exposure to false. However, I cannot find a method to set it in D435i by ROS.

Zoltan3057, 2022-06-08:
I find setting json_file_path parameter in launch to realsense-viewer's preset json file can set the exposure time constant.  Thanks.

closed 2022-06-08 by Zoltan3057