realsenseai/realsense-ros #2373: SLAM with D455 issue running it.

remilamoureuxlevesque, 2022-06-08:
I have some issues following the [SLAM with D435i](https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i). The installation when well, but my issue are in the running part. 

My platform is a Jetson nano developpement board with Ubuntu 18.04 and ROS melodic running on it. 
My sensor is D455 RGBD camera.

My first issue is when I launch the rviz the map topics is not present. I got the following error : Fixed Frame [map] does not exist.
But I am still able to see the point cloud when I change de Fixed Frame to : camera_aligned_depth_to_color_frame and clic on the PointCloud2 checkbox.

My second issue is when I start the recording with
```
rosbag record -O my_bagfile_1.bag /camera/aligned_depth_to_color/camera_info  camera/aligned_depth_to_color/image_raw /camera/color/camera_info /camera/color/image_raw /camera/imu /camera/imu_info /tf_static
```

while rviz is running I don't see my record updating the map in real time, the only thing I see is what the camera is recording now (only the frame that she get). But I achived to generate a bag file with some information.

My last issue is when I try to replay the bag file, I did not see anything in rviz, but when I run
```
rosbag play my_bagfile_1.bag --clock
```
I see the content being display in the terminal.

Hope you can help me with my issues.

MartyG-RealSense, 2022-06-08:
Hi @remilamoureuxlevesque  Some time after Intel's D435i SLAM guide was published, the gyro and accel IMU topics were disabled by default in the RealSense ROS wrapper.  Please try enabling those topics in the roslaunch instruction to see whether this resolves your Fixed Frame issue.

`roslaunch realsense2_camera opensource_tracking.launch enable_gyro:=true enable_accel:=true`

remilamoureuxlevesque, 2022-06-09:
Hi @MartyG-RealSense,

Thanks for your answer, this solve my first issue. When I try to record I the map is still not updating in real time, the only thing I can see is the camera live feed. 

When trying to recored I do that : 
1. Start rviz with ```roslaunch realsense2_camera opensource_tracking.launch enable_gyro:=true enable_accel:=true```
2. Then start the record with ```rosbag record -O my_bagfile_1.bag /camera/aligned_depth_to_color/camera_info  camera/aligned_depth_to_color/image_raw /camera/color/camera_info /camera/color/image_raw /camera/imu /camera/imu_info /tf_static```
3. Stop the record process with ```Ctrl+C```.
4. Stop rviz by clossing the window and ```Ctrl+C``` in the terminal.

Then I get a bag file of about 2GB.

Atfer that I try to replay the bagfile by following the wiki page for the SLAM with D435i and I dont see anything on rviz.

Let me know if I do something wrong.

MartyG-RealSense, 2022-06-10:
When you replayed the bag file, did you again enable the IMU topics?

`roslaunch realsense2_camera opensource_tracking.launch offline:=true enable_gyro:=true enable_accel:=true`

remilamoureuxlevesque, 2022-06-10:
Yes I did enable the gyro and the accel and still have the same result. I also got an error when I launch rviz. 

[image]

And this is what I got when I run : 
```roslaunch realsense2_camera opensource_tracking.launch offline:=true enable_gyro:=true enable_accel:=true```

[image]

MartyG-RealSense, 2022-06-10:
And you entered the full list of commands for bag file playback in the D435i SLAM guide including the roscore and rosparam lines, please?

```
roscore >/dev/null 2>&1 &
rosparam set use_sim_time true
rosbag play my_bagfile_1.bag --clock
roslaunch realsense2_camera opensource_tracking.launch offline:=true enable_gyro:=true enable_accel:=true
```

remilamoureuxlevesque, 2022-06-10:
Yes I did all the command one at the time and I still have the same result.

[image]

MartyG-RealSense, 2022-06-11:
In the **Frames** section of RViz's **TF** category, have you followed the D435i SLAM guide's instructions to have everything unticked except 'map' and 'camera_link'

[image]

remilamoureuxlevesque, 2022-06-13:
[image]

I don't have anything in the frames tab.

remilamoureuxlevesque, 2022-06-13:
Still the same result
[image]

MartyG-RealSense, 2022-06-13:
I went through your bag record and replay process again.  It seems to be fine up uptil the point in https://github.com/IntelRealSense/realsense-ros/issues/2373#issuecomment-1152364600 where you launch RViz and receive the mass of red errors.

[image]

So you performed an offline roslaunch using the bag file with this command?

**roslaunch realsense2_camera opensource_tracking.launch offline:=true enable_gyro:=true enable_accel:=true**

In the **opensource_tracking.launch** launch file, my recollection is that RViz opens automatically.  Is that correct, please?  And if yes, the red errors occur after RViz automatically opens and uses the bag file as its offline data source instead of a live camera, suggesting that it cannot visualize the bag contents?

Thanks very much for your patience!

remilamoureuxlevesque, 2022-06-13:
Yes, everything you mention seem to be true. I also got this error when I launch the command to launch rviz and try to see the pointcloud.

```roslaunch realsense2_camera opensource_tracking.launch enable_gyro:=true enable_accel:=true```

The map attribute not always show in rviz when I launch the command.

MartyG-RealSense, 2022-06-14:
The RealSense ROS wrapper's pointcloud filter is apparently disabled by default in the D435i SLAM project, so I would guess that the pointcloud is being generated by **rtabmap_ros** instead.  

It would be helpful as a diagnostic method to try running the ROS wrapper's simpler pointcloud generation example to provide insight about whether a pointcloud can be generated in RViz at all on your Jetson.  The procedure would be:

1.  Generate a pointcloud with the **rs_camera** launch file with the roslaunch instruction below.

**roslaunch realsense2_camera rs_camera.launch filters:=pointcloud**

2.  Input **rosrun rviz rviz** into the ROS terminal to launch RViz and visualize the pointcloud.

[image]

https://github.com/IntelRealSense/realsense-ros#point-cloud

Jetson devices specifically have problems with generating a pointcloud or performing depth-color alignment in modern ROS wrapper versions - as described in detail at https://github.com/IntelRealSense/realsense-ros/issues/1967 - with the current best solution for ROS1 being to use version **2.43.0** of librealsense built with CUDA support and version **2.2.23** of the ROS wrapper.

MartyG-RealSense, 2022-06-20:
Hi @remilamoureuxlevesque  Do you require further assistance with this case, please?  Thanks!

remilamoureuxlevesque, 2022-06-20:
Yeah sorry, I did not have a lot of time the last few days to work on it. 

I try to the ros wrapper point cloud example you mention and this is what I got :
I have had the pointcloud2 and tf in rviz to visuialize them.

[image]

I also got this warning when running the command : 
```roslaunch realsense2_camera rs_camera.launch filters:=pointcloud```
The warning 

[image]

I will try to run it with the version you mention (ROS 2.2.23 and librealsense 2.43.0)

I will update you when it will be done.

MartyG-RealSense, 2022-06-20:
Thanks very much for the update.  The **Resource temporarily unavailable** type of control_transfer warning can indicate a communication problem between the camera and computer.= if the warning is being generated continuously instead of occurring several times and then stopping.

closed 2022-07-02 by MartyG-RealSense