realsenseai/realsense-ros #2374: Updating parameters for resolution and frame rate for D435i in ROS 2 Galactic doesn't work

hello-chintan, 2022-06-09:
Hello,

I want to run the depth and RGB modules for the D435i camera with 1280x720 resolution at 15 fps. I am trying to set the sensor parameters through a .py launch file in ROS 2 Galactic but it doesn't seem to be working. According to the documentation, **It is a string of the following format: <width>X<height>X<fps> (The dividing character can be X, x or ",". Spaces are ignored.)**. I tried updating the parameters to the launch file from the CLI as well as mentioned **ros2 launch realsense2_camera rs_launch.py depth_module.profile:=1280x720x15** but the camera still works with the default resolution and frame rate. Am I doing something wrong?

Attaching the link to the launch file here for reference:
https://github.com/hello-robot/stretch_ros2/blob/feature/calibration/stretch_core/launch/d435i_basic.launch.py (trying to switch to 15 fps here)
https://github.com/hello-robot/stretch_ros2/blob/feature/calibration/stretch_core/launch/d435i_high_resolution.launch.py (trying to switch resolution and frame rate here)

Appreciate the help! Thanks!

MartyG-RealSense, 2022-06-09:
Hi @hello-chintan  Please try changing 15 FPS to 6.  This should confirm whether or not your camera is being detected as being on a USB 2 connection instead of USB 3.  This is because on a USB 2 connection the 1280x720 resolution only supports 6 FPS on D435i.  If a custom stream configuration is detected by the launch as being invalid then it uses the camera's default configuration instead.

hello-chintan, 2022-06-09:
@MartyG-RealSense Thanks for the quick follow-up. I tried lowering the frame rate to 6 fps without any change in the end result. The depth module still runs with default settings. Attaching the log here:
https://gist.github.com/hello-chintan/315c9a34018672b9f2b89e5a04b08ce0

MartyG-RealSense, 2022-06-10:
I note from the log that you are using ROS2 wrapper 3.2.3.  The **depth_profile** featurewas introduced in the newer **ros2_beta** wrapper, the current version of which is **4.0.4**.  In the older **ros2** branch that you are apparently using, it had a different method of setting a custom camera configuration where you must specify three factors for each stream type used (stream width, height and FPS).  If those three factors are not provided then the custom configuration is deemed invalid and the camera's default stream settings are applied by the launch instead.

In the 3.2.3 wrapper, a depth profile of 1280x720x15 would be configured with the ros2 launch instruction below.

`ros2 launch realsense2_camera rs_launch.py depth_width:=1280 depth_height:=720 depth_fps:=15.0`

Whilst here is an example of setting depth and color at 1280x720 and 15 FPS:

`ros2 launch realsense2_camera rs_launch.py depth_width:=1280 depth_height:=720 depth_fps:=15.0 color_width:=1280 color_height:=720 color_fps:=15.0`

hello-chintan, 2022-06-13:
Ah! Thanks for catching that. Since I am working with ROS 2 Galactic I'll need to update the realsense2_camera package and then try to reconfigure the camera using the new depth_module.profile parameter. I'll report back if I run into any issues. Closing this issue for now. Thanks again!

closed 2022-06-13 by hello-chintan

MartyG-RealSense, 2022-06-14:
You are very welcome.  Thanks very much for the update and good luck!