realsenseai/realsense-ros #2375: Turn off low light compensation?

kfu02, 2022-06-09:
With autoexposure on and the cameras running in bright sunlight, I sometimes have the exact issue described here, where autoexposure increases the gain causing oversaturation of the color image:
https://support.intelrealsense.com/hc/en-us/community/posts/360033445713-D435-Sunlight-saturation-problem

I have tried the fixes given at the bottom and they do not resolve this issue. However, the metadata shows that the `low_light_compensation` is on when the gain is unnecessarily bumped up, but is never on when the image is fine. Is there a way to manually ensure `low_light_compensation` is always off?

MartyG-RealSense, 2022-06-10:
Hi @kfu02  The option that likely corresponds to 'low light compensation' is **color backlight compensation**, whose on / off status can be controlled with the instruction RS2_OPTION_BACKLIGHT_COMPENSATION or by enabling / disabling the Backlight Compensation setting in the **RGB** section of the RealSense Viewer's options side-panel.

https://intelrealsense.github.io/librealsense/doxygen/rs__option_8h.html#a8b9c011f705cfab20c7eaaa7a26040e2a1b190100d8ca308207b8661d5cd7e049

[image]

Whilst there is not an existing example of controlling this setting with scripting, as it is an RS2_OPTION setting it could likely be disabled with code such as the C++ scripting below to access the RGB sensor.

```
auto color_sensor = selected_device.first<rs2::color_sensor>();
if (color_sensor.supports(RS2_OPTION_BACKLIGHT_COMPENSATION ))
{
color_sensor.set_option(RS2_OPTION_BACKLIGHT_COMPENSATION , 0.f); // Disable backlight compensation
}
```

*************

If you have a D455 camera model like the RealSense user in the linked case and are experiencing RGB sensor saturation, Intel recommend in their official guidance about this D455-specific issue that as well as using an ND filter you could also try increasing the FPS speed to see whether it helps.

The official guidance for D455 RGB sensor saturation is on page 8 (the final page) of a PDF document at the link below.

https://www.intelrealsense.com/download/13629/

Setting a Region of Interest (ROI) in the lower half of the image can also help to avoid over-exposure in scenes where the camera is directly observing bright sunlight.

[image]

kfu02, 2022-06-13:
rqt_reconfigure does not show backlight compensation or low light compensation under rgb_camera. I will try changing the ROI instead. Thanks!

closed 2022-06-26 by MartyG-RealSense