realsenseai/realsense-ros #2376: [camera/realsense2_camera_manager-2] process has died Cam died after running ros

tobiasonkes, 2022-06-11:
Dear Sir and Madam,

Currently I have problems to use my Realsense camera. I pull the docker and would like to start with the single camera intrinsic calibration. For that I would like to run` roslaunch realsense2_camera rs_camera.launch`. Unfortunatily he didn't recognize the camera themself. I am able to see the cameraoutput via realsense-viewer. So the camera themself is definitely connect via usb3.2. I don't know what is the problem cuz like I understand when I using docker successfully I don't have to concentrate on the software part. Enclose I submit the shelloutputt.

`root@tobias-Latitude-5580:~# ls /dev/video*
/dev/video0  /dev/video2  /dev/video4  /dev/video6
/dev/video1  /dev/video3  /dev/video5  /dev/video7
root@tobias-Latitude-5580:~# roslaunch realsense2_camera rs_camera.launch enable_gyro:=true enable_accel:=true
... logging to /root/.ros/log/e7dd92d2-e96b-11ec-b371-f894c28bcba3/roslaunch-tobias-Latitude-5580-461.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tobias-Latitude-5580:41791/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
[... 377 lines cut ...]
    Color	  320x240	@ 6Hz	   RGB8
    Color	  320x240	@ 6Hz	   Y16
    Color	  320x240	@ 6Hz	   BGRA8
    Color	  320x240	@ 6Hz	   RGBA8
    Color	  320x240	@ 6Hz	   BGR8
    Color	  320x240	@ 6Hz	   YUYV
    Color	  320x180	@ 60Hz	   RGB8
    Color	  320x180	@ 60Hz	   Y16
    Color	  320x180	@ 60Hz	   BGRA8
    Color	  320x180	@ 60Hz	   RGBA8
    Color	  320x180	@ 60Hz	   BGR8
    Color	  320x180	@ 60Hz	   YUYV
    Color	  320x180	@ 30Hz	   RGB8
    Color	  320x180	@ 30Hz	   Y16
    Color	  320x180	@ 30Hz	   BGRA8
    Color	  320x180	@ 30Hz	   RGBA8
    Color	  320x180	@ 30Hz	   BGR8
    Color	  320x180	@ 30Hz	   YUYV
    Color	  320x180	@ 6Hz	   RGB8
    Color	  320x180	@ 6Hz	   Y16
    Color	  320x180	@ 6Hz	   BGRA8
    Color	  320x180	@ 6Hz	   RGBA8
    Color	  320x180	@ 6Hz	   BGR8
    Color	  320x180	@ 6Hz	   YUYV
`

MartyG-RealSense, 2022-06-11:
Hi @tobiasonkes  I note that your camera has firmware driver version **5.13.0.50** installed.  This is not ideal for use with librealsense SDK version 2.32.1.  This is because firmwares that are 5.12.8.200 or newer recommend SDK version **2.39.0** or newer due to internal changes in the firmware from 5.12.8.200 onwards.  

So for SDK versions before 2.39.0 the firmware version **5.12.7.100** is recommendable, as this was the final version before the internal firmware changes were implemented.

Please also try adding the command **initial_reset:=true** onto the end of your roslaunch instruction to see whether resetting the camera at launch improves the launch's reliability.

Also, if you want to perform a calibration of the camera intrinsics then you should not require ROS to do that, as the **On-Chip Calibration** tool available from the drop-down menu of the 'More' option in the RealSense Viewer side-panel can calibrate camera intrinsics.

[image]

tobiasonkes, 2022-06-11:
Thank you for you quick response. I was able to update the firmware and kind its solve my probleme. But when I want to start the the cam via rosrun I got this respond. 

`11/06 11:38:20,548 WARNING [139689125066496] (ds5-timestamp.cpp:76) UVC metadata payloads not available. Please refer to the installation chapter for details.
[ WARN] [1654947500.550376090]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)`

Do you maybe know what is the issue now. 
Thanks in advance

MartyG-RealSense, 2022-06-11:
I'm pleased to hear that you made significant progress.  Good luck with your remaining tests!

tobiasonkes, 2022-06-11:
Right now I have the issue 
`[ WARN] [1654948361.792477095]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
 11/06 11:52:42,076 WARNING [140368093886208] (ds5-timestamp.cpp:76) UVC metadata payloads not available. Please refer to the installation chapter for details.
`
Do you have any Idea why this error message is shown?

MartyG-RealSense, 2022-06-11:
What method did you use to install the librealsense SDK, please?  If the SDK was built from source code but the kernel was not patched then the SDK would be missing support for hardware metadata.

If librealsense was installed from Debian packages or built from source code with the **RSUSB backend** method then hardware metadata support should be automatically included.

tobiasonkes, 2022-06-11:
ok actually I use an docker container to run open p track. So I don't know it exactly. Like I understood is that if I run the cam now with the correct firmware it should working for all what openPtrack supose to run. 

https://github.com/IntelRealSense/realsense-ros/issues/803

In this issue I understand that I can ignore it right?

MartyG-RealSense, 2022-06-11:
https://github.com/IntelRealSense/realsense-ros/issues/803 used to be true but support for the publishing of metadata topics was added recently in the RealSense ROS wrapper.  Hpwever, since you are using the old wrapper version 2.2.12, it would still be true that the wrapper that you are using was not publishing metadata topics.

Even before then though, having accurate metadata was important to some RealSense ROS users.  It depends whether metadata is important to your particular project.

MartyG-RealSense, 2022-06-18:
Hi @tobiasonkes  Do you require further assistance with this case, please?  Thanks!

tobiasonkes, 2022-06-22:
Thanks a lot now it working fine :). Just have been on Vocations recently.
Just want to ask if you have experience with openPtrack? Because They don't reply to my messages.

MartyG-RealSense, 2022-06-22:
I have not had experience with it personally but there are some RealSense related discussions on their support forum at https://github.com/OpenPTrack/open_ptrack_v2/issues/51 and https://github.com/OpenPTrack/open_ptrack_v2/issues/54

MartyG-RealSense, 2022-06-28:
Hi @tobiasonkes  As your project is working, do you require further assistance please or can this case be closed?  Thanks!

closed 2022-06-29 by tobiasonkes