realsenseai/realsense-ros #2377: How to ensure get_frame_metadata strictly correspond to frame

Zoltan3057, 2022-06-12:
Hi, 
I wonder how the metadata got by "get_frame_metadata" is strictly correspond to the frame in "frame_callback." There is no lock. Is it based on the assumption that the processing time is quick enough. If there is an exposure change, will the metadata happen to be the changed exposure time value.

MartyG-RealSense, 2022-06-13:
Hi @Zoltan3057  Metadata attributes are updated and recalculated on a per-frame basis.

If depth exposure changes whilst depth auto-exposure is enabled then this change will be reflected in the metadata attribute 
**ACTUAL_EXPOSURE** that represents the "real" exposure value, since the exposure value does not update whilst auto-exposure is active.  

The ACTUAL_EXPOSURE for the RGB stream cannot be retrieved whilst RGB auto-exposure is enabled though.

MartyG-RealSense, 2022-06-19:
Hi @Zoltan3057  Do you require further assistance with this case, please?  Thanks!

Zoltan3057, 2022-06-20:
Sorry for late reply.
Previously, I recorded images and exposure times while changing camera's exposure time. And there are some mismatches between images and exposure times. Somehow those mismatches don't appear anymore.
Add sleep(0.1) after sensor.set_option may be the reason. 
Thanks for your explanation. It helps a lot.

closed 2022-06-20 by Zoltan3057