realsenseai/realsense-ros #2378: SLAM with L515: map not appearing,

ssgarimella-create, 2022-06-14:
I am trying to follow the "[SLAM with RealSense™ D435i camera on ROS](https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i)" but with a L515. I followed the instructions exactly; RViz opens up but I do not see the SLAM map. I am running ROS noetic Ubuntu 20.04. What is the process for using the SLAM program on a L515 instead of D435i camera? Please help urgently, I have also included the log files.

[camera-realsense2_camera-2-stdout.log](https://github.com/IntelRealSense/realsense-ros/files/8896608/camera-realsense2_camera-2-stdout.log)
[ImuFilter-3-stdout.log](https://github.com/IntelRealSense/realsense-ros/files/8896609/ImuFilter-3-stdout.log)
[roslaunch-sandilya-XPS-13-9380-21142.log](https://github.com/IntelRealSense/realsense-ros/files/8896610/roslaunch-sandilya-XPS-13-9380-21142.log)
.

MartyG-RealSense, 2022-06-14:
Hi @hunterlineage21  May I first confirm that you have enabled accel and gyro in your roslaunch instruction by adding the commands **enable_accel:=true enable_gyro:=true** to it.  The IMU topics were disabled by default in the RealSense ROS wrapper  some time after Intel's D435i SLAM guide was published.

`roslaunch realsense2_camera opensource_tracking.launch enable_accel:=true enable_gyro:=true`

hunterlineage21, 2022-06-14:
Thank you for getting back @MartyG-RealSense . I tried the command you suggested and also reversing the direction of the USB micro connector. The problem is that the map is showing now, but this time it shoots off to infinity, even though the camera is perfectly still. The camera frame in the map rotates and moves on its own further and further away from the origin such that it is not visible anymore, and a SLAM map is not being created.

MartyG-RealSense, 2022-06-15:
Has your L515's IMU been calibrated with the Python IMU calibration tool at the link below?

https://dev.intelrealsense.com/docs/imu-calibration-tool-for-intel-realsense-depth-camera

The IMU component used in RealSense cameras does not have internal hardware calibration and must be calibrated with software.  An uncalibrated IMU may have a Y accel value that keeps changing even when the camera is stationary on a flat surface.

hunterlineage21, 2022-06-16:
Yes, I tried calibrating it and the problem still persists. Please help!

MartyG-RealSense, 2022-06-16:
There is a RealSense ROS L515 SLAM project at the link below.

https://github.com/wh200720041/MMS_SLAM

It is a **Semantic SLAM** project, which usually means that it is SLAM that does not require use of the IMU.  In that project, they are using the **rs_camera.launch** launch file with a pointcloud filter enabled. 

**roslaunch realsense2_camera rs_camera.launch filters:=pointcloud**

They are also installing a package called **hector trajectory server** to provide the SLAM visualization in RViz.

http://wiki.ros.org/hector_trajectory_server

[image]

With the rs_camera launch file, RViz would be opened with the command **rosrun rviz rviz**

MartyG-RealSense, 2022-06-17:
1.  Thanks very much for the confirmation of your computer model.

2.  The ROS wrapper version that you are using is displayed in the roslaunch log, when it is working correctly.  Please see if you can display it when using the rs_camera launch file.

[image]

I replied at your recorded rosbag link about the **Failed to meet update rate** error.

In regard to C++ and Python scripts for generating a pointcloud from a bag, you could use the official pointcloud example programs and set them to use a bag file as the data source instead of a live camera.

**C++**
https://github.com/IntelRealSense/librealsense/blob/master/examples/pointcloud/rs-pointcloud.cpp

**Python**
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_pointcloud_viewer.py

***********

You can set a bag as the data source by placing the **cfg.enable_device_from_file** instruction on the line befor the pipeline start instruction.

**C++**
```
rs2::config cfg;
cfg.enable_device_from_file("test.bag");
pipe.start(cfg); 
```

**Python**
```
pipe = rs.pipeline()
cfg = rs.config()
cfg.enable_device_from_file("test.bag");
pipe.start(cfg)
```

hunterlineage21, 2022-06-17:
The ROS realsense version that I am using is: [ INFO] [1655493100.229075284]: **RealSense ROS v2.3.2**
Thank you for showing the C++ and Python resources. I will try them and revert. I was also wondering if you could help me with how to save the bag file using a Python script that is using OpenCV or Open3D to create a real time slam map? As in, is there a way to save the point cloud data coming from the L515, running from a Python script, in bag format?

MartyG-RealSense, 2022-06-18:
It looks as though the creator of BreezySLAM wrote scripts for specific lidar device models and placed them in the **examples** sub-folder of BreezySLAM.

https://github.com/simondlevy/BreezySLAM/tree/master/examples

[image]

So you would presumably have to create a .py script file similar to the highlighted ones to add L515 support if it were possible to do so.  

As a possible alternative to BreezySLAM, the **RTABMAP** SLAM tool has a standalone version which can be used without ROS and has been demonstrated as working with L515.

http://introlab.github.io/rtabmap/

https://www.youtube.com/watch?v=e_KS3RHRwSE

There is also a ROS-based SLAM solution specifically for L515 here:

https://github.com/wh200720041/SSL_SLAM

hunterlineage21, 2022-06-20:
Thank you for the information. How do I use the [ROS-based SLAM solution specifically for L515](https://github.com/wh200720041/SSL_SLAM) with python? Would I have to create a new ros node that can let me create the SLAM map in python? I will also try creating a .py script for the BreezySLAM.

Going back to the "[SLAM with RealSense™ D435i camera on ROS](https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i)" problem, my IMU is calibrated and the map is showing now, but it keeps rotating away even though the L515 is stationary.

MartyG-RealSense, 2022-06-21:
There is a Python version of SSL_SLAM worked on by the same developer at the link below.

https://github.com/wh200720041/MMS_SLAM

closed 2022-06-27 by hunterlineage21