realsenseai/realsense-ros #2380: Realsense dmesg spam

BryanStuurman, 2022-06-16:
I have a 415 on my desk and four 435's deployed in the field that are generating a large amount of traffic on dmesg when they're first started up:
```
[ normal boot stuff excised ]
[  349.623072] usb 4-3.2: new SuperSpeed Gen 1 USB device number 3 using xhci_hcd
[  349.643534] usb 4-3.2: New USB device found, idVendor=8086, idProduct=0ad3, bcdDevice=50.d0
[  349.643539] usb 4-3.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  349.643542] usb 4-3.2: Product: Intel(R) RealSense(TM) Depth Camera 415 
[  349.643545] usb 4-3.2: Manufacturer: Intel(R) RealSense(TM) Depth Camera 415 
[  349.643548] usb 4-3.2: SerialNumber: 110423023290
[  349.685543] mc: Linux media interface: v0.10
[  349.691424] videodev: Linux video capture interface: v2.00
[  349.701782] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[  349.701793] uvcvideo: Unknown video format 20303157-0000-0010-8000-00aa00389b71
[  349.701851] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[  349.774403] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3.2/4-3.2:1.0/input/input25
[  349.774524] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[  349.774526] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[  349.811628] usbcore: registered new interface driver uvcvideo
[  349.811630] USB Video Class driver (1.1.1)
[  415.261271] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[  415.261282] uvcvideo: Unknown video format 20303157-0000-0010-8000-00aa00389b71
[  415.261373] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[  415.334686] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3.2/4-3.2:1.0/input/input26
[  415.402724] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[  415.402748] uvcvideo: Unknown video format 20303157-0000-0010-8000-00aa00389b71
[  415.403389] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[  415.476832] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3.2/4-3.2:1.0/input/input27
[  421.834619] usb 4-3.2: Disable of device-initiated U1 failed.
[  421.850101] usb 4-3.2: Disable of device-initiated U2 failed.
[  422.020394] usb 4-3.2: reset SuperSpeed Gen 1 USB device number 3 using xhci_hcd
[  422.236164] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[  422.236179] uvcvideo: Unknown video format 20303157-0000-0010-8000-00aa00389b71
[  422.236253] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[  422.309851] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3.2/4-3.2:1.0/input/input28
[  422.430863] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[  422.430886] uvcvideo: Unknown video format 20303157-0000-0010-8000-00aa00389b71
[  422.430968] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[  422.503900] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3.2/4-3.2:1.0/input/input29
[  422.582724] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[  422.582740] uvcvideo: Unknown video format 20303157-0000-0010-8000-00aa00389b71
[  422.582802] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[  422.656118] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3.2/4-3.2:1.0/input/input30

[ 500 lines removed for clarity. Yes, 500 lines.]

[  438.486795] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[  438.486811] uvcvideo: Unknown video format 20303157-0000-0010-8000-00aa00389b71
[  438.487774] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[  438.560487] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3.2/4-3.2:1.0/input/input114
[  438.651244] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[  438.651264] uvcvideo: Unknown video format 20303157-0000-0010-8000-00aa00389b71
[  438.651344] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 415  (8086:0ad3)
[  438.723982] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3.2/4-3.2:1.0/input/input115

```
The 415 (for which the logs are pictured above) is running the most current firmware. Others are running a sprinkling of firmware versions dating back to early 2021; they all do this.
The systems all run:
ROS 1 noetic
Ubuntu 20.04
RealSense ROS v2.3.2
Built with LibRealSense v2.50.0
Running with LibRealSense v2.50.0

Why does this occur, and can it be limited in the future? This makes debugging issues on remote machines that much harder.

MartyG-RealSense, 2022-06-16:
Hi @BryanStuurman  These messages originate from the kernel.  There was a request in 2021 to be able to mute kernel messages to ease debugging but it was declined by Intel for reasons provided at https://github.com/IntelRealSense/librealsense/issues/8215#issuecomment-789909398

You could try building the librealsense SDK from source code with the **RSUSB backend** installation method, which bypasses the kernel to see whether this reduces the volume of kernel-related messages.  An RSUSB source code build is not dependent on Linux versions or kernel versions and does not require patching.

BryanStuurman, 2022-06-16:
Interesting. I'm less interested in muting the messages and more so interested in why they occur in the first place.
in the example above we get up to input device 115
`/devices/pci0000:00/0000:00:14.0/usb4/4-3/4-3.2/4-3.2:1.0/input/input115`   
and after 2 hotplugs my desktop has it as /input304

This suggests me to some enumeration issue in the realsense itself, can you comment as to the root cause of this?

MartyG-RealSense, 2022-06-16:
Some kernel versions may conflict with librealsense, especially kernels that are unsupported by the SDK.  The newest supported kernels are 5.4 if building from Debian packages, and 5.8 or 5.11 if building from source code.  Unsupported kernels, such as 5.13, can work with the SDK but there may be unpredictable consequences in regards to stability.  An RSUSB build of librealsense can help to avoid kernel conflicts as it bypasses the kernel.

BryanStuurman, 2022-06-16:
Okay. I am running kernel `5.4.0-120-generic` and using the debian packages.

We are having connection issues with one robot in the field, probably related to the USB cable (the camera is on the end effector). Would the RSUSB build be more reliable with a spotty USB connection?

MartyG-RealSense, 2022-06-16:
An unreliable USB connection will negatively affect any type of librealsense build, so it is better to try to address the cause of the USB issue. 

Movement of the camera's USB cable during robot motion may be slightly dislodging the cable connector in the USB port and causing disconnection.  There are a pair of small holes either side of the camera's micro-USB port for a **screw lock** USB cable to be used for firmly affixing the cable into the port so that motion does not dislodge it.  Information about this can be found at https://github.com/IntelRealSense/librealsense/issues/3118#issuecomment-456392939

BryanStuurman, 2022-06-16:
Indeed we are using the screw lock cables, I am hoping to find a software solution that adds some robustness while the client waits for replacement cables/parts.
Thanks for your time, i will investigate RSUSB.

MartyG-RealSense, 2022-06-16:
You are very welcome.  It is possible that the USB cable on the affected robot may have internal damage if it gets flexed a lot during motion, or it could simply be a bad cable.  The quality of USB-C cables can vary when using your own choice of cable instead of the official one supplied with the camera that is validated by Intel for use with RealSense.

If your robot uses 6 FPS speed then I would also recommend increasing to 15 FPS, as errors such as stream time-outs may occur at 6 FPS (due to frames not arriving quickly enough) that disappear at 15 FPS.

BryanStuurman, 2022-06-16:
We spend a fair bit of money on these cables, but I will be investigating the failure thoroughly when the hardware comes back. We run 30FPS - would 15 FPS be preferable to 30?

MartyG-RealSense, 2022-06-16:
Reducing from 30 FPS to 15 would be advisable if there is evidence that the computer board on the robot is struggling to process the data stably at 30. The higher the FPS, the higher the volume of data being transmitted from the camera to the USB port of the computer board, increasing the risk that the computer cannot keep up with processing the incoming frames.  

Using a system monitoring tool on Ubuntu such as **htop** would show whether the CPU usage percentage is near 100 percent, which would indicate that the CPU is over-burdened and so negatively affecting overall performance.

BryanStuurman, 2022-06-16:
CPU load is not a problem, this is an i5 6/12 core/thread chip, the realsense2_camera_manager process uses about 30% peak of a single execution port.

What is the failure mode when USB errors delay a frame transfer beyond when the next frame is due? Does the node crash or does it try again ASAP, or wait until the next scheduled frame time?

MartyG-RealSense, 2022-06-16:
If new frames are not received then the SDK will listen for frames for 5 seconds before declaring a time-out.  If frames arrive within 5 seconds then the pipeline should resume streaming automatically from the point where it left off.

If a script uses **try_wait_for_frames()** instead of wait_for_frames then the program should not exit with an exception error when time-out occurs and instead continue running.

BryanStuurman, 2022-06-16:
Sounds like pretty reasonable behaviour. 

Also implies that the realsense is configured to autonomously generate frames, instead of being requested for them by the SDK - true?

MartyG-RealSense, 2022-06-16:
A camera will not be generating frames until it is requested by a program to do so and once started it will continue streaming frames until stopped by the program, whether due to an intended stop or an exit due to an error such as **RuntimeError: Frame didn't arrived within 5000**

MartyG-RealSense, 2022-06-23:
Hi @BryanStuurman  Do you require further assistance with this case, please?  Thanks!

BryanStuurman, 2022-06-23:
@MartyG-RealSense we had the client replace the cable for us and our connectivity issue is now solved. I'd like to do some forensics to see how many packets were lost - but you can close this issue. 

Please let me know if you'd like to me reopen it when I make my findings.

MartyG-RealSense, 2022-06-24:
Hi @BryanStuurman  I'm pleased to hear that you had success after a cable change.  It is fine to keep this case open until you are ready to report your findings.  Thanks again!

Idnobu, 2022-07-10:
I just need log to spam money

closed 2022-07-16 by MartyG-RealSense