realsenseai/realsense-ros #2383: ROS2 Failure to connect: failed to set power state

ajordan5, 2022-06-17:
I recently built the ROS2 beta driver and am unable to connect to my D435. I get the following error when I run `ros2 launch realsense2_camera rs_launch.py`

```
[realsense2_camera_node-1] [WARN] [1655500276.588853501] [camera.camera]: Device 1/1 failed with exception: failed to set power state
[realsense2_camera_node-1] [ERROR] [1655500276.588920207] [camera.camera]: The requested device with  is NOT found. Will Try again.
[realsense2_camera_node-1]  17/06 15:11:16,587 ERROR [139780834711296] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
[realsense2_camera_node-1]  17/06 15:11:16,587 ERROR [139781335738112] (sensor.cpp:572) acquire_power failed: failed to set power state
[realsense2_camera_node-1]  17/06 15:11:16,588 WARNING [139781335738112] (rs.cpp:310) null pointer passed for argument "device"

```

Is there a parameter or something that I am missing to be able to connect? I know that it finds my device, because if I unplug it, I get the following message: 

```
[realsense2_camera_node-1] [WARN] [1655500285.993681950] [camera.camera]: No RealSense devices were found!

```

MartyG-RealSense, 2022-06-18:
Hi @ajordan5  Is the camera able to be detected if you use the launch instruction below, please?

`ros2 launch realsense2_camera rs_launch.py device_type:=d435 initial_reset:=true`

ajordan5, 2022-06-18:
Sure, I tried that out for both the d435 and d435i, then tried a similar command for the d415. Unfortunately I still get the same error

```
INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [165809]
[realsense2_camera_node-1] [INFO] [1655557086.916781141] [camera.camera]: RealSense ROS v4.0.4
[realsense2_camera_node-1] [INFO] [1655557086.916945080] [camera.camera]: Built with LibRealSense v2.50.0
[realsense2_camera_node-1] [INFO] [1655557086.916975557] [camera.camera]: Running with LibRealSense v2.50.0
[realsense2_camera_node-1] [WARN] [1655557086.947198381] [camera.camera]: Device 1/1 failed with exception: failed to set power state
[realsense2_camera_node-1] [ERROR] [1655557086.947261593] [camera.camera]: The requested device with device name containing d435 is NOT found. Will Try again.
[realsense2_camera_node-1]  18/06 06:58:06,946 ERROR [140005884270336] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
[realsense2_camera_node-1]  18/06 06:58:06,946 ERROR [140005909448448] (sensor.cpp:572) acquire_power failed: failed to set power state
[realsense2_camera_node-1]  18/06 06:58:06,947 WARNING [140005909448448] (rs.cpp:310) null pointer passed for argument "device"

```

MartyG-RealSense, 2022-06-18:
Thank you.  Next, please test whether the camera can be launched successfully with **ros2 run** instead of a launch file.

`ros2 run realsense2_camera realsense2_camera_node`

ajordan5, 2022-06-21:
Thank you for your help. After looking into realsense-viewer I realized that I assumed I had the latest version of the RealSense SDK but I actually didn't. Installing the latest version fixed this problem.

closed 2022-06-21 by ajordan5

MartyG-RealSense, 2022-06-21:
You are very welcome.  It's great to hear that you achieved a solution.  Thanks for the update!