realsenseai/realsense-ros #2385: is OpenCV needed for librealsense with CUDA on Xavier AGX

MrOCW, 2022-06-20:
Hi, I am trying to build librealsense from source on a Jetson Xavier AGX. Do I need to first build OpenCV with CUDA or librealsense has no dependency on OpenCV w/ CUDA?
Also, is building from source the only way for CUDA usage? I've read that debian is compiled with CUDA but its not applicable for realsense-ros? 
Thanks

MartyG-RealSense, 2022-06-20:
Hi @MrOCW  Installing OpenCV is not necessary to add CUDA support to librealsense.  

If librealsense is installed from Debian packages for Jetson at the link below then CUDA support will be automatically included.  

If building librealsense from source code with CMake and patching the kernel then CUDA support can be added by including the build flag **-DBUILD_WITH_CUDA=true** in the CMake build instruction.

If librealsense is built from source code with the **RSUSB backend** installation method then CUDA support is included automatically and you do not need to patch the kernel.

If librealsense and the RealSense ROS wrapper are built from packages together at the same time with the wrapper's **Method 1** instructions then CUDA support is not included in the packages.

MrOCW, 2022-06-20:
I am building it on Jetpack 5 with Ubuntu 20.04
I did not patch the kernel but the build went fine. With this in mind, can I confirm that RSUSB backend still needs to be set to true?

MartyG-RealSense, 2022-06-20:
A reason for patching the kernel is to add support for hardware metadata.  I would advise patching the kernel, which is done before building, if you are not using the RSUSB installation method but if your project is not dependent on camera metadata then you may not need to.

MrOCW, 2022-06-20:
Got it. In that case, I can safely say that for my case, (RSUSB=false, no patch, with_cuda=true) CUDA will still be used? Just that I don't have access to the meta data?

MartyG-RealSense, 2022-06-20:
CUDA should be supported but hardware metadata will be unavailable.  Metadata provides information about a range of camera information such as frame timestamps, exposure values, temperature, FPS, etc.  

https://github.com/IntelRealSense/librealsense/blob/master/doc/frame_metadata.md#employing-metadata-attributes-in-demos

MartyG-RealSense, 2022-06-20:
Nvidia introduced Jetson support for Ubuntu 20.04 in JetPack 5.  

[image]

AS https://github.com/IntelRealSense/librealsense/issues/10448 mentions though, the librealsense SDK does not currently have support for Arm package installation on 20.04 at the time of writing this.  The support is planned to be provided as part of the next-generation networking system in an upcoming SDK release.

MrOCW, 2022-06-20:
got it. so the only way is to build from source now then :)

closed 2022-06-20 by MrOCW