realsenseai/realsense-ros #2386: (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

hugoyvrn, 2022-06-20:
Hello,

Camera: Realsense depth camera D455
OS: Linux (Ubutuntu 20.04, same issue in Ubuntu 18.04)
ROS distribution: ros1 noetic (same issue on ros1 with melodic and ros2 with galactic)

No problem seen with realsense-viewer (neither on Windows 10 nor Ubuntu 20.04).

I changed the USB port and the USB C cable, tested on different computers and reset the camera with realsense-viewer.

I am want to do localization with the camera `roslaunch realsense2_camera rs_camera.launch` or `roslaunch realsense2_camera rs_rgbd.launch`, whatever the parameters used I never get Pose, odom nor tf publication.

Two errors seem to be the problem :
```bash
control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
```
and
```bash
(messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0
```
Does anyone have any idea what is causing this problem?

Thanks in advance.

# Some terminal outputs:

>```bash
>$ roslaunch realsense2_camera rs_camera.launch
>... logging to /home/hugo/.ros/log/6193d680-f07f-11ec-bc70-611a233072b6/roslaunch-hugo-victus-pc-50095.log
>Checking log directory for disk usage. This may take a while.
>Press Ctrl-C to interrupt
>Done checking log file disk usage. Usage is <1GB.
>
>started roslaunch server http://hugo-victus-pc:38211/
>
>SUMMARY
>========
>
>PARAMETERS
> * /camera/realsense2_camera/accel_fps: -1
> * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
> * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
> * /camera/realsense2_camera/align_depth: False
> * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
> * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
> * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
> * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
> * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
> * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
> * /camera/realsense2_camera/allow_no_texture_points: False
> * /camera/realsense2_camera/base_frame_id: camera_link
> * /camera/realsense2_camera/calib_odom_file: 
> * /camera/realsense2_camera/clip_distance: -2.0
> * /camera/realsense2_camera/color_fps: -1
> * /camera/realsense2_camera/color_frame_id: camera_color_frame
> * /camera/realsense2_camera/color_height: -1
> * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
> * /camera/realsense2_camera/color_width: -1
> * /camera/realsense2_camera/confidence_fps: -1
> * /camera/realsense2_camera/confidence_height: -1
> * /camera/realsense2_camera/confidence_width: -1
> * /camera/realsense2_camera/depth_fps: -1
> * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
> * /camera/realsense2_camera/depth_height: -1
> * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
> * /camera/realsense2_camera/depth_width: -1
> * /camera/realsense2_camera/device_type: 
> * /camera/realsense2_camera/enable_accel: False
> * /camera/realsense2_camera/enable_color: True
> * /camera/realsense2_camera/enable_confidence: True
[... 602 lines cut ...]
>[ WARN] (2022-06-20 11:21:25.149) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=98) between -1 and 243"
>[ INFO] [1655716885.149890075]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.091411s
>[ WARN] (2022-06-20 11:21:25.241) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=110) between -1 and 244"
>[ INFO] [1655716885.241869196]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.090804s
>[ WARN] (2022-06-20 11:21:25.340) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=105) between -1 and 245"
>[ INFO] [1655716885.341199164]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.097723s
>[ WARN] (2022-06-20 11:21:25.436) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=92) between -1 and 246"
>[ INFO] [1655716885.437088340]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.093939s
>[ WARN] (2022-06-20 11:21:25.534) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=95) between -1 and 247"
>[ INFO] [1655716885.535575840]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.097187s
>[ WARN] (2022-06-20 11:21:25.624) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=89) between -1 and 248"
>[ INFO] [1655716885.625547620]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.088747s
>[ WARN] (2022-06-20 11:21:25.733) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=95) between -1 and 249"
>[ INFO] [1655716885.734145576]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.100148s
>[ WARN] (2022-06-20 11:21:25.835) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=85) between -1 and 250"
>[ INFO] [1655716885.836490613]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.100000s
>[ERROR] (2022-06-20 11:21:25.838) Rtabmap.cpp:1277::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 639 is ignored!
>[ INFO] [1655716885.838836772]: rtabmap (19): Rate=1.00s, Limit=0.000s, Conversion=0.0020s, RTAB-Map=0.0001s, Maps update=0.0000s pub=0.0000s (local map=11, WM=11)
>[ WARN] (2022-06-20 11:21:25.936) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=84) between -1 and 251"
>[ INFO] [1655716885.936707641]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.098461s
>[ WARN] (2022-06-20 11:21:26.038) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=84) between -1 and 252"
>[ INFO] [1655716886.038966847]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.101214s
>[ WARN] (2022-06-20 11:21:26.135) OdometryF2M.cpp:525::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=81) between -1 and 253"
>[ INFO] [1655716886.136475223]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.096497s
>```

MartyG-RealSense, 2022-06-20:
Hi @hugoyvrn  Does the warning still occur if you reset the camera in the ROS wrapper by including the command **initial_reset:=true** in your roslaunch instruction, please?

`roslaunch realsense2_camera rs_camera.launch initial_reset:=true`

hugoyvrn, 2022-06-20:
Thanks @MartyG-RealSense for your quick response.

This is the first thing I tried, it doesn't seem to change anything.

```bash
$ roslaunch realsense2_camera rs_camera.launch initial_reset:=true
... logging to /home/hugo/.ros/log/a106dcb8-f07f-11ec-bc70-611a233072b6/roslaunch-hugo-victus-pc-58553.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://hugo-victus-pc:41301/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
[... 77 lines cut ...]
[ INFO] [1655731392.094896203]: Device Sensors: 
[ INFO] [1655731392.190691378]: Stereo Module was found.
[ INFO] [1655731392.197778718]: RGB Camera was found.
[ INFO] [1655731392.197887306]: Motion Module was found.
[ INFO] [1655731392.197919868]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1655731392.197945948]: num_filters: 0
[ INFO] [1655731392.197961457]: Setting Dynamic reconfig parameters.
[ INFO] [1655731395.240527707]: Done Setting Dynamic reconfig parameters.
[ INFO] [1655731395.240949643]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1655731395.241797553]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1655731395.241821919]: setupPublishers...
[ INFO] [1655731395.243467794]: Expected frequency for depth = 30.00000
[ INFO] [1655731395.264402819]: Expected frequency for color = 30.00000
[ INFO] [1655731395.278194679]: setupStreams...
 20/06 15:23:15,526 WARNING [139947239515904] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/06 15:23:15,577 WARNING [139947239515904] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/06 15:23:15,627 WARNING [139947239515904] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1655731395.677900580]: SELECTED BASE:Depth, 0
[ INFO] [1655731395.685538742]: RealSense Node Is Up!
[ WARN] [1655731395.740031650]: 
 20/06 15:23:15,740 WARNING [139947239515904] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/06 15:23:15,791 WARNING [139947239515904] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/06 15:23:15,841 WARNING [139947239515904] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/06 15:23:15,891 WARNING [139947239515904] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
```

MartyG-RealSense, 2022-06-20:
Thank you.  Next, let's test whether the problem is related to the ROS wrapper having difficulty with processing the volume of data by using a lower stream configuration (depth only, 640x480, 15 FPS).

roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=15

Do the **control_transfer_returned** warnings still occur with this launch?

hugoyvrn, 2022-06-20:
Okay, do you have any idea how I can fix this? To me, it seems to be a camera software issue.

There is no change with these settings (I've already tested a lot of configurations from what you gave in the other github issues).

```bash
$ roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=15
... logging to /home/hugo/.ros/log/a106dcb8-f07f-11ec-bc70-611a233072b6/roslaunch-hugo-victus-pc-70981.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://hugo-victus-pc:46029/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: 15
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
[... 83 lines cut ...]
[ INFO] [1655754320.474307173]: Expected frequency for color = 30.00000
[ INFO] [1655754320.488904340]: setupStreams...
 20/06 21:45:20,704 WARNING [140378555541248] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/06 21:45:20,754 WARNING [140378555541248] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/06 21:45:20,804 WARNING [140378555541248] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1655754320.854922430]: SELECTED BASE:Depth, 0
 20/06 21:45:20,854 WARNING [140378555541248] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1655754320.860755640]: RealSense Node Is Up!
[ WARN] [1655754321.025031797]: 
 20/06 21:45:21,025 WARNING [140378555541248] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/06 21:45:21,075 WARNING [140378555541248] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/06 21:45:21,125 WARNING [140378555541248] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/06 21:45:21,176 WARNING [140378555541248] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
 20/06 21:45:21,930 WARNING [140378555541248] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/06 21:45:22,080 WARNING [140378555541248] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 20/06 21:45:22,130 WARNING [140378555541248] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
```

MartyG-RealSense, 2022-06-21:
Can you confirm that you do not have any other RealSense programs, such as realsense-viewer, already streaming with the camera before you perform the ROS launch?

Do the warnings reduce if you add **enable_color:=false** to your roslaunch instruction to disable the color stream?

hugoyvrn, 2022-06-21:
I can you confirm that I do not have any other RealSense programs lunch in the same time.

Today, even after rebooting my computer, I have the other issue (the error `(messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Success, number: 0`). But with an other computer I still get the same issue (the error `(messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11`). Anyway `enable_color:=false` does not change a thing.

```bash
$ roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=15 enable_color:=false
... logging to /home/hugo/.ros/log/27819598-f144-11ec-afad-a598d86d1557/roslaunch-hugo-victus-pc-19792.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.started roslaunch server http://hugo-victus-pc:40621/SUMMARY
========PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file:
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: 15
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type:
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: False
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters:
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
[... 550 lines cut ...]
[ INFO] [1655804154.522886468]: setupPublishers...
[ INFO] [1655804154.525888814]: Expected frequency for depth = 15.00000
[ INFO] [1655804154.590587965]: Expected frequency for color = 30.00000
[ INFO] [1655804154.645225192]: setupStreams...
 21/06 11:35:54,878 WARNING [140275083101952] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 21/06 11:35:54,928 WARNING [140275083101952] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 21/06 11:35:54,978 WARNING [140275083101952] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1655804155.029405519]: SELECTED BASE:Depth, 0
[ INFO] [1655804155.063051780]: RealSense Node Is Up!
[ WARN] [1655804155.158494832]:
[... cut ...]
 21/06 11:35:55,878 WARNING [140275083101952] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 21/06 11:35:56,028 WARNING [140275083101952] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 21/06 11:35:56,079 WARNING [140275083101952] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

```

MartyG-RealSense, 2022-06-21:
Let's look at the basics of your librealsense and ROS wrapper installations.  You are correctly matching wrapper version 2.3.2 and librealsense version 2.50.0.  

Which methods did you use to install librealsense and the ROS wrapper, please?  Did you:

1.  Install 2.3.2 and 2.50.0 from packages together at the same time using the ROS1 wrapper's **Method 1** instructions

2.  Install 2.3.2 and 2.50.0 separately from source code.

Sometimes people build librealsense on its own and then install librealsense and the wrapper together from packages with the wrapper's Method 1.  By doing so, they are effectively installing librealsense twice.  Or instead, they install librealsense with both packages and source code on the same computer, resulting in conflicts between the two SDK installation types such as the **Multiple udev rules detected** warning.

hugoyvrn, 2022-06-23:
Thank you for your answer. I installed both packages with the command line `sudo apt-get install ros-noetic-realsense2-camera` or `sudo apt-get install ros-noetic-realsense2-*` (with melodic or galactic instead of noetic depending on the ros distribution).

I have never tried to build something from source. Should I try?

I also tried to remove the packages and reinstall them again.

MartyG-RealSense, 2022-06-23:
All RealSense related packages should be uninstalled with the 'purge' command below. Is that the command that you used to uninstall, please?

`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

Once uninstall has been performed, I would recommend only installing Noetic or Galactic but not both, and purging the packages to remove one version before installing another from packages (for example, removing a Noetic installation before installing Galactic from packages).  This is because performing an apt-get build with the command in the wrapper documentation will install both librealsense and the wrapper.  So doing an apt-get build for both Noetic and Galactic would be installing librealsense twice.

There is no harm in trying a source code build so long as the package build is uninstalled first before doing so.  But you would have to reinstall librealsense separately first (either from packages or source code) and then install the ROS wrapper from source code afterwards.

MartyG-RealSense, 2022-06-29:
Hi @hugoyvrn  Do you require further assistance with this case, please?  Thanks!

zhaojiaran111, 2022-06-29:
I also got this error on my ubuntu 18.04 :

roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=15 enable_color:=false
... logging to /home/ubuntu/.ros/log/df7d8cb2-f77d-11ec-9bef-000c297dee61/roslaunch-ubuntu18-12606.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu18:46121/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: 15
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: False
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
[... 69 lines cut ...]
[ INFO] [1656488054.158218501]: Align Depth: Off
[ INFO] [1656488054.158276365]: Sync Mode: Off
[ INFO] [1656488054.158359392]: Device Sensors: 
[ INFO] [1656488054.218524188]: Stereo Module was found.
[ INFO] [1656488054.228115784]: RGB Camera was found.
[ INFO] [1656488054.228243143]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1656488054.228321297]: num_filters: 0
[ INFO] [1656488054.228336193]: Setting Dynamic reconfig parameters.
[ INFO] [1656488056.161143504]: Done Setting Dynamic reconfig parameters.
[ INFO] [1656488056.161598069]: depth stream is enabled - width: 640, height: 480, fps: 15, Format: Z16
[ INFO] [1656488056.161631006]: setupPublishers...
[ INFO] [1656488056.162978551]: Expected frequency for depth = 15.00000
[ INFO] [1656488056.165121158]: setupStreams...
[ INFO] [1656488056.257729293]: SELECTED BASE:Depth, 0
 29/06 15:34:16,259 WARNING [139784887027456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1656488056.263402398]: RealSense Node Is Up!
[ WARN] [1656488056.452406858]: 
 29/06 15:34:16,453 WARNING [139784887027456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 29/06 15:34:16,504 WARNING [139784887027456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 29/06 15:34:16,556 WARNING [139784887027456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
 29/06 15:34:16,711 WARNING [139784887027456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 29/06 15:34:16,864 WARNING [139784887027456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 29/06 15:34:16,916 WARNING [139784887027456] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

zhaojiaran111, 2022-06-29:
I have used all the methods above and failed

hugoyvrn, 2022-06-29:
> Let's look at the basics of your librealsense and ROS wrapper installations. You are correctly matching wrapper version 2.3.2 and librealsense version 2.50.0.
>
> Which methods did you use to install [...]

> All RealSense related packages should be uninstalled with the 'purge' command below. Is that the command that you used to uninstall, please?
> 
> `dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`
> 
> Once uninstall has been performed, I would recommend only installing Noetic or Galactic but not both, and purging the packages to remove one version before installing another from packages (for example, removing a Noetic wrapper installation before installing the Galactic wrapper from packages). This is because performing an apt-get build with the command in the wrapper documentation will install both librealsense and the wrapper. So doing an apt-get build for both Noetic and Galactic would be installing librealsense twice.
> 
> There is no harm in trying a source code build so long as the package build is uninstalled first before doing so. But you would have to reinstall librealsense separately first (either from packages or source code) and then install the ROS wrapper from source code afterwards.

Hi @MartyG-RealSense, sorry for my late reply, I am now back with the camera.

I used `apt remove --purge ros-noetic-realsense2-*` to remove the packages, but it makes no difference when I do it with `dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`. I don't think the problem is in the installation, I tested on three PCs, one with only melodic (Ubuntu 18), two with noetic and galactic (Ubuntu 20). I had this problem in every test I did. I had to keep galactic on my PC, but I will only install realsense packages on noetic.

Any other ideas?
Thanks,

MartyG-RealSense, 2022-06-30:
@hugoyvrn  What kernel version(s) are you using on your computers, please?  

A package installation of the librealsense SDK supports up to kernel 5.4, whilst the most recent supported kernels on a source-code build of librealsense are 5.8 and 5.11.  Whilst newer unsupported kernels such as 5.13 can work with librealsense, there may be unpredictable consequences in regards to stability.

MartyG-RealSense, 2022-07-06:
Hi @MariamFdil  Can you confirm if you have tried the **initial_reset:=true** method mentioned earlier in this discussion at https://github.com/IntelRealSense/realsense-ros/issues/2386#issuecomment-1160379686 as initial_reset is False in your log.  Thanks!

MariamFdil, 2022-07-06:
@MartyG-RealSense yes I’ve tried it and it didn’t change anything.

MartyG-RealSense, 2022-07-06:
@MariamFdil  It looks as though the pointcloud filter is being enabled.  Do the control_transfer returned warnings still occur if pointcloud is not enabled?

[image]

MariamFdil, 2022-07-06:
@MartyG-RealSense  I tried this command right now roslaunch rs_camera.launch enable_pointcloud:=false. 
Here is the log info : 

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_rgb: False
[... 71 lines cut ...]
[ INFO] [1657118155.763547039]: Done Setting Dynamic reconfig parameters.
[ INFO] [1657118155.768725702]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1657118155.776326929]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1657118155.776599793]: setupPublishers...
[ INFO] [1657118155.790293924]: Expected frequency for depth = 30.00000
[ INFO] [1657118155.976413504]: Expected frequency for color = 30.00000
[ INFO] [1657118156.097102821]: setupStreams...
 06/07 10:35:56,551 WARNING [281472398180720] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/07 10:35:56,602 WARNING [281472398180720] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/07 10:35:56,653 WARNING [281472398180720] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1657118156.704446753]: SELECTED BASE:Depth, 0
 06/07 10:35:56,705 WARNING [281472398180720] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1657118156.752354979]: RealSense Node Is Up!
[ WARN] [1657118156.829460268]: 
 06/07 10:35:56,831 WARNING [281472398180720] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
 06/07 10:35:57,352 WARNING [281472398180720] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/07 10:36:02,318 WARNING [281472398180720] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 06/07 10:36:02,368 WARNING [281472398180720] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

MartyG-RealSense, 2022-07-06:
And the warnings still occur if you use a lower 640x480 resolution for both depth and color?

**roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=30 color_width:=640 color_height:=480 color_fps:=30**

MariamFdil, 2022-07-06:
@MartyG-RealSense Yes still occurs...

MartyG-RealSense, 2022-07-06:
@MariamFdil  Could you check the CPU percentage usage of your Ubuntu 20.04 computer with a system monitoring tool such as **htop** please and see whether CPU usage % is near 100%.  If it is then this could indicate that your warnings might be occurring because the computer's CPU is over-burdened with processing.

prasuchit, 2022-07-06:
I did everything aforementioned and still end up with this while running this command:

roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=30 color_width:=640 color_height:=480 color_fps:=30

` 06/07 17:25:26,663 ERROR [140619812943616] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130

 06/07 17:25:26,679 WARNING [140620232382208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

 06/07 17:25:26,730 WARNING [140620232382208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

 06/07 17:25:26,780 WARNING [140620232382208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

 06/07 17:25:26,830 WARNING [140620232382208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

[ WARN] [1657142727.542752535]: Hardware Notification:Depth stream start failure,1.65714e+12,Error,Hardware Error

 06/07 17:25:27,888 WARNING [140620232382208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

^C[camera/realsense2_camera-3] killing on exit

[camera/realsense2_camera_manager-2] killing on exit

 06/07 17:25:28,523 WARNING [140620232382208] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
`

I didn't want to spam the page with redundant logs so I've only added limited errors and warnings that seemed relevant.

MartyG-RealSense, 2022-07-07:
Hi @prasuchit  Which computer and operating system are you using?  Also, which versions of ROS, the librealsense SDK and the RealSense ROS wrapper are you using?

Could you also provide a complete roslaunch log, please?

prasuchit, 2022-07-07:
Hi @MartyG-RealSense ,

Computer - Dell Precision 5820 Tower with Intel(R) Xeon(R) W-2104 CPU @ 3.20GHz, 72 gigs RAM and Nvidia quadro RTX 4000 gpu.

OS - Ubuntu 20.04; ROS - Noetic; RealSense ROS v2.2.21; LibRealSense v2.50.0;

**roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=30 color_width:=640 color_height:=480 color_fps:=30**

Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Whisler:42071/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
[... 79 lines cut ...]
 07/07 10:00:53,603 WARNING [140168329684736] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
 07/07 10:00:53,989 WARNING [140168329684736] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: Resource temporarily unavailable, number: b
[ INFO] [1657202454.511580952]: Done Setting Dynamic reconfig parameters.
[ INFO] [1657202454.512084190]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1657202454.512575191]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1657202454.512592616]: setupPublishers...
[ INFO] [1657202454.514952459]: Expected frequency for depth = 30.00000
[ INFO] [1657202454.550824595]: Expected frequency for color = 30.00000
[ INFO] [1657202454.571025313]: setupStreams...
[ INFO] [1657202454.617124685]: insert Depth to Stereo Module
[ INFO] [1657202454.617325477]: insert Color to RGB Camera
 07/07 10:00:54,725 WARNING [140168329684736] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1657202454.775347209]: SELECTED BASE:Depth, 0
[ INFO] [1657202454.786561663]: RealSense Node Is Up!
 07/07 10:00:54,852 WARNING [140168329684736] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 07/07 10:00:54,902 WARNING [140168329684736] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 07/07 10:00:54,953 WARNING [140168329684736] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[... cut ...]
 07/07 10:00:56,246 WARNING [140168329684736] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 07/07 10:01:01,500 WARNING [140168329684736] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 07/07 10:01:05,551 WARNING [140168329684736] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

MartyG-RealSense, 2022-07-07:
It is important to match the RealSense ROS wrapper version as closely as possible with the librealsense SDK version listed in that particular wrapper version's release notes.  The librealsense version that matches with the **2.2.21** wrapper version that you are using is **2.41.0**

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.21

If you are using SDK 2.50.0 then the correct Noetic-compatible wrapper version to match it with is **2.3.2**

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.3.2

prasuchit, 2022-07-07:
Another very important problem I forgot to mention(might be because of the same error) is that 2 out of 5 times when I run the driver launch file, the whole system freezes, nothing I do unfreezes it and I've to hard restart my PC. Please let me know how this can be fixed as well. Thanks.

MartyG-RealSense, 2022-07-08:
@prasuchit  Thanks again for your patience.  Whilst RealSense 400 Series cameras can work with any Intel or Arm processor and so should be fine with your Intel Xeon processor, the overall hardware specification of your Xeon server machine may be causing some conflicts with the librealsense SDK that runs underneath the ROS wrapper as the specification is not typical of a consumer PC.  This could be the reason for your frequent launch freezes.

What method do you use to install librealsense and the ROS wrapper, please?

1.  Install librealsense and the ROS wrapper together at the same time from packages using the **Method 1** apt-get instructions in the wrapper documentation.

or

2.  Install librealsense first from source code and then the ROS wrapper secondly from source code.

or

3.  Some other method.

prasuchit, 2022-07-08:
Hi @MartyG-RealSense ,

I used the [this method](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages) for the installation.

MartyG-RealSense, 2022-07-08:
So you installed the librealsense SDK from packages with apt-get and then built the ROS wrapper from source code with the wrapper's **Method 2**, please?

prasuchit, 2022-07-08:
@MartyG-RealSense yes I did.

MartyG-RealSense, 2022-07-08:
Are there any problems in the **RealSense Viewer** tool if you have tried it.  If you have tried the Viewer, does it work normally but the ROS wrapper has problems?

prasuchit, 2022-07-08:
@MartyG-RealSense I found the problem. Since I've been working with realsense-ros for a really long time, somewhere along the way, I installed the packages through both methods 1 and 2(viz. through apt-get instructions on the wrapper documentation and through [these instructions](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages))

So I uninstalled everything related to realsense and reinstalled it just using [this method](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages). The apt-get method definitely has many problems, but [this method](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages) works fine now. 

The control_transfer error is gone and both the viewer and the roslaunch file are running smoothly. You may now close this issue, I'll raise a new one if I run into any further problems. Thanks for all the help. :)

MartyG-RealSense, 2022-07-08:
You are very welcome.  It's great to hear that you achieved a solution.  Thanks very much for the update.  As you suggested, i will close the case.  Thanks again!

MartyG-RealSense, 2022-07-08:
@hugoyvrn @MariamFdil @zhaojiaran111  Do you have updates about your own issues that you can provide, please?  Thanks!

MariamFdil, 2022-07-08:
@MartyG-RealSense As I said yesterday it's not a problem of space.. I tried a new installation in an another workspace and I built from source but I got the same error when I launch the rs_camera launch file. 
PS : realsense_viewer work very well I have one little problem with the motion mode that doesn't work. 

Thank you for your help!

MartyG-RealSense, 2022-07-09:
@MariamFdil  Your mention of JetPack 5.0.1 at the start of you section of this discussion at https://github.com/IntelRealSense/realsense-ros/issues/2386#issuecomment-1175336352 indicates that you are using an Nvidia Jetson as your computer.  You also mention in that initial comment that you built librealsense from source.  

Have you tried an **RSUSB backend** source code build of librealsense on Jetson like the one in the instructions at https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-706643114 by including the CMake build flag **-DFORCE_RSUSB_BACKEND=ON** please?  An RSUSB build bypasses the Linux kernel and so is not dependent on Linux versions or kernel versions and does not require patching.

MariamFdil, 2022-07-09:
@MartyG-RealSense I’ve tried both methods building from source using native backend method and RSUSB backend method. The first didn’t work because of patches. The second Worked and here is the command that I used : 

sudo cmake -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -OpenGL_GL_PREFERENCE=LEGACY 

I followed the steps in this issue :  https://github.com/IntelRealSense/librealsense/issues/6710

MartyG-RealSense, 2022-07-09:
The build flag **-OpenGL_GL_PREFERENCE=LEGACY** is usually not used in librealsense source code builds and in a few cases where it is they have been for Raspberry Pi instead of Jetson.  It does not seem likely that this flag will be having an effect on the particular problems that you are experiencing in the ROS wrapper though.

Which model of Jetson board are you using, please?  If you have a Nano then Intel strongly recommend enabling the **barrel jack** connector to provide extra power using the instructions at the link below if the particular Nano model owned is equipped with a barrel jack.

https://jetsonhacks.com/2019/04/10/jetson-nano-use-more-power/

It is also common with librealsense source code builds on Jetson to enable support for CUDA graphics acceleration by including the build flag **-DBUILD_WITH_CUDA=true** in the CMake build instruction.

MartyG-RealSense, 2022-07-11:
@MariamFdil  Have you tried changing the USB cable to eliminate the cable as a possible cause of your warnings, please?

MariamFdil, 2022-07-11:
@MartyG-RealSense  I just tried nothing  has changed... 
PS : The same error occurs when I use the launch file rs_camera and  Realsense_viewer even if I have the image

MartyG-RealSense, 2022-07-18:
Hi @hugoyvrn @MariamFdil and @zhaojiaran111 Do you have updates about your problems that you can provide, please? Thanks!

MartyG-RealSense, 2022-07-19:
Hi @MariamFdil   That's excellent news that you achieved a solution  :)  Thanks for the update!

MartyG-RealSense, 2022-07-25:
Case closed due to no further comments received.

closed 2022-07-25 by MartyG-RealSense

kerolex, 2022-08-03:
Hi,

I am trying to acquire some RGB + depth image sequences with a Intel RealSense **D435i** device, but I got the same issue raised in this thread, when running roslaunch.

I created a custom rs_cam.launch that also sets the parameter **initial_reset:=true**.

Here details of my setup:
- Machine: Laptop Dell XPS 15
- OS: Linux Ubuntu 18.04.6 LTS (64 bit)
- Kernel: 4.15.0-189-generic
- ROS: melodic 1.14.13
- RealSense ROS v2.3.2
- LibRealSense v2.50.0

Note that installing RealSense-ROS with Method 1, I cannot see realsense-viewer from terminal. However, in a previous installation of the SDK followed by the installation of RealSense ROS, I could access realsense-viewer.
When using realsense-viewer the streaming of both RGB and depth was OK.

The USB ports are all 3.2.

Summary of the log is attached. 
Also note that my laptop tends to freeze any now and then after calling roslaunch.

FYI - I also tried all the procedure on a Dell All-in-One and I could acquired the images. So, the problem seems more specific to the laptop. 

Would you have any advise on how to fix the issue for the laptop, as I would need to acquire image sequences with some mobility? I am available to provide any addition information if needed.

[realsense-ros_launch.log](https://github.com/IntelRealSense/realsense-ros/files/9252576/realsense-ros_launch.log)

MartyG-RealSense, 2022-08-03:
Hi @kerolex  Was your previous SDK and wrapper installation on the same Dell XPS 15 computer as your later installation with Method 1, please?  Because Method 1 installs both the SDK and wrapper from packages at the same time, you may now have two different SDK installations on the same computer (built from source and packages) that may be conflicting.  For example, source and package builds on the same computer can result in a **Multiple realsense udev-rules were found!** warning.  The udev rules are responsible for device handling.

All RealSense SDK related packages on a Linux computer can be uninstalled with the command below:

**dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge**

kerolex, 2022-08-04:
Thank you for the previous reply @MartyG-RealSense 

The problem was/is still there even after the removal of the double installation, and re-installation of only realsense-ros. 
Would you have any other suggestions on how to solve this address?

P.S.: I also tried to reduce the frame resolution to 640x380, and framerate to 15 or 6 fps, with no success.

MartyG-RealSense, 2022-08-04:
@kerolex  As you were using a custom launch file **rs_cam.launch**, do you experience the same problem with the ROS wrapper's rs_camera.launch ?

EuguneHayashida, 2022-08-26:
I am using ROS Melodic and with ros wrapper 2.3.2 still encounter this problem.

My system:

Jetson AGX Xavier Jetpack 4.6
Ubuntu 18.04
Realsense SDK 2.51.1

(messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

MartyG-RealSense, 2022-08-26:
Hi @EuguneHayashida  I note that you are using librealsense version 2.51.1.  It is important to match a particular RealSense ROS wrapper version as closely as possible with the librealsense version listed on that particular wrapper's release notes.

The correct librealsense version to match with ROS1 wrapper version 2.3.2 is **2.50.0** as there is not a ROS1 wrapper for the new 2.51.1 SDK.  There are no longer updates of the ROS1 wrapper, as the RealSense ROS developer team's focus is now on the **ros2_beta** wrapper.

Do you also experience the **control_transfer returned** warning with librealsense version 2.50.0?

EuguneHayashida, 2022-08-28:
Hi. I have changed to 2.50.0 sdk and wrapped it using ROS1 2.3.2. The problem still persist.
I checked rviz to see whether it is publishing data. I only managed to receive raw depth image.

MartyG-RealSense, 2022-08-29:
@EuguneHayashida  Have you tried launching with the **rs_rgbd** ROS1 launch file using the instructions at https://github.com/IntelRealSense/realsense-ros/issues/2454#issuecomment-1221254065 please?

EuguneHayashida, 2022-08-29:
@MartyG-RealSense At first all failed to show raw color image, but I can show raw  color image using this line.

**roslaunch realsense2_camera rs_camera.launch Initial_reset:=true color_fps:=30**

MartyG-RealSense, 2022-08-30:
Thanks very much @EuguneHayashida for the update about your success in obtaining the color image.

StefanT83, 2022-10-01:
Hi, I got the same type of errors with my D455 on Ubuntu 20.04 and eventually managed to find a solution, so I decided to share my story here. My feeling is that there needs to be a match between the firmware of the D455 (I used 5.12.15.50) and librealsense2 (I used 2.51) and the ROS wrapper. 
Step1. removed any previous install related to 'realsense':
$ dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge
Step2. installed librealsense2 according to https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md > Installing the packages
Make sure to unplug the sensor at the end, wait 10 seconds, then plug it back again. Test 
$realsense-viewer 
and saw the depth camera is operational.
Step3. Check version of librealsense that was just installed:  
$dpkg -l|grep realsense
where I got 2.51
Step4. Firmware upgrade/downgrade of the D455 to match the librealsense 2.51
Now my interpretation is that we need to check the column 'SDK ver' on https://dev.intelrealsense.com/docs/firmware-releases and match that to, in my case, librealsense 2.51
Save the bin file and flash it to the D455 using https://www.intelrealsense.com/developers/  > Firmware update guide > 'Firmware Update Tool (rs-fw-update)'
Make sure to unplug the sensor at the end, wait 10 seconds, then plug it back again.
Step5. Install RealSense Wrapper https://github.com/IntelRealSense/realsense-ros > Method 2
Step6. Test in ROS:
$roslaunch realsense2_camera rs_camera.launch
Troubleshooting: in case of errors, make sure to unplug the sensor at the end, wait 10 seconds, then plug it back again.

I tested this approach on a D435 and was successfull too. It worked on Ubuntu 20.04 native as well as a virtual image on VMWare Player 16.
Cheers!

MartyG-RealSense, 2022-10-02:
Thanks so much @StefanT83 for sharing with the RealSense community the solution that worked for you!

MartyG-RealSense, 2022-12-13:
Thanks so much @HanzhiC for sharing the knowledge of what worked for you in your own situation!

snix00125, 2024-06-25:
Hi, I got the same type of error.

Ubuntu - Ubuntu 20.04.6 LTS
Distro - ROS Noetic
kERNEL - 5.10.104-tegra
ROS Version - 1.16.0
board - Jetson NX Xavier 
nvidia-jetpack version - 5.1.1-b56

I tried out all possible solutions from above threads, [#2467](https://github.com/IntelRealSense/realsense-ros/issues/2467) and all the relevant threads. Still i am unable to resolve my error.
@MartyG-RealSense 
I installed librealsense SDK as  you have mentioned in [#6710](https://github.com/IntelRealSense/librealsense/issues/6710). I tried multiple fresh installations. Below is my roslaunch rs_camera.launch output:

`

roslaunch realsense2_camera rs_camera.launch initial_reset:=true filters:=pointcloud
... logging to /home/art/.ros/log/8b9139e6-3326-11ef-b7f8-204ef6c5f2ab/roslaunch-art-desktop-11602.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://art-desktop:36905/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
[... 104 lines cut ...]
hwmon command 0x80( 5 0 0 0 ) failed (response -7= HW not ready)
[ INFO] [1719342625.418956841]: Done Setting Dynamic reconfig parameters.
[ INFO] [1719342625.421772945]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1719342625.432607692]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1719342625.432793135]: setupPublishers...
[ INFO] [1719342625.446420946]: Expected frequency for depth = 30.00000
[ INFO] [1719342625.590006587]: Expected frequency for color = 30.00000
[ INFO] [1719342625.702198466]: setupStreams...
 26/06 00:40:26,033 WARNING [281471931629936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1719342626.086007101]: SELECTED BASE:Depth, 0
[ WARN] [1719342626.159362360]: 
 26/06 00:40:26,221 WARNING [281471931629936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1719342626.259732182]: RealSense Node Is Up!
 26/06 00:40:26,404 WARNING [281471931629936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/06 00:40:26,455 WARNING [281471931629936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/06 00:40:26,517 WARNING [281471931629936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/06 00:40:26,569 WARNING [281471931629936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/06 00:40:26,620 WARNING [281471931629936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/06 00:40:26,672 WARNING [281471931629936] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 26/06 00:41:10,554 ERROR [281471814197616] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132

`
I feel lost after trying out all the possible solutions. Help me out!

Thanks in advance.

MartyG-RealSense, 2024-06-26:
Hi @snix00125   The camera firmware driver version listed in your log is 5.16.0.1.  The librealsense SDK 2.50.0 that you are using with wrapper 2.3.2 requires the older firmware version **5.13.0.50**, as firmwares are designed to work with specific librealsense versions.

MartyG-RealSense, 2024-06-26:
I note that you are enabling the pointcloud.  The ROS1 wrapper has an alternative method of publishing a pointcloud with an **RGBD** launch instead of using rs_camera.launch if you have not tried it already.  Instructions for an RGBD launch can be found at https://github.com/IntelRealSense/librealsense/issues/10223#issuecomment-1152864261

snix00125, 2024-06-26:
Thanks for the information!! I guess it worked.

I installed RGBD launch using:

Noetic
sudo apt-get install ros-noetic-rgbd-launch

Output:
`
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  gir1.2-goa-1.0 libfwupdplugin1 libxmlb1
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  ros-noetic-rgbd-launch
0 upgraded, 1 newly installed, 0 to remove and 8 not upgraded.
Need to get 9,184 B of archives.
After this operation, 56.3 kB of additional disk space will be used.
Get:1 http://packages.ros.org/ros/ubuntu focal/main arm64 ros-noetic-rgbd-launch arm64 2.4.0-1focal.20240111.192956 [9,184 B]
Fetched 9,184 B in 3s (3,204 B/s)                 
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package ros-noetic-rgbd-launch.
(Reading database ... 225523 files and directories currently installed.)
Preparing to unpack .../ros-noetic-rgbd-launch_2.4.0-1focal.20240111.192956_arm64.deb ...
Unpacking ros-noetic-rgbd-launch (2.4.0-1focal.20240111.192956) ...
Setting up ros-noetic-rgbd-launch (2.4.0-1focal.20240111.192956) ...
`

Then i checked the rostopic "/camera/depth_registered/points" and points were published. I also got the pointcloud2 output in rviz. But the error in terminal did not disappear. Do I just ignore or does it mean it's not fully functioning still? Below are some reference screenshots for your reference:

[image]
[image]

[51 more replies]