realsenseai/realsense-ros #2387: Slam Toolbox "Message Filter dropping message"

RNorlie, 2022-06-20:
Hello. Apologies if this is a repeat issue.

I'm attempting to run slam_toolbox with my D435i using depthimage_to_laserscan as a bridge for the data. The topics `/camera/depth/camera_info`, `/camera/depth/image_rect_raw`, and `/scan` all appear to be publishing properly, however instead of publishing to `/map`, the slam_toolbox terminal gives the following output:

```
[sync_slam_toolbox_node-1] [INFO] [1655749660.201251796] [slam_toolbox]: Message Filter dropping message: frame 'camera_depth_frame' at time 1655749660.155 for reason 'Unknown'
[sync_slam_toolbox_node-1] [INFO] [1655749660.234953534] [slam_toolbox]: Message Filter dropping message: frame 'camera_depth_frame' at time 1655749660.188 for reason 'Unknown'
[sync_slam_toolbox_node-1] [INFO] [1655749660.269013418] [slam_toolbox]: Message Filter dropping message: frame 'camera_depth_frame' at time 1655749660.222 for reason 'Unknown'

```

I'm unsure if I need to tweak some of the parameters within the realsense node, or the toolbox node. If this is not the appropriate place to ask for help, I can take this over to rosanswers, but I'd appreciate any insight you might have!

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i   | 
| Operating System & Version |   Linux (Ubuntu 20.04LTS) Jetpack 5.01| 
| ROS2 Distro | Foxy (Built from source) |
| Kernel Version                        |  5.10.65-tegra                                         | 
| Platform                                 | NVIDIA Jetson Xavier NX         |
| Architecture | Arm64 |
| SDK Version                            |  2.50.0                           | 

## Param Dump:

```
/slam_toolbox:
  ros__parameters:
	angle_variance_penalty: 1.0
	base_frame: base_footprint
	ceres_dogleg_type: TRADITIONAL_DOGLEG
	ceres_linear_solver: SPARSE_NORMAL_CHOLESKY
	ceres_loss_function: None
	ceres_preconditioner: SCHUR_JACOBI
	ceres_trust_strategy: LEVENBERG_MARQUARDT
	coarse_angle_resolution: 0.0349
	coarse_search_angle_offset: 0.349
	correlation_search_space_dimension: 0.5
	correlation_search_space_resolution: 0.01
	correlation_search_space_smear_deviation: 0.1
	debug_logging: false
	distance_variance_penalty: 0.5
	do_loop_closing: true
	enable_interactive_mode: true
	fine_search_angle_offset: 0.00349
	interactive_mode: false
	link_match_minimum_response_fine: 0.1
	link_scan_maximum_distance: 1.5
	loop_match_maximum_variance_coarse: 3.0
	loop_match_minimum_chain_size: 10
	loop_match_minimum_response_coarse: 0.35
	loop_match_minimum_response_fine: 0.45
	loop_search_maximum_distance: 3.0
	loop_search_space_dimension: 8.0
	loop_search_space_resolution: 0.05
	loop_search_space_smear_deviation: 0.03
	map_file_name: ''
	map_frame: map
	map_name: /map
	map_start_at_dock: null
	map_start_pose: null
	map_update_interval: 5.0
	minimum_angle_penalty: 0.9
	minimum_distance_penalty: 0.5
	minimum_time_interval: 0.5
	minimum_travel_distance: 0.5
	minimum_travel_heading: 0.5
	mode: mapping
	odom_frame: odom
	paused_new_measurements: false
	paused_processing: false
	resolution: 0.05
	scan_buffer_maximum_scan_distance: 10.0
	scan_buffer_size: 10
	scan_topic: /scan
	solver_plugin: solver_plugins::CeresSolver
	tf_buffer_duration: 30.0
	throttle_scans: 1
	transform_publish_period: 0.02
	transform_timeout: 0.2
	use_response_expansion: true
	use_scan_barycenter: true
	use_scan_matching: true
	use_sim_time: true
/camera/camera:
  ros__parameters:
	accel_fps: 63
	accel_info_qos: DEFAULT
	accel_qos: SENSOR_DATA
	align_depth:
  	enable: false
  	frames_queue_size: 16
	angular_velocity_cov: 0.01
	base_frame_id: link
	camera_name: camera
	clip_distance: -2.0
	color_info_qos: DEFAULT
	color_qos: SYSTEM_DEFAULT
	colorizer:
  	color_scheme: 0
  	enable: false
  	frames_queue_size: 16
  	histogram_equalization_enabled: true
  	max_distance: 6.0
  	min_distance: 0.0
  	stream_filter: 1
  	stream_format_filter: 1
  	stream_index_filter: -1
  	visual_preset: 0
	decimation_filter:
  	enable: false
  	filter_magnitude: 2
  	frames_queue_size: 16
  	stream_filter: 1
  	stream_format_filter: 1
  	stream_index_filter: -1
	depth_info_qos: DEFAULT
	depth_module:
  	auto_exposure_limit: 200000
  	auto_exposure_limit_toggle: false
  	auto_exposure_roi:
    	bottom: 479
    	left: 0
    	right: 847
    	top: 0
  	auto_gain_limit: 248
  	auto_gain_limit_toggle: false
  	emitter_always_on: false
  	emitter_enabled: 1
  	emitter_on_off: false
  	enable_auto_exposure: true
  	error_polling_enabled: true
  	exposure: 8500
  	frames_queue_size: 16
  	gain: 16
  	global_time_enabled: true
  	hdr_enabled: false
  	inter_cam_sync_mode: 0
  	laser_power: 150.0
  	output_trigger_enabled: false
  	profile: 848x480x30
  	sequence_id: 0
  	sequence_name: 0
  	sequence_size: 2
  	visual_preset: 0
	depth_qos: SYSTEM_DEFAULT
	device_type: ''
	diagnostics_period: 0.0
	disparity_filter:
  	enable: false
	disparity_to_depth:
  	enable: false
	enable_accel: true
	enable_color: true
	enable_depth: true
	enable_gyro: true
	enable_infra1: false
	enable_infra2: false
	enable_sync: false
	filter_by_sequence_id:
  	enable: false
  	frames_queue_size: 16
  	sequence_id: 1
	gyro_fps: 200
	gyro_info_qos: DEFAULT
	gyro_qos: SENSOR_DATA
	hdr_merge:
  	enable: false
  	frames_queue_size: 16
	hold_back_imu_for_frames: false
	hole_filling_filter:
  	enable: false
  	frames_queue_size: 16
  	holes_fill: 1
  	stream_filter: 1
  	stream_format_filter: 1
  	stream_index_filter: -1
	infra1_info_qos: DEFAULT
	infra1_qos: SYSTEM_DEFAULT
	infra2_info_qos: DEFAULT
	infra2_qos: SYSTEM_DEFAULT
	initial_reset: true
	json_file_path: ''
	linear_accel_cov: 0.01
	motion_module:
  	enable_motion_correction: true
  	frames_queue_size: 16
  	global_time_enabled: true
	pointcloud:
  	allow_no_texture_points: false
  	enable: false
  	filter_magnitude: 2
  	frames_queue_size: 16
  	ordered_pc: false
  	pointcloud_qos: DEFAULT
  	stream_filter: 2
  	stream_format_filter: 0
  	stream_index_filter: 0
	publish_odom_tf: true
	publish_tf: true
	reconnect_timeout: 6.0
	rgb_camera:
  	auto_exposure_priority: false
  	auto_exposure_roi:
    	bottom: 719
    	left: 0
    	right: 1279
    	top: 0
  	backlight_compensation: false
  	brightness: 0
  	contrast: 50
  	enable_auto_exposure: true
  	enable_auto_white_balance: true
  	exposure: 166
  	frames_queue_size: 16
  	gain: 64
  	gamma: 300
  	global_time_enabled: true
  	hue: 0
  	power_line_frequency: null
  	profile: 1280x720x30
  	saturation: 64
  	sharpness: 50
  	white_balance: 4600.0
	rosbag_filename: ''
	serial_no: ''
	spatial_filter:
  	enable: false
  	filter_magnitude: 2
  	filter_smooth_alpha: 0.5
  	filter_smooth_delta: 20
  	frames_queue_size: 16
  	holes_fill: 0
  	stream_filter: 1
  	stream_format_filter: 1
  	stream_index_filter: -1
	temporal_filter:
  	enable: false
  	filter_smooth_alpha: 0.4000000059604645
  	filter_smooth_delta: 20
  	frames_queue_size: 16
  	holes_fill: 3
  	stream_filter: 1
  	stream_format_filter: 1
  	stream_index_filter: -1
	tf_publish_rate: 0.0
	unite_imu_method: 0
	usb_port_id: ''
	use_sim_time: false
	wait_for_device_timeout: -1.0
/depthimage_to_laserscan_node:
  ros__parameters:
	output_frame: camera_depth_frame
	range_max: 10.0
	range_min: 0.45
	scan_height: 1
	scan_time: 0.033
	use_sim_time: false
```

[image]

MartyG-RealSense, 2022-06-21:
Hi @RNorlie  At the ROS Answers link below there was another person who experienced the **Message Filter dropping message: frame 'camera_depth_frame'** warning with a RealSense D455 camera, the SLAM Toolbox and depth_image_to_laserscan.  In their case though, the reason given was 'discarding message because the queue is full' instead of 'unknown'.

https://answers.ros.org/question/393773/slam-toolbox-message-filter-dropping-message-for-reason-discarding-message-because-the-queue-is-full/

The user in that case found that setting depth in the RealSense ROS wrapper to 15 FPS instead of 30 reduced the frequency of the warning but did not eliminate it entirely.

RNorlie, 2022-06-21:
Hi @MartyG-RealSense, thanks for your response!

 I was unsure which setting the user altered in the link that you shared. These are the only available parameters relating depth in rqt. Unless I am supposed to configure something outside of the node, I'm not sure what would accomplish this.

[image]

This user's situation appears to be slightly different as they did have some messages being published to the `/map` topic, whereas none are being published in my setup. 

running `top` shows that the realsense2_camera node is at ~80% cpu usage, and no other nodes take up a significant amount. 

The user did mention the potential of there being a tf tree issue. I'll look into this, as I am receiving warnings from RVIZ.

[image]

MartyG-RealSense, 2022-06-21:
Thanks very much for the further information.  As neither yourself or the case that I linked to directly mention the RealSense ROS wrapper, may I confirm whether your **/camera/depth/image_rect_raw** topic is being published by the ROS wrapper by using a launch instruction such as **roslaunch realsense2_camera rs_camera.launch**

If you are using a RealSense ROS wrapper launch to publish the depth topic then the depth stream FPS could be custom-configured from the roslaunch instruction or by editing the launch file.

RNorlie, 2022-06-21:
Hello,

> may I confirm whether your **/camera/depth/image_rect_raw** topic is being published by the ROS wrapper by using a launch instruction such as **roslaunch realsense2_camera rs_camera.launch**

Yes, I am launching the node using the instruction: `ros2 launch realsense2_camera rs_launch.py enable_gyro:=true enable_accel:=true initial_reset:=true &`

`Running ros2 topic echo /camera/depth/image_rect_raw` confirms that the node is indeed publishing data to the topic.

RNorlie, 2022-06-21:
As an update, I have modified the tf tree to resolve the rviz warnings, and I am now receiving a different output from my slam_toolbox terminal:

```
[sync_slam_toolbox_node-1] Warning: Invalid frame ID "base_footprint" passed to canTransform argument source_frame - frame does not exist
[sync_slam_toolbox_node-1]      	at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[sync_slam_toolbox_node-1] [WARN] [1655843318.026403848] [slam_toolbox]: Failed to compute odom pose
[sync_slam_toolbox_node-1] Warning: Invalid frame ID "base_footprint" passed to canTransform argument source_frame - frame does not exist
[sync_slam_toolbox_node-1]      	at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[sync_slam_toolbox_node-1] [WARN] [1655843318.093144103] [slam_toolbox]: Failed to compute odom pose

```

More troubleshooting to come tomorrow.

RNorlie, 2022-06-22:
Hello @MartyG-RealSense I have looked into the warning:
>[WARN] [1655843318.093144103] [slam_toolbox]: Failed to compute odom pose

using `ros2 topic echo /camera/imu` reveals that no odometry data is being published. I tried using `unite_imu_method:=copy` and `unite_imu_method:=linear_interpolation` but neither option showed data being published. 

If I omit `unite_imu_method` and echo `/camera/gyro/imu_info` and `/camera/accel/imu_info` I see no data either.

I looked to calibrate the imu, as I have not done that yet, but I was unable to run the calibration script as I did not install librealsense with pyrealsense2.

So I am left with my latest issue of trying to rebuild librealsense with the `-DBUILD_PYTHON_BINDINGS:bool=true` option. My build has not been successful, and I have been unable to find a workaround or anyone with a similar setup of the Jetson Xavier NX with Jetpack 5.01 (Ubuntu 20.04). 

I receive the following error after inputting: 

`cmake ../ -DCMAKE_BUILD_TYPE=Release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA=true -DFORCE_RSUSB_BACKEND:=false -DBUILD_PYTHON_BINDINGS:bool=true 
`

............

```
HEAD is now at 8de7772c chore: prepare for the 2.6.2 release (#2821)
[ 37%] No patch step for 'pybind11'
[ 50%] Performing update step for 'pybind11'
[ 62%] No configure step for 'pybind11'
[ 75%] No build step for 'pybind11'
[ 87%] No install step for 'pybind11'
[100%] Completed 'pybind11'
[100%] Built target pybind11
-- pybind11 v2.6.2
-- Found PythonInterp: /usr/bin/python3.9 (found version "3.9.5")
-- Found PythonLibs: python3.9
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
CMake Error at /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
  Could NOT find Python (missing: Python_INCLUDE_DIRS Development) (found
  version "2.7.18")
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:393 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.16/Modules/FindPython.cmake:347 (find_package_handle_standard_args)
  wrappers/python/CMakeLists.txt:63 (find_package)

-- Configuring incomplete, errors occurred!
```

Let me know if you have any recommendations as to how to help. Thank you for your continued support!

MartyG-RealSense, 2022-06-23:
1.  The IMU of a RealSense camera can function in the realsense-viewer tool and rs-motion without having to be calibrated first.

2.  Some RealSense users use a tool called **Kalibr** for IMU calibration. 

https://github.com/ethz-asl/kalibr

As mentioned above though, a calibration is not required in order for IMU data to be displayed in realsense-viewer or rs-motion.

3.  I note that you have already tried echoing the **imu** topic and the gyro / accel topics individually to check whether they were publishing and they were not.

I also note that you were not able to display IMU data in realsense-viewer.  Are there any warning / error messages in the log when you enable the **Motion Module** in the Viewer?  The Viewer's log can be expanded open by left-clicking on the small upward triangle at the bottom corner of the Viewer window.

[image]

RNorlie, 2022-06-23:
Thank you for the information! I will have to look into that third-party tool!

Here is the log when I enable the **Motion Module** in realsense-viewer:

[image]

And my terminal window prints this message:
 
23/06 10:55:58,421 WARNING [281473115254800] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
 23/06 10:55:58,442 WARNING [281472493679024] (backend-hid.cpp:715) HID set_power 1 failed for /sys/devices/platform/3610000.xhci/usb2/2-3/2-3.1/2-3.1:1.5/0003:8086:0B3A.0004/HID-SENSOR-200073.2.auto/iio:device0/buffer/enable

RNorlie, 2022-06-23:
Sorry, my mistake
[image]
This does display the data! 

So the imu is functioning in the viewer and rs-motion, but I am still unable to echo the data from a ros topic.

Do you have any ideas as to why that might be?
Thanks for all your help!

MartyG-RealSense, 2022-06-23:
Your results make sense.  A **Motion Module Failure** notification in the Viewer does not prevent the IMU data from working normally, but it can make the IMU topics in the ROS wrapper fail to provide data.

Going back to your use of JetPack 5 for its Ubuntu 20.04 support, the most recently supported JetPack in the RealSense SDK is **L4T 32.6.1 (JetPack 4.6)**.  This support was added in the current 2.50.0 librealsense SDK.  So I could not rule out that the ROS problem with IMU data is related to use of JetPack 5.

There was also a case of ROS Motion Module Failure at https://github.com/IntelRealSense/realsense-ros/issues/1960#issuecomment-875801977 where the cause of the failure may have been related to doing a **rosbag record** of a lot of topics, as reducing the number of topics being recorded apparently solved the error.  Are you performing any recording?

Also, have you tried accessing the IMU topics without slam_toolbox and depthimage_to_laserscan being active?

RNorlie, 2022-06-23:
@MartyG-RealSense I do have access to an L515 as well. This camera does not have the **Motion Module Failure** message, but it does display this message upon launching the node.

`[realsense2_camera_node-1]  23/06 13:53:29,091 WARNING [281471730301344] (backend-hid.cpp:715) HID set_power 1 failed for /sys/devices/platform/3610000.xhci/usb2/2-3/2-3.1/2-3.1:1.7/0003:8086:0B64.000D/HID-SENSOR-200073.2.auto/iio:device0/buffer/enable`

I do not see any data when I echo **/camera/imu** with the different `unite_imu_method` settings nor **/camera/gyro/imu_info** and **/camera/accel/imu_info** when I omit `unite_imu_method`.

I am able to echo data from **/camera/gyro/sample** and **/camera/accel/sample** though, so perhaps the odometry from the L515 will be more viable for SLAM?

When I try to run the camera, depthimage_to_laserscan, and slam_toolbox nodes with this device, I receive the same message as I did with the D435i:

```
[sync_slam_toolbox_node-1] [INFO] [1656011180.041164815] [slam_toolbox]: Message Filter dropping message: frame 'camera_depth_frame' at time 1656011177.902 for reason 'Unknown'
[sync_slam_toolbox_node-1] [INFO] [1656011180.073619057] [slam_toolbox]: Message Filter dropping message: frame 'camera_depth_frame' at time 1656011177.936 for reason 'Unknown'
[sync_slam_toolbox_node-1] [INFO] [1656011180.106913075] [slam_toolbox]: Message Filter dropping message: frame 'camera_depth_frame' at time 1656011177.969 for reason 'Unknown'
```

I suspect there is a simple fix to this issue, but I'm not familiar enough with ros2 or these devices. Thank you for helping me get this far!

MartyG-RealSense, 2022-06-24:
A RealSense team member advises a D435i owner at https://github.com/IntelRealSense/librealsense/issues/5901 who was experiencing the **(backend-hid.cpp:715) HID set_power 1 failed** message that it is for information only and can be ignored if there is not the problem of no IMU data.  If there is also a problem with IMU data then unplugging and re-inserting the camera should correct the problem.  Though the RealSense user in that case did report that they had to repeat the unplug-replug each time that they started a new roslaunch.

RNorlie, 2022-06-24:
I've tried unplugging and replugging the camera in after starting the node, and the warning persisted, and the only way I am able to obtain imu data in ros is still from topics **/camera/accel/sample** and **/camera/gyro/sample** Perhaps there is a way to utilize these topics through remapping or using some other package that subscribes to these topics.

MartyG-RealSense, 2022-06-24:
Is there any difference if you add **enable_sync:=true** to your ros launch instruction?

When set to true, this setting gathers the closest frames of different sensors to be sent with the same time tag.

RNorlie, 2022-06-24:
That causes me to receive the same output from the slam_toolbox terminal, however the messages seem to appear at a slightly different time interval than they did in the past:

```
ros2 launch slam_toolbox online_sync_launch.
[INFO] [launch]: All log files can be found below /home/jetson_dev/.ros/log/2022-06-24-10-58-21-876407-jetson-dev-desktop-8549
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [sync_slam_toolbox_node-1]: process started with pid [8555]
[sync_slam_toolbox_node-1] [INFO] [1656086302.941716214] [slam_toolbox]: Node using stack size 40000000
[sync_slam_toolbox_node-1] [INFO] [1656086303.218250171] [slam_toolbox]: Using solver plugin solver_plugins::CeresSolver
[sync_slam_toolbox_node-1] [INFO] [1656086303.218785468] [slam_toolbox]: CeresSolver: Using SCHUR_JACOBI preconditioner.
[sync_slam_toolbox_node-1] [INFO] [1656086303.404230764] [slam_toolbox]: Message Filter dropping message: frame 'camera_depth_frame' at time 1656086301.957 for reason 'Unknown'
[sync_slam_toolbox_node-1] [INFO] [1656086303.446686129] [slam_toolbox]: Message Filter dropping message: frame 'camera_depth_frame' at time 1656086302.056 for reason 'Unknown'
[sync_slam_toolbox_node-1] [INFO] [1656086303.476397154] [slam_toolbox]: Message Filter dropping message: frame 'camera_depth_frame' at time 1656086302.090 for reason 'Unknown'
[sync_slam_toolbox_node-1] [INFO] [1656086303.509439800] [slam_toolbox]: Message Filter dropping message: frame 'camera_depth_frame' at time 1656086302.123 for reason 'Unknown'
[sync_slam_toolbox_node-1] [INFO] [1656086303.567219959] [slam_toolbox]: Message Filter dropping message: frame 'camera_depth_frame' at time 1656086302.156 for reason 'Unknown'
[sync_slam_toolbox_node-1] [INFO] [1656086303.602067024] [slam_toolbox]: Message Filter dropping message: frame 'camera_depth_frame' at time 1656086302.222 for reason 'Unknown'
[sync_slam_toolbox_node-1] [INFO] [1656086303.635250438] [slam_toolbox]: Message Filter dropping message: frame 'camera_depth_frame' at time 1656086302.255 for reason 'Unknown'
[sync_slam_toolbox_node-1] [INFO] [1656086303.667212827] [slam_toolbox]: Message Filter dropping message: frame 'camera_depth_frame' at time 1656086302.289 for reason 'Unknown'

```

My Camera node terminal displays the following:

```
jetson_dev@jetson-dev-desktop:~/ros2_ws$ ros2 launch realsense2_camera rs_launch.py enable_gyro:=true enable_accel:=true initial_reset:=true unite_imu_method:=linear_interpolation enable_sync:=true &
[1] 7789
jetson_dev@jetson-dev-desktop:~/ros2_ws$ [INFO] [launch]: All log files can be found below /home/jetson_dev/.ros/log/2022-06-24-10-52-40-884634-jetson-dev-desktop-7789
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [7791]
[realsense2_camera_node-1] [INFO] [1656085961.966865420] [camera.camera]: RealSense ROS v4.0.4
[realsense2_camera_node-1] [INFO] [1656085961.967203884] [camera.camera]: Built with LibRealSense v2.50.0
[realsense2_camera_node-1] [INFO] [1656085961.967294412] [camera.camera]: Running with LibRealSense v2.50.0
[realsense2_camera_node-1] [INFO] [1656085962.319888377] [camera.camera]: Device with serial number f1120471 was found.
[realsense2_camera_node-1]
[realsense2_camera_node-1] [INFO] [1656085962.320090585] [camera.camera]: Device with physical ID /sys/devices/platform/3610000.xhci/usb2/2-3/2-3.2/2-3.2:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1656085962.320163514] [camera.camera]: Device with name Intel RealSense L515 was found.
[realsense2_camera_node-1] [INFO] [1656085962.321647068] [camera.camera]: Device with port number 2-3.2 was found.
[realsense2_camera_node-1] [INFO] [1656085962.321756476] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1656085962.321869468] [camera.camera]: Resetting device...
[realsense2_camera_node-1] [INFO] [1656085968.748827272] [camera.camera]: Device with serial number f1120471 was found.
[realsense2_camera_node-1]
[realsense2_camera_node-1] [INFO] [1656085968.749032744] [camera.camera]: Device with physical ID /sys/devices/platform/3610000.xhci/usb2/2-3/2-3.2/2-3.2:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1656085968.749106824] [camera.camera]: Device with name Intel RealSense L515 was found.
[realsense2_camera_node-1] [INFO] [1656085968.750434474] [camera.camera]: Device with port number 2-3.2 was found.
[realsense2_camera_node-1] [INFO] [1656085968.750551594] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1656085968.768223715] [camera.camera]: getParameters...
[realsense2_camera_node-1] [WARN] [1656085968.770364262] [camera.camera]: Could not set param: unite_imu_method with 0 Range: [0, 2]: parameter 'unite_imu_method' has invalid type: expected [integer] got [string]
[realsense2_camera_node-1] [INFO] [1656085968.770534246] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1656085968.770648454] [camera.camera]: Device Name: Intel RealSense L515
[realsense2_camera_node-1] [INFO] [1656085968.770752998] [camera.camera]: Device Serial No: f1120471
[realsense2_camera_node-1] [INFO] [1656085968.770862247] [camera.camera]: Device physical port: /sys/devices/platform/3610000.xhci/usb2/2-3/2-3.2/2-3.2:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1656085968.770934791] [camera.camera]: Device FW version: 01.05.08.01
[... cut ...]
[realsense2_camera_node-1]  24/06 10:52:49,500 WARNING [281472510429600] (backend-hid.cpp:715) HID set_power 1 failed for /sys/devices/platform/3610000.xhci/usb2/2-3/2-3.2/2-3.2:1.7/0003:8086:0B64.0008/HID-SENSOR-200073.2.auto/iio:device0/buffer/enable
[realsense2_camera_node-1] [INFO] [1656085969.508240882] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1656085969.510480949] [camera.camera]:
[realsense2_camera_node-1] [WARN] [1656085969.510721782] [camera.camera]: frame's time domain is HARDWARE_CLOCK. Timestamps may reset periodically.
```

MartyG-RealSense, 2022-06-25:
Noting that the documentation for slam_toolbox refers to ROS2 **Eloquent** but not the Foxy version that you are using, I wondered whether the issue may be related to Foxy. 

https://github.com/SteveMacenski/slam_toolbox

A Foxy user of slam_toolbox who was dropping messages found that a solution that worked for them was to set the time with the command **sudo date -s$(date -Ins)**

https://www.reddit.com/r/ROS/comments/tth202/ros2_foxy_issues_with_using_slam_toolbox_to_map/

That user was using the **robot_localization** package, which Intel's own D435i SLAM guide also makes use of.

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

My understanding is that robot_localization is one way in which the odom can be published.

https://navigation.ros.org/setup_guides/odom/setup_odom.html

MartyG-RealSense, 2022-06-28:
It's excellent news that you were able to receive data from the /imu topic after adjusting the unite_imu_method setting!  

I look forward to hearing the results of your testing with robot_localization.  Good luck!

RNorlie, 2022-06-29:
Hi @MartyG-RealSense, I've been experimenting with the robot_localization package. I have not been able to create an 
**odom->base_link** transform other than by publishing my own via a static_tf_publisher. This does not resolve the error message from slam_toolbox, as I am still not providing any odometry data.

I am looking into utilizing wheel odometry later on and simulating my own odometry in gazebo for now, but I was curious if there were any alternatives to this as well, perhaps using the LiDAR from the L515. 

Are you aware of any implementations of the L515 to obtain odometry within ROS2?

I have not seen many examples, and those that I have noticed are for ROS1 only: [SSL SLAM](https://github.com/wh200720041/SSL_SLAM)

If you have seen any promising applications, let me know! otherwise I will likely move forward with investing in some encoders for wheel odometry and close the issue. 

Thank you!

RNorlie, 2022-06-29:
Thank you for your help! 
I'm going to begin exploring other options for odometry and/or different slam packages.

Thank you for helping me reach full functionality with my realsense cameras.
I may open a new issue later if I find that I also need the D435i imu functioning, but with the L515 imu working, I don't suspect that will be a problem.

closed 2022-06-29 by RNorlie

MartyG-RealSense, 2022-06-29:
You are very welcome  :)

ryanmaxwell96, 2022-12-18:
Did you get slam to work with Jetpack 5.1? Or did you have to revert to Jetpack 4.6 and do some manipulation to get Ubuntu 20.04 and Foxy?

RNorlie, 2022-12-21:
> Did you get slam to work with Jetpack 5.1? Or did you have to revert to Jetpack 4.6 and do some manipulation to get Ubuntu 20.04 and Foxy?

I was able to get slam_toolbox working with JetPack 5.01, give it a shot!

ryanmaxwell96, 2022-12-24:
> > Did you get slam to work with Jetpack 5.1? Or did you have to revert to Jetpack 4.6 and do some manipulation to get Ubuntu 20.04 and Foxy?
> 
> 
> 
> @ryanmaxwell96,
> 
> I was able to get slam_toolbox working with JetPack 5.01, give it a shot!

That's great! What camera did you use?

ryanmaxwell96, 2022-12-24:
Sorry dumb question. I see at the top you used D435i

furkansariyildiz, 2023-08-07:
Hello guys, I had same problem. You should publish your odometry value to /odom topic with time stamp. I mean there is a problem on your TF tree. I hope it solves your problem.