realsenseai/realsense-ros #2388: D455 and ROS2 on Jetson Xavier AGX

MrOCW, 2022-06-21:
Hi, I have enabled pointcloud and align depth but there is no image on /camera/aligned_depth_to_color/image_raw and there is no point cloud topic at all using ros2 topic list. The RGB images and IMU works fine.

```
[realsense2_camera_node-1] [INFO] [1655772882.805624557] [camera.camera]: RealSense ROS v4.0.4
[realsense2_camera_node-1] [INFO] [1655772882.805855097] [camera.camera]: Built with LibRealSense v2.50.0
[realsense2_camera_node-1] [INFO] [1655772882.805916188] [camera.camera]: Running with LibRealSense v2.50.0
[realsense2_camera_node-1] [INFO] [1655772882.880980698] [camera.camera]: Device with serial number 207322250954 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1655772882.881190117] [camera.camera]: Device with physical ID /sys/devices/platform/3610000.xhci/usb2/2-1/2-1.3/2-1.3:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1655772882.881271497] [camera.camera]: Device with name Intel RealSense D455 was found.
[realsense2_camera_node-1] [INFO] [1655772882.882483848] [camera.camera]: Device with port number 2-1.3 was found.
[realsense2_camera_node-1] [INFO] [1655772882.882557004] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1655772882.882618319] [camera.camera]: Resetting device...
[realsense2_camera_node-1] [INFO] [1655772889.089706135] [camera.camera]: Device with serial number 207322250954 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1655772889.089877024] [camera.camera]: Device with physical ID /sys/devices/platform/3610000.xhci/usb2/2-1/2-1.3/2-1.3:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1655772889.089932323] [camera.camera]: Device with name Intel RealSense D455 was found.
[realsense2_camera_node-1] [INFO] [1655772889.091038908] [camera.camera]: Device with port number 2-1.3 was found.
[realsense2_camera_node-1] [INFO] [1655772889.091117120] [camera.camera]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1655772889.097136537] [camera.camera]: getParameters...
[realsense2_camera_node-1] [INFO] [1655772889.098149038] [camera.camera]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1655772889.098265172] [camera.camera]: Device Name: Intel RealSense D455
[realsense2_camera_node-1] [INFO] [1655772889.098383226] [camera.camera]: Device Serial No: 207322250954
[realsense2_camera_node-1] [INFO] [1655772889.098470559] [camera.camera]: Device physical port: /sys/devices/platform/3610000.xhci/usb2/2-1/2-1.3/2-1.3:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1655772889.098537794] [camera.camera]: Device FW version: 05.13.00.50
[realsense2_camera_node-1] [INFO] [1655772889.098618438] [camera.camera]: Device Product ID: 0x0B5C
[realsense2_camera_node-1] [INFO] [1655772889.098663305] [camera.camera]: Sync Mode: On
[realsense2_camera_node-1] [INFO] [1655772889.269765798] [camera.camera]: Set Exposure.1 to 7500
[realsense2_camera_node-1] [WARN] [1655772889.296555575] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [WARN] [1655772889.313208024] [camera.camera]: Could not set param: rgb_camera.power_line_frequency with 3 Range: [0, 2]: parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {} doesn't comply with integer range.
[realsense2_camera_node-1] [WARN] [1655772889.333493143] [camera.camera]: re-enable the stream for the change to take effect.
[realsense2_camera_node-1] [INFO] [1655772889.341041791] [camera.camera]: Set ROS param gyro_fps to default: 200
[realsense2_camera_node-1] [INFO] [1655772889.341588251] [camera.camera]: Set ROS param accel_fps to default: 100
[realsense2_camera_node-1] [INFO] [1655772889.429231943] [camera.camera]: Start Sensor: Stereo Module
[realsense2_camera_node-1] [INFO] [1655772889.543188810] [camera.camera]: Open profile: stream_type: Depth(0), Format: Z16, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1655772889.543709349] [camera.camera]: Start Sensor: RGB Camera
[realsense2_camera_node-1] [INFO] [1655772889.607801257] [camera.camera]: Open profile: stream_type: Color(0), Format: RGB8, Width: 1280, Height: 720, FPS: 30
[realsense2_camera_node-1] [INFO] [1655772889.608026900] [camera.camera]: Start Sensor: Motion Module
[realsense2_camera_node-1]  21/06 00:54:49,610 WARNING [281472380570016] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[realsense2_camera_node-1] [INFO] [1655772889.624815517] [camera.camera]: Open profile: stream_type: Accel(0)Format: MOTION_XYZ32F, FPS: 100
[realsense2_camera_node-1] [INFO] [1655772889.625075978] [camera.camera]: Open profile: stream_type: Gyro(0)Format: MOTION_XYZ32F, FPS: 200
[realsense2_camera_node-1] [INFO] [1655772889.630744433] [camera.camera]: RealSense Node Is Up!
[realsense2_camera_node-1] [WARN] [1655772889.641866067] [camera.camera]: 
[realsense2_camera_node-1]  21/06 00:54:49,643 WARNING [281471631595936] (backend-hid.cpp:715) HID set_power 1 failed for /sys/devices/platform/3610000.xhci/usb2/2-1/2-1.3/2-1.3:1.5/0003:8086:0B5C.0009/HID-SENSOR-200073.2.auto/iio:device0/buffer/enable
[realsense2_camera_node-1] [WARN] [1655772890.497657717] [camera.camera]: XXX Hardware Notification:Motion Module failure,1.65577e+12,Error,Hardware Error
[realsense2_camera_node-1] [WARN] [1655772890.498679018] [camera.camera]: Hardware Notification:Motion Module failure,1.65577e+12,Error,Hardware Error
```

[image]

MartyG-RealSense, 2022-06-21:
Hi @MrOCW  There are known issues with using pointclouds or align_depth on Jetson boards specifically - as described in detail at https://github.com/IntelRealSense/realsense-ros/issues/1967 - and Intel are aware of the issue, which does not occur on desktop and laptop PCs.  

For ROS1, using librealsense **2.43.0** and wrapper **2.2.23** is the best available solution at the time of writing this.  For ROS2 though there is not an equivalent solution available unfortunately.

closed 2022-06-22 by MrOCW

iraadit, 2022-07-20:
Hi @MrOCW , did you find a working configuration with ROS2?
Thanks

MrOCW, 2022-07-20:
@iraadit I am currently using ROS1 just for the realsense and bridging the topics over to ROS2. librealsense 2.50 and realsense-ros 2.3.2

iraadit, 2022-07-25:
> @iraadit I am currently using ROS1 just for the realsense and bridging the topics over to ROS2. librealsense 2.50 and realsense-ros 2.3.2

Thank you @MrOCW, I'll try that in the coming days if I don't find a native solution with  ROS2.

Our open issue could interest you: https://github.com/IntelRealSense/realsense-ros/issues/2396