realsenseai/realsense-ros #2390: Build problem on D455 on Xavier NX Jetpack 5.01 - OpenCV

martinerk0, 2022-06-21:
Hi, I recently installed Jetpack 5.01 on Jetson Xavier NX ( Ubuntu 20.04 and ROS Noetic )
Firstly I installed realsense lib and ros wrapper using _sudo apt-get_ and while it worked, I thought pointcloud was very low quality and low fps (or pps? - pointcloud per seconds :D ) (around 2 hz with rostopic hz)

I tried to build from source [librealsense](https://github.com/IntelRealSense/librealsense) and it worked with [librealsense](https://github.com/IntelRealSense/librealsense) script.

Then I built realsense-ros from source, but when I wanted to run `roslaunch realsense2_camera rs_camera.launch` I got error:
`/opt/ros/noetic/lib/nodelet/nodelet: symbol lookup error: /home/martinerk0/catkin_ws/devel/lib//librealsense2_camera.so: undefined symbol: _ZN2cv3MatC1Ev`

I found info at some [chinese website](https://blog.csdn.net/weixin_45703465/article/details/122977877) that It might me OpenCV dependency issue, but it didn't quite solve it, but because I build some package with OpenCV previously I added this to [CMakeLists.txt](https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/CMakeLists.txt)

```
find_package(OpenCV REQUIRED)

include_directories(
    ${OpenCV_INCLUDE_DIRS}
)

target_link_libraries(${PROJECT_NAME}
    ${realsense2_LIBRARY}
    ${catkin_LIBRARIES}
    ${CMAKE_THREAD_LIBS_INIT}
    ${OpenCV_LIBRARIES}
    )
```
After building with cmake everything runs fine. Only problem is it is still the same hz arond 2-3 but I think pointclouds generated are "cleaner". Or I don't know, I'm just happy that I resolved it and maybe this could prove useful for future Jetpack 5 users who will want to build from source. What do you think caused this OpenCV dependency issue? I have not had any problems usng apt-get.

MartyG-RealSense, 2022-06-22:
Thanks very much @martinerk0 for sharing the method that worked for you in resolving your OpenCV issue in the RealSense ROS wrapper!  It is unclear why the **_ZN2cv3MatC1Ev** occurs on rare occasions, though one possibility is if a different version of OpenCV is installed after OpenCV has already been installed.

Issues with pointcloud, and also depth-color alignment, on Jetson boards specifically are a known issue in the RealSense ROS wrapper (desktop / laptop PCs are not affected by it).  The problems can include very low FPS of around 3 or missing color.  For ROS1 the current best solution is to use librealsense **2.43.0** and wrapper **2.3.23**.  More information can be found at https://github.com/IntelRealSense/realsense-ros/issues/1967

RNorlie, 2022-06-22:
@martinerk0 Thank you for your response. I'll open my own issue!

martinerk0, 2022-06-22:
@MartyG-RealSense Btw, it is interesting that realsense-viewer on Xavier NX runs at 30fps but ROS wrapper only at 2-3. I don't think it could be just overhead of packing data into the messages, I will try your (  librealsense 2.43.0 and wrapper 2.2.23 ) recommendation when I get another Jetson device and see if it helps.

martinerk0, 2022-06-30:
No, you can close this. Thanks!

MartyG-RealSense, 2022-06-30:
Thanks very much @martinerk0 for the update!  As you suggested, I will close the case.  Thanks again!

closed 2022-06-30 by MartyG-RealSense

Hypothesis-Z, 2023-03-08:
> Hello, @martinerk0. I was just trying to install librealsense with the same setup. I have had success in the past with this setup, but I am trying once again to build with the additional option: `-DBUILD_PYTHON_BINDINGS:bool=true `, but I am receiving the following errors:
> 
> ```
> HEAD is now at 8de7772c chore: prepare for the 2.6.2 release (#2821)
> [ 37%] No patch step for 'pybind11'
> [ 50%] Performing update step for 'pybind11'
> [ 62%] No configure step for 'pybind11'
> [ 75%] No build step for 'pybind11'
> [ 87%] No install step for 'pybind11'
> [100%] Completed 'pybind11'
> [100%] Built target pybind11
> -- pybind11 v2.6.2
> -- Found PythonInterp: /usr/bin/python3.9 (found version "3.9.5")
> -- Found PythonLibs: python3.9
> -- Performing Test HAS_FLTO
> -- Performing Test HAS_FLTO - Success
> CMake Error at /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
>   Could NOT find Python (missing: Python_INCLUDE_DIRS Development) (found
>   version "2.7.18")
> Call Stack (most recent call first):
>   /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:393 (_FPHSA_FAILURE_MESSAGE)
>   /usr/share/cmake-3.16/Modules/FindPython.cmake:347 (find_package_handle_standard_args)
>   wrappers/python/CMakeLists.txt:63 (find_package)
> 
> 
> -- Configuring incomplete, errors occurred!
> ```
> 
> Were you able to build with this option, or did you omit it?

Hi @RNorlie  How did you slove the problem? It looks like python 3.8 is not found. Here is my error info:
```
-- pybind11 v2.6.1
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.10")
-- Found PythonLibs: /usr/lib/aarch64-linux-gnu/libpython3.8.so
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
CMake Error at /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:146 (message):
  Could NOT find Python (missing: Python_INCLUDE_DIRS Development) (found
  version "2.7.18")
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPackageHandleStandardArgs.cmake:393 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.16/Modules/FindPython.cmake:347 (find_package_handle_standard_args)
  wrappers/python/CMakeLists.txt:62 (find_package)
```

arnoluken, 2024-05-16:
Hello everyone, same problem here: i am stuck here for almost a day. can anyone help? -- Found PythonInterp: /usr/bin/python3.8 (found suitable version "3.8.10", minimum required is "3.6") 
-- Found PythonLibs: /usr/lib/libpython3.8.so
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
CMake Error at /usr/local/share/cmake-3.27/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Python (missing: Python_INCLUDE_DIRS Development
  Development.Module Development.Embed) (found version "2.7.18")
Call Stack (most recent call first):
  /usr/local/share/cmake-3.27/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/share/cmake-3.27/Modules/FindPython.cmake:616 (find_package_handle_standard_args)
  CMake/external_pybind11.cmake:74 (find_package)
  CMake/external_pybind11.cmake:121 (get_pybind11)
  CMake/global_config.cmake:68 (include)
  CMakeLists.txt:46 (global_set_flags)

MartyG-RealSense, 2024-05-16:
Hi @arnoluken  A RealSense user with Python 3.8 who experienced the same error messages posted the full solution that worked in their particular case at https://github.com/IntelRealSense/librealsense/issues/12626#issuecomment-1923212052