realsenseai/realsense-ros #2391: Realsense Image Turns White

dverstandig, 2022-06-22:
Hello, 

I am using a Realsense D455 and I am having issues when bringing up the Realsense camera through ROS. 

Every so often, the ROS image will look like this:

[image]

We have a perception stack running using the Realsense ROSimage, so having this white image is pretty problematic. 
The resolution so far has been to re-launch the camera until this issue goes away, but we would like a more permanent fix. 

The camera is running on an Nvidia Jetson with RealSense ROS v2.3.2 and LibRealSense v2.50.0. 

I've also attached a copy of the logs when running the bring up: [realsense_bringup.txt](https://github.com/IntelRealSense/realsense-ros/files/8960570/realsense_bringup.txt)

Any help resolving this issue would be greatly appreciated!

Thanks,
Dylan

dverstandig, 2022-06-23:
Hi, 

We've had this happen on numerous cameras in several different environments. We are currently using the default USB cable provided by Realsense. 

We've had some environments that had access to sunlight, but the latest test had no access to sunlight, only light provided by indoor lighting.

MartyG-RealSense, 2022-06-23:
Are you using fluorescent lighting such as ceiling strip lights in your office?  These can cause interference with RealSense images because they contain a heated gas that flickers at frequencies that are difficult to see with the human eye.

If you do have such lights then you can test whether they are a disruptive influence by changing an RGB setting called **power line frequency** to match the operating frequency of the lights in your particular location.  In North America regions this frequency is typically **60** Hz, whilst in European regions it is **50** Hz.  

In ROS the controlling parameter is **/camera/rgb_camera/power_line_frequency** with values '0' for Disabled, '1' for 50Hz and '2' for 60Hz.

dverstandig, 2022-06-23:
Thanks for the reply, seems like this is exactly what is happening. 

I am trying to find the rosparam /camera/rgb_camera/power_line_frequency, but I dont see it anywhere. rosparam list does not return a 'power_line_frequency' and dynamic recofnigre does not show this as an option:
[image]

Could you help me locate this param? I also dont see it referenced anywhere in the code. 

Thanks,
Dylan

dverstandig, 2022-06-23:
Thanks!

I was able to set the value to 2 (60Hz), but after a few launches I had the same issue of the image turning white. I doubled checked and the value is set to 60Hz in dynamic reconfigure:

[image]

MartyG-RealSense, 2022-06-24:
What is the temperature of the camera when the issue occurs?  In the RealSense ROS wrapper you can monitor temperature in the **/diagnostics** topic.

If the ASIC temperature is exceeding 42 degrees C when the problem occurs then this could suggest that image problems are being caused by an excessive internal temperature event inside the camera.  If the camera is disconnected then temperature may fall back to a lower level and then the excessive temperature of > 42 degrees C may return at some point after the camera is reconnected.

dverstandig, 2022-06-24:
Hi,

It seems like the temperature is ok, looking at **/diagnostics** when the image turns white, the value looks to be 35 degrees C

```
---
header:
  seq: 1136
  stamp:
    secs: 1656095516
    nsecs: 972740476
  frame_id: ''
status:
  -
    level: 0
    name: "camera/front/realsense2_camera_manager: Asic Temperature"
    message: "OK"
    hardware_id: "105322250472"
    values:
      -
        key: "Index"
        value: "35"
```

MartyG-RealSense, 2022-06-25:
I note that you have auto-exposure enabled in the rgb_camera settings in the above rqt_reconfigure diagram.  If the indoor location that the camera is used in has a constant lighting level, would it be possible to try disabling auto-exposure to see whether it prevents the image from turning white?

MartyG-RealSense, 2022-07-01:
Hi @dverstandig  Do you require further assistance with this case, please?  Thanks!

dverstandig, 2022-07-05:
Hi,

Thanks for getting back to me. Disabling the auto exposure mitigates the issue for now, but we have some environments where the lighting conditions are not constant. 

For now, I think we can close this issue, and when the time comes, I can open a new issue if required. 

Thank you for the help troubleshooting!

closed 2022-07-05 by dverstandig