realsenseai/realsense-ros #2394: I got double depth hz?

byhang, 2022-06-30:
Hi @MartyG-RealSense,
I set depth_fps 1x, but I got 2x depth fps.
example: 
depth_fps=6 in rs_camera.launch, got 12 hz  of /camera/depth/iamge_rect_raw.
15 got 30,
30 got 60.
Is it normal?

MartyG-RealSense, 2022-06-30:
Hi @byhang   This effect would not be normal.  Are you able to achieve 6 FPS if you use the roslaunch instruction below that defines a custom stream configuration, please?

**roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=6**

byhang, 2022-06-30:
@MartyG-RealSense ,
roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=6
Got 6 hz.

MartyG-RealSense, 2022-06-30:
Thank you very much for testing.  Do you also correctly get 15 and 30 FPS if you change the depth_fps value in the launch instruction to 15 or 30?

byhang, 2022-06-30:
15 and 30 are normal.

My launch file below in another project, is any err?

### realsense.launch file

<?xml version="1.0"?>
<launch>
    <arg name="cam"                         default="cam"/>
    <arg name="camera"                      default="$(arg cam)_cam"/>
    <arg name="usb_port_id"                 default=""/>
    <arg name="device_type"                 default=""/>
    <arg name="json_file_path"              default="/opt/raccoon/share/realsense_node/param/HighAccuracyPreset.json"/>
    <arg name="tf_prefix"                   default="$(arg camera)"/>
    <arg name="external_manager"            default="false"/>
    <arg name="manager"                     default="$(arg camera)_manager"/>
    <arg name="output"                      default="screen"/>  
    <arg name="respawn"                     default="true"/>
    <arg name="initial_reset"               default="false"/>

    <arg name="enable_color"                default="false"/>

    <arg name="depth_width"                 default="480"/>
    <arg name="depth_height"                default="270"/>
    <arg name="enable_depth"                default="true"/>
    <arg name="depth_fps"                   default="6"/>

    <arg name="confidence_width"            default="$(arg depth_width)"/>
    <arg name="confidence_height"           default="$(arg depth_height)"/>
    <arg name="enable_confidence"           default="true"/>
    <arg name="confidence_fps"              default="$(arg depth_fps)"/>
    <arg name="infra_fps"                   default="$(arg depth_fps)"/>

    <arg name="align_depth"                 default="true"/>
    <arg name="enable_pointcloud"           default="true"/>
    <arg name="pointcloud_texture_stream"   default="RS2_STREAM_ANY"/>  
    <arg name="publish_tf"                  default="true"/>

    <env name="ROSCONSOLE_CONFIG_FILE" value="$(env HOME)/.raccoon_config/system/log/rgbd_$(arg cam)_log.conf"/>
    <include file="$(find realsense2_camera)/launch/rs_camera.launch">
        <arg name="camera"                      value="$(arg camera)"/>
        <arg name="usb_port_id"                 value="$(arg usb_port_id)"/>
        <arg name="device_type"                 value="$(arg device_type)"/>
        <arg name="json_file_path"              value="$(arg json_file_path)"/>
        <arg name="tf_prefix"                   value="$(arg tf_prefix)"/>
        <arg name="external_manager"            value="$(arg external_manager)"/>
        <arg name="manager"                     value="$(arg manager)"/>
        <arg name="output"                      value="$(arg output)"/>  
        <arg name="respawn"                     value="$(arg respawn)"/>
        <arg name="initial_reset"               value="$(arg initial_reset)"/>

        <arg name="enable_color"                value="$(arg enable_color)"/>

        <arg name="depth_width"                 value="$(arg depth_width)"/>
        <arg name="depth_height"                value="$(arg depth_height)"/>
        <arg name="enable_depth"                value="$(arg enable_depth)"/>
        <arg name="depth_fps"                   value="$(arg depth_fps)"/>

        <arg name="confidence_width"            value="$(arg confidence_width)"/>
        <arg name="confidence_height"           value="$(arg confidence_height)"/>
        <arg name="enable_confidence"           value="$(arg enable_confidence)"/>
        <arg name="confidence_fps"              value="$(arg confidence_fps)"/>
        <arg name="infra_fps"                   value="$(arg infra_fps)"/>

        <arg name="align_depth"                 value="$(arg align_depth)"/>
        <arg name="enable_pointcloud"           value="$(arg enable_pointcloud)"/>
        <arg name="pointcloud_texture_stream"   value="$(arg pointcloud_texture_stream)"/>  
        <arg name="publish_tf"                  value="$(arg publish_tf)"/>
    </include>
    <group ns="$(arg camera)">
        <node pkg="nodelet" type="nodelet" name="depth_points"
            args="load depth_image_proc/point_cloud_xyz $(arg manager) --no-bond"
            respawn="$(arg respawn)" output="$(arg output)">
        <remap from="image_rect" to="/$(arg camera)/aligned_depth_to_color/image_raw"/> 
        <remap from="points"     to="/$(arg camera)/aligned_depth_to_color/points" />
        </node>
    </group>
</launch>

### my launch file:

    <include file="$(find realsense_node)/launch/realsense.launch">
        <arg name="cam" value="a"/>
        <arg name="usb_port_id" value="2-2"/>
    </include>
    <include file="$(find realsense_node)/launch/realsense.launch">
        <arg name="cam" value="b"/>
        <arg name="usb_port_id" value="2-3"/>
    </include>

byhang, 2022-06-30:
roslaunch /.../realsense.launch depth_width:=640 depth_height:=480 depth_fps:=30 usb_port_id:=2-2

MartyG-RealSense, 2022-06-30:
Looking at the ROS wrapper's **opensource_tracking.launch** file, which also imports rs_camera.launch settings like your launch file does, it looks as though it only lists the parameters that it wants to change the settings of - for example, setting align_depth to True instead of the default False in rs_camera.  

https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/opensource_tracking.launch#L4-L7

You do not seem to be changing any of the rs_camera default settings in your arg list, so I wonder whether all that you actually need in that section is this:

 <include file="$(find realsense2_camera)/launch/rs_camera.launch">
</include>

byhang, 2022-06-30:
@MartyG-RealSense , By the way, where can i get old official version deb of librealsense2(-2.45.0) and realsense-ros-2.3.0 in Ubuntu-16.04, is there have a url?

The ddynamic-reconfigure-0.3.2 and librealsense2-2.50.0 and realsense-ros-2.3.2 build by myself, it work well on my pc, but does not take effect in my robot which have the same version kernel.
### dmesg in my robot
[   14.812304] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[   14.813926] uvcvideo: Unable to create debugfs 2-3 directory.
[   14.813951] uvcvideo 2-6:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[   14.813952] uvcvideo 2-6:1.0: Entity type for entity Processing 2 was not initialized!
[   14.813953] uvcvideo 2-6:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[   14.813954] uvcvideo 2-6:1.0: Entity type for entity Camera 1 was not initialized!
[   14.814004] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/input/input9
[   14.814118] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[   14.814931] uvcvideo: Unable to create debugfs 2-3 directory.
[   14.814950] uvcvideo 2-6:1.3: Entity type for entity Processing 7 was not initialized!
[   14.814952] uvcvideo 2-6:1.3: Entity type for entity Extension 8 was not initialized!
[   14.814952] uvcvideo 2-6:1.3: Entity type for entity Camera 6 was not initialized!
[   15.410353] uvcvideo: Found UVC 1.00 device USB 2.0 Camera (1871:0130)

[   30.237623] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[   30.273043] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[   30.288881] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[   30.323388] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[   30.374009] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[   30.402506] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[   30.424984] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[   30.475542] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[   30.635510] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[   30.640485] uvcvideo: Failed to query (GET_CUR) UVC control 2 on unit 3: -32 (exp. 1).
[   30.690614] uvcvideo: Failed to query (GET_CUR) UVC control 2 on unit 3: -32 (exp. 1).
[   30.740735] uvcvideo: Failed to query (GET_CUR) UVC control 2 on unit 3: -32 (exp. 1).
[   30.747044] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[   30.790904] uvcvideo: Failed to query (GET_CUR) UVC control 2 on unit 3: -32 (exp. 1).

MartyG-RealSense, 2022-06-30:
Advice is provided at https://github.com/IntelRealSense/librealsense/issues/5773#issuecomment-581477785 and https://github.com/IntelRealSense/realsense-ros/issues/1925#issuecomment-860157761 about installing a specific package version of librealsense instead of the latest version.

You would have to then install the matching ROS wrapper version for that particular librealsense version from source code instead of packages though.

byhang, 2022-06-30:
@MartyG-RealSense I am glad to tell you, comment this line, fixed the depth hz.

But i don not know why?
color=true   align_depth=true    dps=15  got 15hz
color=false  align_depth=true   dps=15  got 30hz
color=false  align_depth=false  dps=15  got 15hz

MartyG-RealSense, 2022-06-30:
I have not seen this particular effect before.  Usually if there is a problem with align_fps then the FPS reduces, not increases.

Could you try disabling an RGB option called **auto-exposure_priority** to see if the color FPS maintains its correct speed please, as described at https://github.com/IntelRealSense/realsense-ros/issues/1349#issuecomment-681695259

You can test quickly whether disabling auto_exposure_priority solves the problem by opening the **rqt_reconfigure** interface after launch has completed with the command **rosrun rqt_reconfigure rqt_reconfigure**, going to the RGB section of the options and unticking auto_exposure_priority

[image]

byhang, 2022-06-30:
[image]
my auto_exposure_priority is unticking default.

byhang, 2022-06-30:
> You would have to then build the matching ROS wrapper version for that particular librealsense version from source code instead of packages though.
Sorry, I can't understand that, The "source code" is github repo? I cloned and checkout to v2.50.0, it's same as the librealsense2-2.50.0.tar.gz. Did you mean that use the newest github repo?

byhang, 2022-06-30:
I built ros-kinetic-librealsense2_2.50.0-0xenial_amd64.deb from librealsense2-2.50.0.tar.gz use [this method](https://gist.github.com/awesomebytes/196eab972a94dd8fcdd69adfe3bd1152)
```bash
bloom-generate rosdebian --os-name ubuntu --os-version xenial --ros-distro kinetic
fakeroot debian/rules binary
```

MartyG-RealSense, 2022-06-30:
When speaking of source code, I was referring to downloading the source for a particular version from a file on the wrapper's **Releases** page.

https://github.com/IntelRealSense/realsense-ros/releases

I am not familiar with the **bloom** installation method.  If a basic rs_camera launch works correctly and publishes topics successfully then it is likely that the installation is okay.

byhang, 2022-07-01:
@MartyG-RealSense , There is must a official realsense-ros-2.3.2 deb compile command method, but i not found in official site, can you give me a url?

closed 2022-07-12 by MartyG-RealSense