realsenseai/realsense-ros #2396: D455 on Jetson Xavier very slow

GEngels, 2022-07-01:
I am trying to use the Intel Realsense camera on the Jetson Xavier AGX. When I only run the RGB camera & depth image it seems to run quite fast (30fps). However, when I enable the pointcloud it becomes very slow. 

I tried to run it at 15 fps for RGB & Depth at various resolutions but the result is the same. The RGB camera freezes and no point cloud is displayed in RVIZ. When I block part of the depth camera (so that it doesn't have to create as many points in the point cloud) it seems to work again. The creation of the pointcloud in the realsense node seems to be a bottleneck, I am running the Jetson on max performance.

Some of the errors and warnings that appear are the following:

"incomplete frame received: Incomplete video frame detected! Size 99080 out of 2048255 bytes (4%):"
"Hardware Notification:Depth stream start failure,1.65668e+12,Error,Hardware Error"
"Out of frame resources"

I wonder if it is expected behaviour that you can only run it with maybe 5 FPS on the Jetson on max settings.

MartyG-RealSense, 2022-07-01:
Hi @GEngels  Problems with pointcloud generation, and also enabling align_depth, on Jetson boards specifically are a known issue in the RealSense ROS wrapper.  Symptoms can iclude very low FPS or missing color.  The issue does not occur on non-Jetson computers such as laptop and desktop PCs.  Detailed information about it can be found at https://github.com/IntelRealSense/realsense-ros/issues/1967

For ROS1 (Kinetic, Melodic, Noetic) the current best solution available is to use librealsense version **2.43.0** and ROS1 wrapper version **2.2.23**.

Doch88, 2022-07-04:
Hi @MartyG-RealSense, I am a collegue of @GEngels.
Thank you for your quick answer!
We are currently using ROS2, is there also a version of the ROS2 wrapper that works flawlessly with the Jetson? And I guess the librealsense version will still be 2.43.0 in that case.

MartyG-RealSense, 2022-07-04:
Hi @Doch88  The older RealSense **ros2** wrapper branch would have to be used with older SDK versions, as the current **ros2_beta** wrapper branch is for SDK 2.50.0.

https://github.com/IntelRealSense/realsense-ros/tree/ros2

If SDK 2.43.0 is being used with ROS2 then it would have to be matched with ROS2 wrapper version **3.1.5**

https://github.com/IntelRealSense/realsense-ros/releases/tag/3.1.5

This wrapper version only supports Foxy, Eloquent and Dashing though.  So it would not be suitable for newer ROS2 versions such as Galactic and Rolling as support for those two ROS2 versions was introduced in more recent wrapper versions.

As far as I am aware, Jetson pointcloud generation has not been tested with the combination of SDK 2.43.0 and ROS2 wrapper 3.1.5 though, so I cannot offer a guarantee about pointcloud performance with that configuration.

Doch88, 2022-07-05:
Hi @MartyG-RealSense, I built the SDK 2.43.0 version from source (using RSUSB backend), with CUDA support, and I also used the ROS2 wrapper version 3.1.5 (with Foxy). 
The problem seems to be still there. However, it seems to change depending on which Jetson power option is enabled.

 - With MAXN the FPS seems very low, and the frames are received with a delay, at least as it is shown on RViz and Rqt (this does not happen with the point cloud generation disabled).
 - With MODE 15W Desktop, same as with MAXN.
 - With MODE 30W ALL the color frame is stuck, and the point cloud is not published. The depth frame is shown correctly. If we lower the resolution of the Depth frame to 848x400 the problem seems to not be there anymore.

What looks strange to me is that the 15W is better than the 30W one, I would guess that it depends on some specific power limitation set with these modes on the USB port or on the CPU cores.

Doch88, 2022-07-05:
Sorry, I mean covering the lens on the outside of the camera.

MartyG-RealSense, 2022-07-05:
How about using 1280x720 and applying a **Decimation** filter to reduce the complexity of the depth scene?

Doch88, 2022-07-05:
Yes, we are using alignment, and also the Decimation filter will reduce the resolution of the depth image, which is not something that we want.

MartyG-RealSense, 2022-07-05:
In your project, is 1280x720 a requirement for both depth and color or only for depth?

MartyG-RealSense, 2022-07-06:
Can you confirm please that when you built librealsense from source code with RSUSB, you included the build flag **-DBUILD_WITH_CUDA=TRUE** in the CMake build instruction?  An RSUSB build of librealsense will automatically support **metadata** but require the DBUILD_WITH_CUDA flag to be used in order to add CUDA support to the build too.

Doch88, 2022-07-06:
I used this command to build the library:

`cmake -DBUILD_WITH_CUDA=True -DCMAKE_BUILD_TYPE=Release -DFORCE_LIBUVC=true ..`

Doch88, 2022-07-06:
I checked the differences between MODE 30W ALL and MODE 15W DESKTOP.
The first one uses 8 core with a capped frequency for all of them of 1.2 Ghz. The latter, instead, uses 4 cores, and two of them have a capped frequency of 2.18 Ghz (The other twos still 1.2 Ghz).
As the 15W DESKTOP one works better than the 30W one I guess that the problem is some CPU-intensive operation. 
As I know the spatial filtering and the temporal filtering are executed on the CPU, while I know nothing about the point cloud generation (I see that there is a function called rs2::gl::pointcloud for pointcloud generation using GPU, which is not used directly in the ROS2 node, but maybe it is used indirectly when building the library with CUDA).

Also I looked a bit into the code of the realsense-viewer and at a first sight I noticed that there a thread is used for the post-processing part of the code, while in the node (as I understood) everything is done in the same thread (except for some part not related to filtering and point cloud generation). 
Maybe that could also help a bit to understand what is the actual problem.

MartyG-RealSense, 2022-07-06:
You mentioed earlier at https://github.com/IntelRealSense/realsense-ros/issues/2396#issuecomment-1175108099 that you could run well with 1280x720 depth and 1280x800 color (both at 15 FPS) but with spatial and temporal filters disabled.  Are these filters vital to your ROS project, please?

Normally, Infra and Infra2 are disabled by default when using the **rs_launch.py launch** file.  Are you using that launch file or a custom launch file, and if you are using a custom launch then are Infra and Infra2 enabled in it?

Doch88, 2022-07-06:
> Normally, Infra and Infra2 are disabled by default when using the rs_launch.py launch file. Are you using that launch file or a custom launch file, and if you are using a custom launch then are Infra and Infra2 enabled in it?

We are using a custom launch file that uses rs_launch.py inside, we are not touching the settings regarding infra and infra2.

These are the launch arguments, many of them are useless for the current camera but they are still provided for internal purposes:
```
            launch_arguments={
                'device_type': 'd455',
                'enable_gyro': 'false',
                'enable_accel': 'false',
                'enable_sync': 'true',
                'initial_reset': 'true',
                'align_depth': 'true',
                'enable_pointcloud': 'true',
                'ordered_pc': 'true',
                'enable_fisheye1': 'false',
                'enable_fisheye2': 'false',
                'enable_confidence': 'false',
                'depth_fps': '15.0',
                'depth_width': '1280',
                'depth_height': '720',
                'color_fps': '15.0',
                'color_width': '1280',
                'color_height': '800',
                'filters': '"decimation"',
                'clip_distance': '10.0',
            }
```

MartyG-RealSense, 2022-07-06:
It's no trouble at all  :)

If librealsense is built with the flag **-DBUILD_WITH_OPENMP=TRUE** then YUY to RGB color conversion and depth-color alignment can take advantage of multiple CPU cores, at the expense of slightly higher CPU percentage utilization.

rs2::gl, also known as GLSL, is an alternative acceleration system to CUDA that is built into librealsense and works with any graphics GPU brand instead of just Nvidia.  It is not involved in CUDA processing.  Tthere would not be a significant performance advantage to using GLSL over CUDA and since GLSL only works with rs2:: instructions in librealsense scripts, CUDA will be the best choice for ROS.

Unless you need an ordered pointcloud instead of the default unordered one then you could try removing the **ordered_pc** instruction to see if it makes a positive difference.

As you are using both depth and color, if auto-exposure is enabled then you could also try disabling an RGB setting called **auto_exposure_priority** to force the FPS of the two streams to remain at the defined FPS (such as '30') instead of being permitted to drop to a lower FPS value.  Information about defining this setting in a launch file as a rosparam is at https://github.com/IntelRealSense/realsense-ros/issues/2308#issuecomment-1086628767

Doch88, 2022-07-08:
Thank you for your clarifications!
I will try to build librealsense today with OpenMP enabled and see if it improves the results.

Yes, we need ordered_pc and I have also tried to disable it and the performances did not improve visibly.

I cannot find how to set that parameter with ROS2, that setting in the launch file seems to be a ROS1 one. Also, I cannot find the definition of that parameter inside of the Realsense node, can you provide me a way to do that with ROS2?

MartyG-RealSense, 2022-07-08:
Are you able to test disabling auto_exposure_priority by opening the **rqt_reconfigure** interface in the 3.x ROS2 wrapper with the command **ros2 run rqt_reconfigure rqt_reconfigure**

If you can open the rqt_reconfigure interface and the side-panel has an **rgb_camera** category in its side-panel then the auto_exposure_priority option should be under that section's options.

[image]

Doch88, 2022-07-12:
For some reason rqt_reconfigure was not working, but I managed to change the setting with 
`ros2 param set /camera/camera rgb_camera.auto_exposure_priority false`
Although nothing really changed that much.
Also, I just saw that disabling everything (including pointcloud generation and depth alignment) on Rviz appears to go quite ok, but instead, doing `ros2 topic hz /camera/color/image_raw`, I realize that the publishing rate is always lower than expected. 

If I set the camera to run at 15 fps, the publishing rate is around 10 Hz; if I set the camera to run at 30 fps the publishing rate is around 20 Hz (1280x720 for both cases).
That happens also if I disable auto_exposure_priority.

MartyG-RealSense, 2022-07-12:
Thanks very much for your tests.

When auto_exposure_priority is set to False, is the RGB **auto_exposure** also set to True?  If auto_exposure is true and auto_exposure_priority is false then the FPS rate should be forced to remain constant.

MartyG-RealSense, 2022-07-13:
GLSL differs from CUDA in that programs will not make use of GLSL acceleration unless support for doing so has been deliberately coded into that program.  So the ROS wrapper would not be benefiting from GLSL.

CUDA acceleration of pointclouds, alignment and color conversion is provided automatically though if the librealsense SDK has CUDA support enabled (by building the SDK from packages, or from source code with the **DBUILD_WITH_CUDA=true** flag included).

CUDA support is not included though if the SDK and the ROS wrapper were installed together from packages at the same time using the wrapper's **Method 1** instructions.

Doch88, 2022-07-18:
Then I do not really understand this difference of performance between the node and the realsense-viewer.
Another thing that I saw is that the USB consumption seems to be quite low, even when used:
```
jetsonxavier@ubuntu:~$ lsusb -v | egrep "^Bus|MaxPower"
Bus 002 Device 004: ID 8086:0b5c Intel Corp. 
    MaxPower              180mA
```
I would expect it to be around 700 mA, as it is when I run it in a laptop.
Apart from that, I do not really know what else to look, I feel like I checked everything.

MartyG-RealSense, 2022-07-18:
The RealSense Viewer tool runs directly in the RealSense SDK, whilst the ROS wrapper runs as a layer atop the SDK in the background.  The wrapper also handles some functions differently from librealsense in order to maintain compatibility with ROS standards.  So the camera may work correctly in the Viewer but have some performance issues in the ROS wrapper for some RealSense users.

A Jetson user at https://github.com/IntelRealSense/realsense-ros/issues/1964 who was also using MAXN mode had performance issues such as very high CPU % utilization.  They provided extensive testing logs in that case.  In the end, they found that performance improved if they removed the ROS wrapper with the command **sudo apt remove ros-$ROS_DISTRO-librealsense2** and rebuilt it from source code.

MartyG-RealSense, 2022-07-24:
Hi @GEngels  and @Doch88  Do you require further assistance with this case, please?  Thanks!

iraadit, 2022-07-25:
Hi @MartyG-RealSense,

I'm a colleague of @GEngels and @Doch88, and I'm taking over this issue. I'm currently writing a first long message to try to be the most complete possible.

Thank you.

MartyG-RealSense, 2022-07-25:
Thanks very much @iraadit for the introduction.  I'm pleased to meet you and look forward to your forthcoming message!

iraadit, 2022-07-25:
Hi @MartyG-RealSense,

Here is a recap of my setup:

- Nvidia Jetson AGX Xavier 16 GB (Dev Kit),, in MAXN power mode
- Jetpack 4.2.3 (L4T 32.2.3) (we also used a more recent Jetpack on another Jetson, with the same results) (Ubuntu 18.04)
- ROS2 Dashing
- Powered USB hub
- Intel RealSense D455, firmware version: 05.13.00.50
- Librealsense v2.50.0 (built from source with CUDA and OpenMP support, as well as RSUSB backend)
- RealSense ROS, branch [ros2-beta](https://github.com/IntelRealSense/realsense-ros/tree/ros2-beta) (commit: 725168df2b65935d3a6f7051730e18f2fa2bdcbb, Date:   Tue Jul 19 19:06:55 2022 +0300) (tag 4.0.4) (built from source)

First, **should ros2_beta be considered production ready?** Is ros2-beta branch  having the last development / being compatible with the last librealsense version, while ros2 branch is stale?

We aim at getting the maximum output quality. Our application should target precision and reactivity. Therefore, we want to be able to get the point cloud at max resolution (1280x720) and 30FPS. It would be preferable to get a colored texture stream (_RS2_STREAM_COLOR_), but no texture stream (_RS2_STREAM_ANY_) would be acceptable for now (with *align_depth:=true* though). Soon, we will want to be able to use 2 realsense cameras for one Jetson AGX Xavier. And later on, possibly 4 cameras. It could later be considered to get higher FPS at the expense of the resolution (up to 90FPS at 848x480).

I checked several issues on this Github repository as well as on the Nvidia forum. I didn't find any definitive answer on how to get good point cloud performances on Jetson, without sacrificing output quality or downgrading librealsense and realsense-ros (which seemed to be a solution for ROS1). With your help, I'd like to go to the end of this issue and to find a working configuration. I'd be ready to delve into code if necessary and to publish a pull request if we find a solution.

Here is a collection of the observations I made until now:
- When using **realsense-viewer**, CPU load is shared between different threads. One thread is around 80%, another one around 10%, and then several other ones between 0 and 5%
- When using **realsense-ros ros2 branch**, with _RS2_STREAM_COLOR_, FPS will be really slow or no point cloud will be published. I can observe that a thread is saturating (CPU usage around 90-100%), and that publishing of the point cloud is stopping. If covering one of the IR cameras (hand in front of the lens), CPU usage is going down and publication of the point cloud is starting again at "fast" speed. Taking out my hand, it would publish again a correct point cloud, before slowing down and finally stopping. Multiple *control_transfer returned error*.
  ```
  $ launch realsense2_camera rs_launch.py enable_pointcloud:=true initial_reset:=true pointcloud_texture_stream:=RS2_STREAM_COLOR depth_width:=1280 depth_height:=720 depth_fps:=30.0
  ```
- In the same config as previous point, but with _RS2_STREAM_ANY_, point cloud will be published, but with slow FPS and discernible delay. Total cpu usage around 65-70%, mainly shared between 2 threads at 40 and 15%. Multiple *control_transfer returned error*.
  ```
  $ ros2 launch realsense2_camera rs_launch.py enable_pointcloud:=true initial_reset:=true pointcloud_texture_stream:=RS2_STREAM_ANY depth_width:=1280 depth_height:=720 depth_fps:=30.0
  ```
- In the same config as previous point, but with *align_depth:=true*, same observations. Total cpu usage around 75%. Multiple *control_transfer returned error*. 
  ```
  $ ros2 launch realsense2_camera rs_launch.py enable_pointcloud:=true initial_reset:=true pointcloud_texture_stream:=RS2_STREAM_ANY depth_width:=1280 depth_height:=720 depth_fps:=30.0 align_depth:=true
  ```
- In the previous tests, were using the ros2 branch, which was giving poor results as well as errors (such as _WARNING [547297925568] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11_). I then switched to the **ros2_beta branch**, and this error seems to have disappeared. Publishing of the point cloud is more stable, even with *RS2_STREAM_COLOR*. Delay is reduced too! Total CPU usage is around 40%, shared between a thread at approx. 25%, one at approx. 8% and same smaller ones. Note: *pointcloud.stream_filter:=2* with ros2_beta branch is equivalent to *RS2_STREAM_COLOR* with ros2 branch (respectively, *:=0* and *RS2_STREAM_ANY*)
  ```
  $ ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true depth_module.profile:=1280x720x30 decimation_filter.enable:=false pointcloud.stream_filter:=2
  ```
  The publication rates are still slow and variable though, as showed by the next command:
  ```
  $ ros2 topic hz /camera/depth/color/points -w 10
  average rate: 9.935
  	min: 0.055s max: 0.141s std dev: 0.03570s window: 10
  average rate: 7.864
  	min: 0.050s max: 0.265s std dev: 0.07520s window: 10
  average rate: 6.858
  	min: 0.064s max: 0.334s std dev: 0.08832s window: 10
  average rate: 11.541
  	min: 0.063s max: 0.209s std dev: 0.04425s window: 10
  average rate: 6.488
  	min: 0.064s max: 0.397s std dev: 0.10397s window: 10
  average rate: 7.923
  	min: 0.057s max: 0.271s std dev: 0.08429s window: 10
  average rate: 7.141
  	min: 0.061s max: 0.472s std dev: 0.12704s window: 10
  average rate: 5.008
  	min: 0.062s max: 0.530s std dev: 0.18022s window: 10
  ```
  ```
  $ ros2 topic hz /camera/color/image_raw -w 10
  average rate: 18.758
[... cut ...]
- I don't see any delay for /camera/color/image_raw nor  /camera/depth/image_rect_raw using
  ```
  $ ros2 run rqt_image_view rqt_image_view
  ```
  There is a more visible delay in Rviz with the pointcloud
- Another user had problems with ROS2 and pointcloud, he finally used ROS1 and bridged the topics to ROS2. I still didn't try his solution, I'll do that in the coming days if no native solution with ROS2 is found.
  See https://github.com/IntelRealSense/realsense-ros/issues/2388#issuecomment-1190365341

I'll continue to look into it in the coming days.

Thank you for your help.

MartyG-RealSense, 2022-07-26:
Thanks so much @iraadit for your very detailed report.

*  Whilst the ros2_beta wrapper is a beta, it is stable and so may be the best choice for your production environment if the previous generation ros2 wrapper was providing degraded performance and more errors compared to ros2_beta.

A good rule to apply in a production situation is "one version back" - using one version older than the current version in order to give time for reports of any problems to emerge with a new version before committing to using it in a production or critical reliability environment.  So it may be worth sticking with 4.0.4 for a while even if a newer beta wrapper is released.

The current known issues with using pointcloud and align with Jetson boards specifically (such as FPS drop or missing color) should apply to both the ros2 and ros2_beta wrappers though.  Also, it is likely that there will be a performance drop with using 1280x720 compared to 848x480, especially if using 90 FPS.

*  Typically, the maximum number of RealSense 400 Series cameras that have been attached to the same Jetson boad via a hub has been six.  So four cameras would certainly be an achievable target.

*  If performance degrades and then is corrected by the auto-exposure after placing a hand over the lenses, this can indicate that the infrared sensor may have become saturated with light by a cause such as a strong light source (such as sunlight) flashing directly into the camera's view.

*  As mentioned at https://github.com/IntelRealSense/realsense-ros/issues/2396#issuecomment-1176356778 earlier in this discussion, a common method used to deal with differences between depth and RGB streams, such as FPS lag or one stream being 30 FPS and the other 15, is to have auto-exposure enabled and the RGB option **auto_exposure_priority** disabled.

 *  The developer of the ROS1 wrapper stated at https://github.com/IntelRealSense/realsense-ros/issues/588#issuecomment-457890773 that the **No stream match for pointcloud chosen texture Process - Color** warning can occur if there is a frame drop.

MartyG-RealSense, 2022-08-01:
Hi @iraadit  Do you require further assistance with this case, please?  Thanks!

iraadit, 2022-08-02:
Hi @MartyG-RealSense,

Yes, I'm still working on it / trying to find a solution.

First, answering your last comments:

- It is good to know that 6 cameras have already been achieved on Jetson. Was it while using poincloud though?

- The problem doesn't seem to come from a saturated infrared sensor, the luminosity of the room isn't changing the results. it works with smaller resolutions though.

- **rgb_camera.enable_auto_exposure** is enabled and **rgb_camera.auto_exposure_priority** disabled; these are their default values. I tried playing with those values (as well as with rgb_camera.exposure) with rqt_reconfigure and didn't see real changes; **BUT** I saw changes when changing depth_module.exposure. By default, **depth_module.enable_auto_exposure** is disabled and the value of **depth_module.exposure** is 33000 (33ms). Reducing this value to 32000 will increase the FPS of depth at the expense at the FPS of color, and more instability (pointcloud freezing sometime). Do you have an idea of why chaning this parameter would change the FPS of the color camera?

- It seems indeed that the color and depth frames are going out of sync, leading to the pointcloud problem.

I reinstalled my Jetson with Jetpack 4.6.2 (the last version 4)  to gain access to the latest Nvidia Nsight profiling tools.

I'm now able to debug and profile realsense-ros as well as librealsense.

I can also share that the problem is present with:
* Jetpack 4.2 and 4.6
* ROS2 Dashing, Eloquent and Foxy

From here, I'll dive into code to try to find what is going wrong.

By the way, as it seems it is a problem that occurred often for other users too, and that it is a showstopper for the use of pointclouds with Realsense and Jetson, do you know if the internal dev time of Intel Realsense are working on it?

Thank you

MartyG-RealSense, 2022-08-03:
Typically the 6 cameras have been used with basic depth / color streams rather than pointclouds.  If CUDA support is enabled in the librealsense SDK though then the processing burden of generating a pointcloud should be low, as the Nvidia graphics GPU on the Jetson can process the pointcloud instead of the CPU.

If auto-exposure is enabled then setting a manual exposure value will automatically disable auto- exposure.  When using manual exposure, setting values in a certain range can increase or decrease FPS.  More information about this is at https://github.com/IntelRealSense/librealsense/issues/1957#issuecomment-400715579

The Jetson-specific issues with pointclouds and alignment are being worked on, though there is not a date for implementation at the time of writing this.

MartyG-RealSense, 2022-08-09:
Hi @iraadit   Was the information in the comment above helpful to you, please? Thanks!

iraadit, 2022-08-10:
Hi @MartyG-RealSense 
I'll answer tomorrow
Still not there, but slowly going in the right direction :)

MartyG-RealSense, 2022-08-16:
Hi @iraadit  Do you have an update about this case that you can provide, please?  Thanks!

iraadit, 2022-08-18:
Hi @MartyG-RealSense,
Again sorry for the delayed answer.

I passed the last days installing Jetpack 5 and Isaac ROS (Humble with Docker)
That took me some time but I now have librealsense and realsense-ros working in the Docker.
I hoped new versions of ROS packages would maybe help, it seems at first it isn't the case.
I'll continue to explore with this version though.

In the discoveries of the last weeks:

1. I can get 30 FPS in color and depth (without pointcloud) by setting _depth_module.profile:=1280x720x30 depth_module.exposure:=32000 depth_module.enable_auto_exposure:=false_ in the _ros2 launch realsense2_camera rs_launch.py_ command (confirmed using the topic _/diagnostics_). I also had to modifiy _rs_launch.py_ to accept those new parameters.

2. I profiled realsense-viewer and realsense-ros with Nsight, running on the Jetson. realsense-viewer is using the OpenGL version of the filters, while realsense-ros is using a "basic CPU" version. For poincloud, on a x86_64 host computer, the executed code is in SSE3 (what is not possible on aarch64/Jetson), already better than the "basic CPU" version. On Jetson with realsense-ros, poincloud operations is what is using must of the CPU time and the problematic function is _get_texture_map_.
The CUDA flag is only changing the function [_depth_to_points_](https://github.com/IntelRealSense/librealsense/blob/master/src/proc/cuda/cuda-pointcloud.cpp).
While the [OpenGL](https://github.com/IntelRealSense/librealsense/blob/c94410a420b74e5fb6a414bd12215c05ddd82b69/src/gl/pointcloud-gl.cpp#L424) and [SSE3](https://github.com/IntelRealSense/librealsense/blob/c94410a420b74e5fb6a414bd12215c05ddd82b69/src/proc/sse/sse-pointcloud.cpp#L269) versions are changing _get_texture_map_.
I modified rs-benchmark to take a 1280x720 depth image as input, here is the output:
```
$ rs-benchmark 

|            |     |
|------------|-----|
|**CPU** | ARMv8 Processor rev 0 (v8l) |
|**GPU** | NVIDIA Tegra Xavier (nvgpu)/integrated |
|**Graphics Driver** |4.6.0 NVIDIA 32.7.2 |
|**Device Name** |Intel RealSense D455 |

**Stream Type**: Depth, **Resolution**: 1280 x 720

|Filter Name |Step |Median(m)   |Mean(m)  |STD(m)  |Max(m)  | Max FPS |
|------------|-----|------------|---------|--------|--------|---------|
|colorizer |Calculate |19.138 |19.773 |4.766 |76.215 |30 ![30](https://placehold.it/15/82c13e/000000?text=+) |
|pointcloud |Calculate |4.561 |8.947 |49.191 |609.226 |6 ![6](https://placehold.it/15/d6a726/000000?text=+) |
|spatial_filter |Calculate |56.714 |56.782 |4.908 |103.292 |15 ![15](https://placehold.it/15/eff70c/000000?text=+) |
|temporal_filter |Calculate |7.653 |7.970 |0.753 |10.122 |90 ![90](https://placehold.it/15/35ff4d/000000?text=+) |
|disparity_transform |Calculate |4.532 |4.764 |0.716 |11.145 |90 ![90](https://placehold.it/15/35ff4d/000000?text=+) |
|threshold_filter |Calculate |3.982 |4.537 |0.831 |8.814 |90 ![90](https://placehold.it/15/35ff4d/000000?text=+) |
|decimation_filter |Calculate |3.402 |3.603 |0.538 |6.285 |90 ![90](https://placehold.it/15/35ff4d/000000?text=+) |
|gl::colorizer | Upload |3.602 |4.205 |1.912 |21.720 |90 ![90](https://placehold.it/15/35ff4d/000000?text=+) |
| |Calculate |1.357 |1.951 |1.291 |8.280 |90 ![90](https://placehold.it/15/35ff4d/000000?text=+) |
| |Download |1.685 |2.339 |1.325 |9.559 |90 ![90](https://placehold.it/15/35ff4d/000000?text=+) |
| |**Total** |8.373 |8.495 |3.116 |30.229 |60 ![60](https://placehold.it/15/6fe837/000000?text=+) |
|gl::pointcloud | Upload |3.556 |4.226 |1.446 |9.747 |90 ![90](https://placehold.it/15/35ff4d/000000?text=+) |
| |Calculate |1.223 |1.704 |2.445 |29.984 |90 ![90](https://placehold.it/15/35ff4d/000000?text=+) |
| |Download |0.002 |0.008 |0.073 |0.903 |90 ![90](https://placehold.it/15/35ff4d/000000?text=+) |
| |**Total** |5.043 |5.939 |3.059 |37.744 |90 ![90](https://placehold.it/15/35ff4d/000000?text=+) |

**Stream Type**: Color, **Resolution**: 1280 x 800

|Filter Name |Step |Median(m)   |Mean(m)  |STD(m)  |Max(m)  | Max FPS |
|------------|-----|------------|---------|--------|--------|---------|
|yuy_decoder |Calculate |2.108 |2.462 |1.046 |6.506 |90 ![90](https://placehold.it/15/35ff4d/000000?text=+) |
|gl::yuy_decoder | Upload |1.834 |2.377 |1.131 |6.112 |90 ![90](https://placehold.it/15/35ff4d/000000?text=+) |
| |Calculate |1.159 |1.583 |0.991 |5.516 |90 ![90](https://placehold.it/15/35ff4d/000000?text=+) |
| |Download |1.651 |2.032 |1.166 |10.268 |90 ![90](https://placehold.it/15/35ff4d/000000?text=+) |
| |**Total** |5.276 |5.993 |1.970 |17.424 |90 ![90](https://placehold.it/15/35ff4d/000000?text=+) |
```
It can be seen that _gl::pointcloud_ is faster than _pointcloud_ (6ms vs 9ms, maximum of 38 vs 609ms)
From there, I changed the code of realsense-ros to use OpenGL, getting some inspiration from https://github.com/IntelRealSense/librealsense/tree/master/examples/gl
I know have _get_texture_map_ going faster, less CPU usage and better streams. I'm still not satisfied with the result though.
I also tried to change _align_depth_ by its OpenGL version, but with no luck until now, because of an error about an OpenGL texture.
OpenGL upload should also be used to avoid sending the frames several times to the GPU.
I'll also maybe try to convert the SSE3 instructions to NEON instructions to make it compatible with aarch64.

4. Now, hz of poincloud is higher, but FastDDS seemed to badly cope with the transfer of the data, I switched to Cyclone DDS and got better results.

5. When running some example app, it is possible to load a config file to the camera. Behavior of the camera will be different after applying this config (and until a reset). Is is possible to get a faster pointcloud but losing a certain amount of points (higher confidence threshold). It could be explored but we would like to keep the highest quality possible.

I'll be on vacations until the start of September and will continue to work on this when I will be back. Please, do not close the ticket :)

Thank you

MartyG-RealSense, 2022-08-19:
Thanks so much @iraadit for the highly detailed feedback of your findings.  It is no problem at all to keep the ticket open whilst you are on vacation and resume when you return.

AndreV84, 2022-08-20:
@iraadit Thank you for sharing
@MartyG-RealSense what about Humble ROS support on Jetson Docker? Just about Humble aarch64 support? Could you add on it, please?

MartyG-RealSense, 2022-08-20:
Hi @AndreV84   Humble is now supported by the RealSense **ros2_beta** wrapper, though not by earlier wrapper branches (ROS1 and **ros2**).  At the time of writing this, only building from source code is supported for Humble.  Ubuntu 22.04 (Jammy) Debians are on the roadmap.  https://github.com/IntelRealSense/librealsense/issues/10439#issuecomment-1186588624 and the comments beneath it provide further information.

iraadit, 2022-08-29:
Gld it can help you @AndreV84 :)

Hi @MartyG-RealSense 

I've been able to install the last librealsense v2.51.1 as well as the branch ros2-beta of realsense-ros inside a docker container created with Nvidia Isaac ROS (with ROS 2 Humble). I could install following the first method of the Jetson install (with _FORCE_RSUSB_BACKEND=false_), without needing to apply the kernel patching. Running on a Nvidia Jetson AGX Xavier.

Sadly, the performances are staying the same.

ROS2 logs are now working correctly though and I observed something in the output of the command `ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true decimation_filter.enable:=false pointcloud.stream_filter:=2 diagnostics_period:=1.0 enable_sync:=true align_depth.enable:=false  ordered_pc:=true log_level:=debug output:=log camera_name:=realsense_cam depth_module.profile:=1280x720x30 depth_module.exposure:=32000 depth_module.enable_auto_exposure:=false`:

```
[realsense2_camera_node-1] [DEBUG] [1661793531.267126566] [realsense_cam.realsense_cam]: Frameset arrived.
[realsense2_camera_node-1] [DEBUG] [1661793531.267218956] [realsense_cam.realsense_cam]: List of frameset before applying filters: size: 1
[realsense2_camera_node-1] [DEBUG] [1661793531.267279728] [realsense_cam.realsense_cam]: Frameset contain (Depth, 0, Z16 0) frame. frame_number: 8748 ; frame_TS: 1661793531226.631592 ; ros_TS(NSec): 1661793531226631680
[realsense2_camera_node-1] [DEBUG] [1661793531.267353108] [realsense_cam.realsense_cam]: num_filters: 11
[realsense2_camera_node-1] [DEBUG] [1661793531.276654409] [FrequencyStatus_debug_logger]: TICK 5941
[realsense2_camera_node-1] [DEBUG] [1661793531.288720997] [realsense_cam.realsense_cam]: List of frameset after applying filters: size: 2
[realsense2_camera_node-1] [DEBUG] [1661793531.288904336] [realsense_cam.realsense_cam]: Frameset contain (Depth, 0, XYZ32F) frame. frame_number: 8748 ; frame_TS: 1661793531226.631592 ; ros_TS(NSec): 1661793531226631680
[realsense2_camera_node-1] [DEBUG] [1661793531.289124126] [realsense_cam.realsense_cam]: Frameset contain (Depth, 0, Z16) frame. frame_number: 8748 ; frame_TS: 1661793531226.631592 ; ros_TS(NSec): 1661793531226631680
[realsense2_camera_node-1] [DEBUG] [1661793531.289208483] [realsense_cam.realsense_cam]: frame: 1280 x 720
[realsense2_camera_node-1] [DEBUG] [1661793531.289259942] [realsense_cam.realsense_cam]: publishFrame(...)
[realsense2_camera_node-1] [DEBUG] [1661793531.291451211] [realsense_cam.realsense_cam]: Depth stream published, message address: 0xffff41209ea0
[realsense2_camera_node-1] [DEBUG] [1661793531.291857412] [realsense_cam.realsense_cam]: Frameset arrived.
[realsense2_camera_node-1] [DEBUG] [1661793531.291943465] [realsense_cam.realsense_cam]: List of frameset before applying filters: size: 1
[realsense2_camera_node-1] [DEBUG] [1661793531.291999340] [realsense_cam.realsense_cam]: Frameset contain (Depth, 0, Z16 0) frame. frame_number: 8749 ; frame_TS: 1661793531259.886963 ; ros_TS(NSec): 1661793531259886848
[realsense2_camera_node-1] [DEBUG] [1661793531.292080369] [realsense_cam.realsense_cam]: num_filters: 11
[realsense2_camera_node-1] [DEBUG] [1661793531.297640419] [realsense_cam.realsense_cam]: List of frameset after applying filters: size: 2
[realsense2_camera_node-1] [DEBUG] [1661793531.297817774] [realsense_cam.realsense_cam]: Frameset contain (Depth, 0, XYZ32F) frame. frame_number: 8749 ; frame_TS: 1661793531259.886963 ; ros_TS(NSec): 1661793531259886848
[realsense2_camera_node-1] [DEBUG] [1661793531.298029306] [realsense_cam.realsense_cam]: Frameset contain (Depth, 0, Z16) frame. frame_number: 8749 ; frame_TS: 1661793531259.886963 ; ros_TS(NSec): 1661793531259886848
[realsense2_camera_node-1] [DEBUG] [1661793531.298109599] [realsense_cam.realsense_cam]: frame: 1280 x 720
[realsense2_camera_node-1] [DEBUG] [1661793531.298160610] [realsense_cam.realsense_cam]: publishFrame(...)
[realsense2_camera_node-1] [DEBUG] [1661793531.301872740] [realsense_cam.realsense_cam]: Depth stream published, message address: 0xffff41209ea0
[realsense2_camera_node-1] [DEBUG] [1661793531.302473512] [FrequencyStatus_debug_logger]: TICK 6498
[realsense2_camera_node-1] [DEBUG] [1661793531.306614820] [realsense_cam.realsense_cam]: Frameset arrived.
[realsense2_camera_node-1] [DEBUG] [1661793531.306867859] [realsense_cam.realsense_cam]: List of frameset before applying filters: size: 1
[realsense2_camera_node-1] [DEBUG] [1661793531.306991674] [realsense_cam.realsense_cam]: Frameset contain (Color, 0, RGB8 3) frame. frame_number: 8740 ; frame_TS: 1661793530860.797363 ; ros_TS(NSec): 1661793530860797440
[realsense2_camera_node-1] [DEBUG] [1661793531.307089664] [realsense_cam.realsense_cam]: num_filters: 11
[realsense2_camera_node-1] [DEBUG] [1661793531.307221416] [realsense_cam.realsense_cam]: List of frameset after applying filters: size: 1
[realsense2_camera_node-1] [DEBUG] [1661793531.307278764] [realsense_cam.realsense_cam]: Frameset contain (Color, 0, RGB8) frame. frame_number: 8740 ; frame_TS: 1661793530860.797363 ; ros_TS(NSec): 1661793530860797440
[realsense2_camera_node-1] [DEBUG] [1661793531.307358673] [realsense_cam.realsense_cam]: frame: 1280 x 720
[realsense2_camera_node-1] [DEBUG] [1661793531.307410324] [realsense_cam.realsense_cam]: publishFrame(...)
[realsense2_camera_node-1] [DEBUG] [1661793531.309945102] [FrequencyStatus_debug_logger]: TICK 5942
[... cut ...]
Frameset contain (Color, 0, RGB8 3) frame. frame_number: 8742 num_filters: 11
List of frameset after applying filters: size: 1
Frameset contain (Color, 0, RGB8) frame. frame_number: 8742 
frame: 1280 x 720
publishFrame(...)
Color stream published
```

Perfect execution should always be scenario 1, always having pointcloud data with color texture available as an output.
The frames seem to not be in sync, and it can be seen that the frame number of the filtered frames are going back and forth..

Would you have any idea on how to avoid that?

Thank you

MartyG-RealSense, 2022-08-30:
@iraadit  You should not need to set **enable_sync** to true in your launch instruction as the ROS wrapper documentation for this parameter states that it happens automatically when filters such as **pointcloud** are enabled.  Likewise, the decimation filter is false by default in the rs_launch.py file and so should not need setting to false in the launch instruction.

How does it perform if you remove **depth_module.exposure:=32000 depth_module.enable_auto_exposure:=false** and use auto-exposure?

iraadit, 2022-08-30:
Hi @MartyG-RealSense,

_enble_sync_ as well as _decimation_filter.enable_ are in the launch command because it is easier for me that way to try out different parameters (I don't have to remember their name when I want to try them out, but just to change _true_ to _false_ (and vice-versa)).

By the way, I think that _pointcloud.enable:=true_ must also turn _align_depth.enable_ to true, what is happening then if I add _align_depth.enable:=false_ in the launch command? Is it ignored?

As explained in the point 1 of https://github.com/IntelRealSense/realsense-ros/issues/2396#issuecomment-1219842074, I have to add **depth_module.exposure:=32000 depth_module.enable_auto_exposure:=false** to be able to get 30 FPS in input for depth, if not it saturates at 15FPS. By default, **depth_module.enable_auto_exposure** is true, but doesn't seem to work. (???) If I put **depth_module.enable_auto_exposure** to false, it will use the default value of **depth_module.exposure** that is 33000 (33ms) and seems to be too much to get 30FPS. If I change the value of **depth_module.exposure** to be lower (here 32000), it can runt 30FPS. These FPS values are seen thanks to the /diagnostics topic.

MartyG-RealSense, 2022-08-30:
I would assume that a 'false' state of align_depth will be overriden to 'true' if the pointcloud filter is enabled.

Your colleague @Doch88 reported earlier in the discussion that enabling auto_exposure and disabling **auto_exposure_priority** instead of using manual exposure in order to enforce a constant FPS) did not seem to work.

https://github.com/IntelRealSense/realsense-ros/issues/2396#issuecomment-1175140329 states that you cannot reduce your resolution from 1280x720 to 848x480 to improve performance.

I believe we have covered almost all of the available possibilities for improving performance during this 2 month discussion, unfortunately.

iraadit, 2022-08-31:
Hi @MartyG-RealSense ,
I believe there is still more into this, by modifying code if needed, I'll continue to look into it :)

MartyG-RealSense, 2022-08-31:
Thanks so much @iraadit   :)  Please do feedback any findings that you make.

iraadit, 2022-09-12:
Hi @MartyG-RealSense,

I finally found a way to get a (nearly) 30FPS textured pointcloud, but it is suboptimal. It consists of these different elements:
1. 1280x720x30 for depth and color
2. decimation_filter.enable:=true decimation_filter.filter_magnitude:=3 (kinda work with :=2 (default))
4. json_file_path:=ABSOLUTE_PATH_TO/HighAccuracyPreset.json
5. depth_module.enable_auto_exposure:=false depth_module.exposure:=32000 
6. OpenGL implementation of _get_texture_map_
7. Using the script _topic_hz.py_

With that, I can get nearly 30FPS (more than 25) on color, depth and pointcloud.
But it is at the expense of a lower resolution of the pointcloud (due to use of the High Accuracy preset followed by the decimation filter).

### topic_hz.py

With _rviz2_, I could see the streams framerates dropping when I was using _ros2 topic hz_. Searching for it on internet, I found [this post](https://answers.ros.org/question/350753/ros2-topic-hz-provides-wrong-rate-for-larger-msgs/) describing the same thing, started to write my own suscriber to calculate the FPS aand then discovered it had already been done on realsense-ros repo. For reference, the script is here: [realsense2_camera/scripts/topic_hz.py](https://github.com/IntelRealSense/realsense-ros/blob/37041f661620eaaab5310b5dc0e0969b3bf41481/realsense2_camera/scripts/topic_hz.py). Using this script, I can (finally) get good framerates values. The script is only present in the ros2-beta branch (as it seems it wasn't a problem with ROS 1).

### Depth module

32000 instead of 33000 or auto (because then saturate at 15FPS).

For the exposure of the depth module, would there be something similar to _auto_exposure_priority_ for the color stream?

#### Preset

I got the preset from [the page D400-Series-Visual-Presets wiki on the librealsense repo](https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets).

### Decimation

On this wiki page, it is also said:

> The key to good depth is to start from our recommended defaults:
> 
>     D435: Use 848x480 resolution @30fps, with auto-exposure. Use post processing with downsample 2.
>     D415: Use 1280x720 resolution @30fps, with auto-exposure. Use post processing with downsample 3.

What would be the recommended default for D455?

Concerning decimation, I also saw on the [web documentation of intelrealsense](https://dev.intelrealsense.com/docs/opencv-wrapper#downsample-step) that there would be a better implementation of decimation for collision avoidance problems. I'll add it in our code.

># Decimation
>
> ##  Downsample Step
>
> Downsampling is a very common first step in any depth processing algorithm. The key observation is that downsampling reduces spatial (X-Y) accuracy but preserves Z-accuracy.
> It is so widespread that the SDK offers built-in downsampling method in form of [`rs2::decimation_filter`](https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md#decimation-filter).
> It's important to note that using standard OpenCV downsampling is not ideal for depth images.
> In this example we show another correct way to implement depth downsampling. It is conceptually similar to `rs2::decimation_filter`, picking one of the non-zero depth values for every 4x4 block, but unlike `rs2::decimation_filter` it is picking the *closest* depth value instead of median value. This makes sense in context of collision avoidance, since we want to preserve the minimal distance to an object.
>
> Naive implementation for this approach:
>
> C++
>
> ```cplusplus
> for (int y = 0; y < sizeYresized; y++)
>         for (int x = 0; x < source.cols; x += DOWNSAMPLE_FACTOR)
>         {
>             uint16_t min_value = MAX_DEPTH;
> 
>             // Loop over 4x4 quad
>             for (int i = 0; i < DOWNSAMPLE_FACTOR; i++)
>                 for (int j = 0; j < DOWNSAMPLE_FACTOR; j++)
>                 {
>                     auto pixel = source.at<uint16_t>(y * DOWNSAMPLE_FACTOR + i, x + j);
>                     // Only include non-zero pixels in min calculation
>                     if (pixel) min_value = std::min(min_value, pixel);
>                 }
> 
>             // If no non-zero pixels were found, mark the output as zero
>             if (min_value == MAX_DEPTH) min_value = 0;
> 
>             pDest->at<uint16_t>(y, x / DOWNSAMPLE_FACTOR) = min_value;
>         }
> ```

I also have a modification of the code to execute part of the pointcloud calculation with OpenGL. I should test without this modification to see if I still get the nearly 30 FPS. (CPU usage is for sure down with this modification, as well as pointcloud processing is faster).

### Conclusion

Can you help me with any of the questions I have?

I will finally have to let this case at rest in the coming days to work on other things. I'm still not satisfied with the result and it can be said for sure that the combination of D455, ROS 2 (Humble - Isaac), librealsense 2.51.1, realsense-ros (ros2-beta) and trying to get a textured pointcloud with the highest resolution possible on a Jetson AGX Xavier (Jetpack 4 or 5) is giving subpar results for now.

I will submit some Pull Requests in the coming days.

Thank you for your help

MartyG-RealSense, 2022-09-13:
Thanks so much @iraadit for sharing such detailed feedback of your tests!

Using the **Medium Density** preset - MedDensityPreset.json - may provide a better image than High Accuracy, as HA tends to greatly reduce the amount of detail on the depth image due to confidence-filtering of depth coordinates, whilst Medium Density provides a good balance between accuracy and the amount of detail on the image.

Regarding a similar option to the RGB-only auto_exposure_priority for the depth stream: when auto-exposure is disabled then a constant FPS can be enforced if the manual epxosure value is within a certain range, as mentioned earlier in this discussion at https://github.com/IntelRealSense/realsense-ros/issues/2396#issuecomment-1231663581 but there is not a direct depth-stream equivalent for auto_exposue_priority

The recommended resolution setting for optimal depth accuracy on the D455 camera model is the same as those for D435 / D435i - 848x480 depth at 30 FPS.

MartyG-RealSense, 2022-09-19:
Does anyone who commented on this case require further assistance, please?  Thanks!

AndreV84, 2022-09-19:
not from my side. thanks for following up though

MartyG-RealSense, 2022-09-19:
Thanks very much, @AndreV84  :)

AndreV84, 2022-09-19:
@MartyG-RealSense do you know by any chance
a framework/model/ example
which will detect 3d objects from recorded by realsense pointcloud files? 
Thanks

MartyG-RealSense, 2022-09-19:
@AndreV84  If you mean using .ply pointcloud files then **PyTorch 3D** might be a suitable option if you are able to use Python.

**Pytorch3D**
https://pytorch3d.org/

https://ai.facebook.com/blog/building-3d-deep-learning-models-with-pytorch3d/

**Loading ply files into PyTorch3D**
https://pytorch3d.org/docs/meshes_io

iraadit, 2022-09-27:
Hi @MartyG-RealSense

Sorry, I've been busy on something else for the last two weeks.

I'll work again on the realsense pointcloud optimization this week.

MartyG-RealSense, 2022-09-27:
No problem at all, @iraadit - thanks very much for the update and good luck!

iraadit, 2023-01-10:
Hi @MartyG-RealSense,
We decided to wrap up our work on the realsense optimization a moment ago, it went better thanks to the different tricks mentioned before, but there would certainly still be  room to optimize more.
Thank you for your help.
You can close the ticket.

closed 2023-01-10 by MartyG-RealSense