realsenseai/realsense-ros #2397: By not providing "Findlibrealsense2.cmake" in CMAKE_MODULE_PATH this   project has asked CMake to find a package configuration file provided by   "librealsense2", but CMake did not find one.

liyk14, 2022-07-04:
ubuntu20.04 ;ros2:foxy;ros2_intel_realsense;
I installed ros2:foxy, pull down the feature pack from [](https://github.com/intel/ros2_intel_realsense/) 
Install the realsense SDK at   [](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md,)   the SDK can be used normally
, but the function package build was unsuccessful, and an error was reported

>root@237ecce90d8b:~/ros2_ws# colcon build
Starting >>> realsense_camera_msgs
Finished <<< realsense_camera_msgs [0.43s]                    
Starting >>> realsense_ros2_camera
--- stderr: realsense_ros2_camera                         
CMake Error at CMakeLists.txt:65 (find_package):
  By not providing "Findlibrealsense2.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "librealsense2", but CMake did not find one.

  Could not find a package configuration file provided by "librealsense2"
  with any of the following names:

    librealsense2Config.cmake
    librealsense2-config.cmake

  Add the installation prefix of "librealsense2" to CMAKE_PREFIX_PATH or set
  "librealsense2_DIR" to a directory containing one of the above files.  If
  "librealsense2" provides a separate development package or SDK, be sure it
  has been installed.
---
Failed   <<< realsense_ros2_camera [0.90s, exited with code 1]

Summary: 1 package finished [1.48s]
  1 package failed: realsense_ros2_camera
  1 package had stderr output: realsense_ros2_camera

I have confirmed that the corresponding version of the function package is installed   sudo apt install ros-foxy-librealsense2,
Tried changing the CMake file's, file directory, but failed.

MartyG-RealSense, 2022-07-04:
Hi @liyk14  I note that your log refers to **realsense_ros2_camera**.  It looks as though you might be installing the obsolete RealSense ROS2 wrapper at the link below.  Is this the wrapper that you are using, please?

https://github.com/intel/ros2_intel_realsense

If you are using that old wrapper which is no longer supported then the current RealSense ROS2 wrapper, which is compatible with Foxy, can be found at the link below.

https://github.com/IntelRealSense/realsense-ros/tree/ros2-beta

liyk14, 2022-07-05:
Hi@MartyG-RealSense  Thank you for your advice.I tried to get this resource and have solved the problem.
nice! nice!

MartyG-RealSense, 2022-07-05:
It's excellent news that you achieved a solution!  Do you require further assistance, please?  Thanks!

closed 2022-07-11 by MartyG-RealSense